margelo/react-native-vision-camera #1986: Error Domain=AVFoundationErrorDomain Code=-11800 The operation could not be completed UserInfo={NSUnderlyingError=0x28006b360 { Error Domain=NSOSStatusErrorDomain Code=-16800 "(null)"}, NSLocalizedFailureReason=Anunknown error occurred (-16800), AVErrorRecordingFailureDomainKey=4, NSLocalizedDescription=The operation could not be completed}🐛

LESANF, 2023-10-12:
### What's happening?

Please watch the video below.

https://github.com/mrousavy/react-native-vision-camera/assets/54767632/13da3839-c112-45dc-ae5e-d751797b3bd8

https://github.com/mrousavy/react-native-vision-camera/assets/54767632/1621cc18-65ea-4c56-9abd-9ffc7d498786

========================================================================================

The error only occurs under completely specific conditions.

In my opinion, this condition is when using IOS17 or higher on iPhone 14

The error did not occur when using IOS16 on iPhone 14.

I can't confirm that it's a bug in IOS17. Because when using IOS17 on other devices such as SE, 12, etc., the bug does not occur.

I want to know what the problem is.

The build took place in xcode14.2.

### Reproduceable Code

```tsx
const device = useCameraDevice('back');

try { 
    if (tempPath === '') {
        setIsActive(false);
                const photo =
                    Platform.OS === 'ios'
                        ? await cameraRef.current.takePhoto({
                              flash: 'off',
                          })
                        : await cameraRef.current.takeSnapshot({
                              quality: 80,
                              flash: 'off',
                          });
                let rotatedUri = photo.path;

                await foodLensProcess(rotatedUri);
            } else {
                setTempPath('');
                setFoodInfo(null);
                setUploadPath('');
                setIsActive(true);
            }
        } catch (err) {
            alert(err);
            console.log('takePhoto catch :', err);
        }

<Camera
    ref={cameraRef}
    style={StyleSheet.absoluteFill} 
    device={device}
    isActive={isActive}
    photo={true}
/>
```

### Relevant log output

```shell
Error Domain=AVFoundationErrorDomain Code=-11800 The operation could not be completed UserInfo={NSUnderlyingError=0x28006b360 { Error Domain=NSOSStatusErrorDomain Code=-16800 "(null)"}, NSLocalizedFailureReason=Anunknown error occurred (-16800), AVErrorRecordingFailureDomainKey=4, NSLocalizedDescription=The operation could not be completed}
```

### Camera Device

```json
-
```

### Device

iPhone 14 Pro(IOS 17.0.2, 17.0.3)

### VisionCamera Version

3.3.1

### Can you reproduce this issue in the VisionCamera Example app?

Yes, I can reproduce the same issue in the Example app here

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

LESANF, 2023-10-12:
@mrousavy Thank you for changing the title

mrousavy, 2023-10-16:
Is this related to https://github.com/mrousavy/react-native-vision-camera/issues/1923?

mrousavy, 2023-10-16:
As per [this StackOverflow post](https://stackoverflow.com/questions/24176738/error-domain-avfoundationerrordomain-code-11814-cannot-record): 

> Having your device restrict the camera access under "Settings > General > Restrictions" will also give you this error.

Could it be that the user disabled Camera access? In that case, `useCameraPermission()` would probably return `"restricted"`. Do you check for that?

mrousavy, 2023-10-16:
Also maybe [searching for this error on StackOverflow](https://stackoverflow.com/search?q=%22-11800%22+The+operation+could+not+be+completed) could be any help. If you find anything useful that works for you, please let me know!

LESANF, 2023-10-16:
> As per [this StackOverflow post](https://stackoverflow.com/questions/24176738/error-domain-avfoundationerrordomain-code-11814-cannot-record):
>
>
> Could it be that the user disabled Camera access? In [...]

<hr/>

First of all, thank you for replying.

It looks similar to the #1923 you told me, but I think it's a little different

But it looks very similar, but the reason why my answer might be ambiguous is that I haven't figured out the exact cause yet

I tested it with a lot of IOS models and found the exact case

Only with 14 or higher and at the same time with IOS 17 or higher.

It worked exactly on iPhone 14 Pro and IOS 16 versions. I've tested several iPhone 14 Pro models

> You gave me an opinion about Kwon Hwan and I've already tested it.
> If you didn't grant the camera permission, the camera components themselves from vision shouldn't have appeared
> But Component worked fine and it's an error log from the takephoto method

I still haven't solved this problem

Eventually, I replaced the library with react-native-camera.

I'm curious about the cause, too

LESANF, 2023-10-16:
@mrousavy I'm sorry, I accidentally clicked on 'closed'

kcstechnis, 2023-10-24:
We have similar issues with the QR code scanner on iOS.  

```
[unknown/unknown: Error Domain=AVFoundationErrorDomain Code=-11800 "The operation could not be completed" UserInfo={NSLocalizedFailureReason=An unknown error occurred (-12780), NSLocalizedDescription=The operation could not be completed, NSUnderlyingError=0x282782700 {Error Domain=NSOSStatusErrorDomain Code=-12780 "(null)"}}]
```
Version: 3.5.1
Model: iPhone 11
OS: 15.6.1

siemya, 2023-10-26:
I am having a similar problem on Iphone 15, 17.02

```
<Camera
    photo
    isActive
    enableHighQualityPhotos
    video={false}
    audio={false}
    ref={camera}
    device={device}
    style={{flex: 1}}/>
    
    await camera.current.takePhoto({ flash: 'off' })
```

`Error Domain=AVFoundationErrorDomain Code=-11800 "The operation could not be completed" UserInfo={NSUnderlyingError=0x2803dbdb0 {Error Domain=NSOSStatusErrorDomain Code=-16800 "(null)"}, NSLocalizedFailureReason=An unknown error occurred (-16800)`

michelevettone, 2023-10-30:
Same error on iPhone 15 Pro Max - iOS 17.1

<Camera ref={camera} photo={true} style={styles.camera} device={device} isActive={isFocused} />

const photo = await camera.current.takePhoto({
                    flash: 'off',
              });

[capture/unknown: Error Domain=AVFoundationErrorDomain Code=-11800 "L’operazione non è stata completata" UserInfo={NSUnderlyingError=0x2803994a0 {Error Domain=NSOSStatusErrorDomain Code=-16800 "(null)"}, NSLocalizedFailureReason=Si è verificato un errore sconosciuto (-16800), AVErrorRecordingFailureDomainKey=4, NSLocalizedDescription=L’operazione non è stata completata}]

akshaypingala, 2023-11-21:
I am getting bellow error while capture video in iPhone 11 and iOS 17.1.1 but it is working in iPhone 13 Pro Max and iOS 17.1.1.

capture/unknown: An unknown recording error occured! Error Domain=AVFoundationErrorDomain Code=-11800 "The operation could not be completed" UserInfo={NSLocalizedFailureReason=An unknown error occurred (-12780), NSLocalizedDescription=The operation could not be completed, NSUnderlyingError=0x28391f9f0 {Error Domain=NSOSStatusErrorDomain Code=-12780 "(null)"}}

IvanByRool, 2023-12-02:
Hey,

having the same pb with iOS 17.1.1 & iPhone 13 Pro
It's happening randomly 

Thx

randbytes, 2023-12-04:
Hey!
Same here with iphone 15(pm), iOS 17.1.2(Native Swift)

> photoSettings.photoQualityPrioritization = .speed // instead of .quality

thank you very much! This is resolved for now. I spent the entire week debugging this.

mrousavy, 2024-01-15:
Hey! Closing for now as I think this has been fixed in recent VisionCamera updates. If you still experience this error, please re-open this.

mrousavy, 2024-01-15:
If not, this might be an Apple bug: https://forums.developer.apple.com/forums/thread/717565

closed 2024-01-15 by mrousavy

vishaljak, 2024-02-19:
Having issue again on v3.6.16 on all iOS devices. 

Adding  qualityPrioritization: 'speed' to options while taking a picture fixes it. but not sure why its happening in the first place

mrousavy, 2024-02-19:
> but not sure why its happening in the first place

Me neither.

> Adding `qualityPrioritization: 'speed'` to options while taking a picture fixes it.

great! Maybe this will be fixed in future iOS update.s

ElSguidge, 2024-10-23:
`photoSettings.photoQualityPrioritization = .speed // instead of .quality`

Didnt solve it for me.
Anyone else comes across this I fixed it by unmounting the camera using isActive flag as it uses less resources since the Camera will be completely destroyed when unmounted. (from docs)

`
  const [isCameraActive, setIsCameraActive] = useState(false);

  useEffect(() => {
    if (isFocused && hasPermission) {
      setIsCameraActive(true);
    } else {
      setIsCameraActive(false);
    }
  }, [isFocused, hasPermission]);
<Camera
          style={StyleSheet.absoluteFill}
          device={device}
          isActive={isCameraActive}
          codeScanner={codeScanner}
          enableZoomGesture={true}
          onLayout={onLayout}
          torch={flashMode === 'on' ? 'on' : 'off'}
        />
       `

Rakura-cloud, 2025-01-12:
I had the same issue, but it was probably an error on my part. I ran a deactivated camera before taking a photo (no problem without a flash), but when the flash was firing, I got an error. 

Code with error: 
```JS  
    **setIsCameraActive(false);**

      const photo = await camera.current.takePhoto({
        flash: flash,
      });

      console.log(photo);

      setIsCameraActive(true);
```
code worked:

``` JS
const photo = await camera.current.takePhoto({
        flash: flash,
      });

      **setIsCameraActive(false);**

      console.log(photo);

      setIsCameraActive(true);
```

souravpaul-appycodes, 2025-02-17:
My issue has been fixed by adding Format for the camera.

` const cameraDevice = useCameraDevice(device);
  if (!cameraDevice) {
    console.log('No camera device available!');
    return null;
  }
  const screenAspectRatio = SCREEN_HEIGHT / SCREEN_WIDTH;
  const format = useCameraFormat(cameraDevice, [
    {fps: 60},
    {videoAspectRatio: screenAspectRatio},
    {videoResolution: {width: SCREEN_WIDTH, height: SCREEN_HEIGHT}},
  ]);
  return (
    <>
      <Camera
        format={format}
        ref={cameraRef}
        video={true}
        device={cameraDevice}
        style={StyleSheet.absoluteFill}
        isActive={true}
      />`