margelo/react-native-vision-camera #1987: 🔧 Cannot find type 'FrameProcessorPluginBase' in scope

suwu150, 2023-10-12:
### How were you trying to build the app?

Cannot find type 'FrameProcessorPluginBase' in scope 

[image]

### Full build logs

```tsx
Cannot find type 'FrameProcessorPluginBase' in scope
```

### Project dependencies

```json
"@react-native-async-storage/async-storage": "^1.19.3",
    "@shopify/react-native-skia": "^0.1.213",
    "react": "18.2.0",
    "react-native": "0.72.4",
    "react-native-fast-tflite": "^0.3.1",
    "react-native-reanimated": "^3.5.4",
    "react-native-svg": "^13.14.0",
    "react-native-vision-camera": "^3.3.1",
    "react-native-webview": "^13.6.0",
    "react-native-worklets-core": "^0.2.0",
    "url-loader": "^4.1.1",
    "vision-camera-ocr": "^1.0.0"
```

### VisionCamera Version

3.3.1

### Target platforms

iOS

### Operating system

MacOS

### Can you build the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [ ] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

suwu150, 2023-10-12:
https://github.com/rodgomesc/vision-camera-code-scanner/issues/62

mrousavy, 2023-10-12:
I think it's called `FrameProcessorPlugin` now, no?

mrousavy, 2023-10-12:
https://github.com/mrousavy/react-native-vision-camera/blob/54871022f43f7530bdaade67ca52f36e416d0c67/package/ios/Frame%20Processor/FrameProcessorPlugin.h#L14-L28

mrousavy, 2023-10-12:
Where did you even get the name `FrameProcessorPluginBase` from? It's not in the docs either, maybe this is a left-over artefact of a V2 implementation. In V3 it's just called `FrameProcessorPlugin`

closed 2023-10-12 by mrousavy

suwu150, 2023-10-12:
I integrated it here according to the document [https://react-native-vision-camera.com/docs/guides/frame-processor-plugin-list](https://react-native-vision-camera.com/docs/guides/frame-processor-plugin-list) The document integrates plugins, it seems that the plugins here are not compatible with the v3 version anymore

spiropoulos94, 2023-10-29:
Same here for me. All I did is to follow the manual process from `npx vision-camera-plugin-builder ios`. After registering the global plugin name and setting up the `babel.config.js`, I stumbled upon the issue mentioned above

Rakedd, 2023-11-10:
Any updates on the issue ? It is a real boomer because the lib could be convenient

Rakedd, 2023-11-10:
> Where did you even get the name `FrameProcessorPluginBase` from? It's not in the docs either, maybe this is a left-over artefact of a V2 implementation. In V3 it's just called `FrameProcessorPlugin`

Why did you close the issue if no solutions has been found ?

NoahSimajji, 2023-11-15:
Guys? i encounter this issue too. Any solution?

Rakedd, 2023-11-15:
Ok guys here's the deal, the error come from the plugin vision-camera-ocr, not this repo. Related [here](https://github.com/rodgomesc/vision-camera-code-scanner/issues/62), and there's a ongoing PR [here](https://github.com/aarongrider/vision-camera-ocr/pull/32), so i retract my comment @mrousavy, it's indeed closed and not relevant here
And thank you for your work 🫡.

Superstar721, 2024-07-26:
npm install react-native-vision-camera@latest
or
npm install react-native-vision-camera@latest