margelo/react-native-vision-camera #1988: 🐛 App Crashes When Navigating Back to Camera Screen

tronginc, 2023-10-12:
### What's happening?

The app crashes when I navigate back from the camera screen and then attempt to return to the camera screen.

**Steps to Reproduce:**

1. Navigate to the camera screen within the app.
2. Interact with the camera screen as needed.
3. Use the app's navigation to go back to the previous screen.
4. Navigate back to the camera screen.

Error: 
```
java.lang.IllegalArgumentException: Surface was abandoned
```

### Reproduceable Code

```tsx
function App() {
  const device = useCameraDevice('back', {
    physicalDevices: ['wide-angle-camera'],
  });

  if (device == null) return null;
  return (
    <Camera
      style={StyleSheet.absoluteFill}
      device={device}
      isActive={true}
    />
  )
}
```

### Relevant log output

```shell
10-12 15:20:57.082  9477 10289 E AndroidRuntime: FATAL EXCEPTION: mrousavy/VisionCamera.main
10-12 15:20:57.082  9477 10289 E AndroidRuntime: Process: com.dt.johnny.jdtmelhu, PID: 9477
10-12 15:20:57.082  9477 10289 E AndroidRuntime: java.lang.IllegalArgumentException: Surface was abandoned
10-12 15:20:57.082  9477 10289 E AndroidRuntime: 	at android.hardware.camera2.utils.SurfaceUtils.getSurfaceSize(SurfaceUtils.java:135)
10-12 15:20:57.082  9477 10289 E AndroidRuntime: 	at android.hardware.camera2.params.OutputConfiguration.<init>(OutputConfiguration.java:563)
10-12 15:20:57.082  9477 10289 E AndroidRuntime: 	at android.hardware.camera2.params.OutputConfiguration.<init>(OutputConfiguration.java:345)
10-12 15:20:57.082  9477 10289 E AndroidRuntime: 	at com.mrousavy.camera.core.outputs.SurfaceOutput.toOutputConfiguration(SurfaceOutput.kt:38)
10-12 15:20:57.082  9477 10289 E AndroidRuntime: 	at com.mrousavy.camera.extensions.CameraDevice_createCaptureSessionKt.createCaptureSession(CameraDevice+createCaptureSession.kt:57)
10-12 15:20:57.082  9477 10289 E AndroidRuntime: 	at com.mrousavy.camera.core.CameraSession.getCaptureSession(CameraSession.kt:438)
10-12 15:20:57.082  9477 10289 E AndroidRuntime: 	at com.mrousavy.camera.core.CameraSession.startRunning(CameraSession.kt:532)
10-12 15:20:57.082  9477 10289 E AndroidRuntime: 	at com.mrousavy.camera.core.CameraSession.access$startRunning(CameraSession.kt:49)
10-12 15:20:57.082  9477 10289 E AndroidRuntime: 	at com.mrousavy.camera.core.CameraSession$startRunning$1.invokeSuspend(Unknown Source:14)
10-12 15:20:57.082  9477 10289 E AndroidRuntime: 	at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:33)
10-12 15:20:57.082  9477 10289 E AndroidRuntime: 	at kotlinx.coroutines.DispatchedTask.run(DispatchedTask.kt:106)
10-12 15:20:57.082  9477 10289 E AndroidRuntime: 	at android.os.Handler.handleCallback(Handler.java:958)
10-12 15:20:57.082  9477 10289 E AndroidRuntime: 	at android.os.Handler.dispatchMessage(Handler.java:99)
10-12 15:20:57.082  9477 10289 E AndroidRuntime: 	at android.os.Looper.loopOnce(Looper.java:205)
10-12 15:20:57.082  9477 10289 E AndroidRuntime: 	at android.os.Looper.loop(Looper.java:294)
10-12 15:20:57.082  9477 10289 E AndroidRuntime: 	at android.os.HandlerThread.run(HandlerThread.java:67)
10-12 15:20:57.082  9477 10289 E AndroidRuntime: 	Suppressed: kotlinx.coroutines.DiagnosticCoroutineContextException: [StandaloneCoroutine{Cancelling}@9a35690, mrousavy/VisionCamera.main]
```

### Camera Device

```json
{
  "sensorOrientation": "landscape-right",
  "hardwareLevel": "full",
  "maxZoom": 10,
  "minZoom": 1,
  "supportsLowLightBoost": false,
  "neutralZoom": 1,
  "physicalDevices": [
    "wide-angle-camera"
  ],
  "supportsFocus": true,
  "supportsRawCapture": true,
  "isMultiCam": false,
  "name": "BACK (0)",
  "hasFlash": true,
  "hasTorch": true,
  "position": "back",
  "id": "0"
}
```

### Device

Any Android Devices

### VisionCamera Version

3.3.1

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

amserhii, 2023-10-12:
The same issue on my side

Cirych, 2023-10-12:
Same

```
2023-10-12 15:18:03.235 12356-12805 BufferQueueProducer     com.xxx.app                 E  [ImageReader-4080x3072f21m3-12356-0](id:304400000004,api:0,p:-1,c:12356) connect: BufferQueue has been abandoned
2023-10-12 15:18:03.235  1242-8491  Camera3-OutputStream    cameraserver                         E  configureConsumerQueueLocked: Unable to connect to native window for stream 1
2023-10-12 15:18:03.235  1242-8491  Camera3-Stream          cameraserver                         E  finishConfiguration: Unable to configure stream 1 queue (non-fatal): No such device (-19)
2023-10-12 15:18:03.235  1242-8491  Camera3-Device          cameraserver                         E  Camera 0: configureStreamsLocked: Can't finish configuring output stream 1: No such device (-19)
2023-10-12 15:18:03.235  1242-8491  CameraDeviceClient      cameraserver                         E  endConfigure: Camera 0: Error configuring streams: Broken pipe (-32)
2023-10-12 15:18:03.237 12356-12805 CameraCaptureSession    com.xxx.app                 E  Session 0: Failed to create capture session; configuration failed
```
...
```

2023-10-12 15:18:03.536 12356-12805 AndroidRuntime          com.xxx.app                 E  FATAL EXCEPTION: mrousavy/VisionCamera.main
                                                                                                    Process: com.xxx.app, PID: 12356
                                                                                                    android.hardware.camera2.CameraAccessException: CAMERA_ERROR (3): endConfigure:656: Camera 0: Error configuring streams: Broken pipe (-32)
                                                                                                    	at android.hardware.camera2.CameraManager.throwAsPublicException(CameraManager.java:1489)
                                                                                                    	at android.hardware.camera2.impl.ICameraDeviceUserWrapper.endConfigure(ICameraDeviceUserWrapper.java:119)
                                                                                                    	at android.hardware.camera2.impl.CameraDeviceImpl.configureStreamsChecked(CameraDeviceImpl.java:519)
                                                                                                    	at android.hardware.camera2.impl.CameraDeviceImpl.createCaptureSessionInternal(CameraDeviceImpl.java:726)
                                                                                                    	at android.hardware.camera2.impl.CameraDeviceImpl.createCaptureSession(CameraDeviceImpl.java:680)
                                                                                                    	at com.mrousavy.camera.extensions.CameraDevice_createCaptureSessionKt.createCaptureSession(CameraDevice+createCaptureSession.kt:82)
                                                                                                    	at com.mrousavy.camera.core.CameraSession.getCaptureSession(CameraSession.kt:438)
                                                                                                    	at com.mrousavy.camera.core.CameraSession.startRunning(CameraSession.kt:532)
                                                                                                    	at com.mrousavy.camera.core.CameraSession.access$startRunning(CameraSession.kt:49)
                                                                                                    	at com.mrousavy.camera.core.CameraSession$startRunning$1.invokeSuspend(Unknown Source:14)
                                                                                                    	at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:33)
                                                                                                    	at kotlinx.coroutines.DispatchedTask.run(DispatchedTask.kt:106)
                                                                                                    	at android.os.Handler.handleCallback(Handler.java:942)
                                                                                                    	at android.os.Handler.dispatchMessage(Handler.java:99)
                                                                                                    	at android.os.Looper.loopOnce(Looper.java:201)
                                                                                                    	at android.os.Looper.loop(Looper.java:288)
                                                                                                    	at android.os.HandlerThread.run(HandlerThread.java:67)
                                                                                                    	Suppressed: kotlinx.coroutines.DiagnosticCoroutineContextException: [StandaloneCoroutine{Cancelling}@87284c4, mrousavy/VisionCamera.main]
                                                                                                    Caused by: android.os.ServiceSpecificException: endConfigure:656: Camera 0: Error configuring streams: Broken pipe (-32) (code 10)
                                                                                                    	at android.os.Parcel.createExceptionOrNull(Parcel.java:3025)
[... cut ...]
                                                                                                    	at android.hardware.camera2.ICameraDeviceUser$Stub$Proxy.endConfigure(ICameraDeviceUser.java:705)
                                                                                                    	at android.hardware.camera2.impl.ICameraDeviceUserWrapper.endConfigure(ICameraDeviceUserWrapper.java:116)
                                                                                                    	... 15 more
```

tupywebteam, 2023-10-12:
The same issue

JoeDareZone, 2023-10-12:
I had some similar issues and found using navigation.replace instead of navigation.navigate worked for me.

CarlosMayoralMX, 2023-10-12:
I added `isFocused` to my camera implementation because it was crashing everytime i go back from another screen

`{ device != null && isFocused ? (...`

Also I changed` isActive={true}` to use `isActive={isActive}` where isActive=

```
 // check if camera page is active
  const isFocussed = useIsFocused() // Implementation from '@react-navigation/core'
  const isForeground = useIsForeground() // Hook you can find on the Example app inside the package
  const isActive = isFocussed && isForeground
```

mrousavy, 2023-10-13:
I think this will get fixed with this: https://github.com/mrousavy/react-native-vision-camera/pull/1996

mrousavy, 2023-10-14:
Yea basically I need to do a refactor on the Codebase - I now know exactly what I need for the session configuration, and need to do all of this fully atomically.

810505339, 2023-10-18:
> 我认为这将通过以下方式解决：#1996

thinks

810505339, 2023-10-19:
https://github.com/mrousavy/react-native-vision-camera/assets/50100966/79b7cc8b-5a9c-4467-97d9-55a301eb81dd

Click the back button to see the following situation 

```ts
  const device = useCameraDevice('back');
  const camera = useRef<Camera>(null);
  const isFocused = useIsFocused();
  const navigation = useNavigation();
  const appState = useAppState();
  const [active, setActive] = useState(true);
  useEffect(() => {
    const unsubscribe = navigation.addListener('beforeRemove', () => {
      setTimeout(() => {
        setActive(false);
      }, 0);
    });
    return unsubscribe;
  }, [navigation]);

  const isActive = active && isFocused && appState === 'active';

  const takePhoto = async () => {
    const photo = await camera.current!.takePhoto();
    console.log(photo);
  };

  if (!device) {
    return (<View className="flex-1">
      <Text>请打开摄像头</Text>
    </View>);

  }

  if (!isActive) {
    return (<View className="flex-1">
      <Text>请打开摄像头</Text>
    </View>);
  }

  return <View className="h-full w-full items-center justify-end">
    {isActive && <Camera
      ref={camera}

      style={{ position: 'absolute', left: 0, right: 0, bottom: 0, top: 0 }}
      //https://github.com/mrousavy/react-native-vision-camera/issues/1988  不然会崩溃
      fps={60}

      device={device} //此相机设备包含的物理设备类型列表。
      // video={true} //录像功能打开关闭
      // supportsVideoHDR={true}
      isActive={isActive} //是否打开相机， 可以缓存相机，加快打开速度
      photo={true} //拍照功能是否打开
    />}
    <TouchableOpacity className=" w-16 h-16 rounded-full bg-slate-500 mt-[100%] mb-[20%]" onPress={takePhoto} />
  </View>;
```

mrousavy, 2023-10-19:
@810505339 can you share your adb logcat logs

R4tch3t, 2023-10-19:
NOT FIXED FOR ME I HAVE "react-native-vision-camera": "^3.5.1", "react-native-worklets-core": "^0.2.2", "react-native-reanimated": "^3.5.4"
```
2023-10-19 15:37:23.543 10007-10386 BufferQueueProducer     com.adonaysoft.stepsV2               I  [ImageReader-4000x1800f100m3-10007-0](id:271700000004,api:4,p:1195,c:10007) disconnect: api 4
2023-10-19 15:37:23.546 10007-10386 BufferQueueProducer     com.adonaysoft.stepsV2               I  [ImageReader-1920x864f23m3-10007-1](id:271700000006,api:4,p:1195,c:10007) disconnect: api 4
2023-10-19 15:37:23.556 10007-10386 SurfaceOutput           com.adonaysoft.stepsV2               I  Closing 4000x1800 PHOTO ImageReader..
2023-10-19 15:37:23.556 10007-10386 BufferQueueConsumer     com.adonaysoft.stepsV2               I  [ImageReader-4000x1800f100m3-10007-0](id:271700000004,api:0,p:-1,c:10007) disconnect
2023-10-19 15:37:23.556 10007-10386 SurfaceOutput           com.adonaysoft.stepsV2               I  Closing 1920x864 Video Pipeline..
2023-10-19 15:37:23.556 10007-10386 BufferQueueProducer     com.adonaysoft.stepsV2               I  [SurfaceTexture-0-10007-1](id:271700000005,api:4,p:10007,c:10007) disconnect: api 4
2023-10-19 15:37:23.557 10007-10386 BufferQueueConsumer     com.adonaysoft.stepsV2               I  [ImageReader-1920x864f23m3-10007-1](id:271700000006,api:0,p:-1,c:10007) disconnect
2023-10-19 15:37:23.557 10007-10386 BufferQueueConsumer     com.adonaysoft.stepsV2               I  [SurfaceTexture-0-10007-1](id:271700000005,api:0,p:-1,c:10007) disconnect
2023-10-19 15:37:23.557 10007-10386 CameraSession           com.adonaysoft.stepsV2               I  Camera Session stopped!
2023-10-19 15:37:23.558 10007-10386 CameraSession           com.adonaysoft.stepsV2               I  Setting isActive: false (isRunning: false)
2023-10-19 15:37:23.558 10007-10386 CameraSession           com.adonaysoft.stepsV2               I  Camera became available: 0
2023-10-19 15:37:23.558 10007-10235 CameraDevices           com.adonaysoft.stepsV2               I  Camera #0: Available!
2023-10-19 15:37:23.587 10007-10232 BLASTBufferQueue        com.adonaysoft.stepsV2               I  releaseBufferCallbackThunk bufferId:42979737731137 framenumber:1253 blastBufferQueue is dead
2023-10-19 15:37:23.798 10007-10234 ReactNativeJS           com.adonaysoft.stepsV2               I  topAf: 458x639
2023-10-19 15:37:24.069 10007-10387 libc                    com.adonaysoft.stepsV2               A  Fatal signal 11 (SIGSEGV), code 2 (SEGV_ACCERR), fault addr 0x71b16f2000 in tid 10387 (ionCamera.video), pid 10007 (naysoft.stepsV2)
```

810505339, 2023-10-20:
> @810505339 can you share your adb logcat logs

```
10-20 01:49:08.428 E/WifiScoringParams(  553): Invalid frequency(-1), using 5G as default rssi array
10-20 01:49:08.543 E/Transition(  553): Trying to add a ready-group twice: Display{#0 state=ON size=1440x3120 ROTATION_0}
10-20 01:49:15.445 E/BufferQueueProducer( 5135): [SurfaceView[com.myapp/com.myapp.MainActivity]#3(BLAST Consumer)3](id:140f00000003,api:4,p:468,c:5135) queueBuffer: BufferQueue has been abandoned
10-20 01:49:15.446 E/Surface (  468): queueBuffer: error queuing buffer, -19
10-20 01:49:15.446 E/Camera3-OutputStream(  468): returnBufferCheckedLocked: Stream 0: Error queueing buffer to native window: No such device (-19)
10-20 01:49:15.446 E/BufferQueueProducer( 5135): [SurfaceView[com.myapp/com.myapp.MainActivity]#3(BLAST Consumer)3](id:140f00000003,api:4,p:468,c:5135) dequeueBuffer: BufferQueue has been abandoned
10-20 01:49:15.447 E/Camera3-OutputStream(  468): getBufferLockedCommon: Stream 0: Can't dequeue next output buffer: No such device (-19)
10-20 01:49:15.476 E/BufferQueueProducer( 5135): [SurfaceView[com.myapp/com.myapp.MainActivity]#3(BLAST Consumer)3](id:140f00000003,api:4,p:468,c:5135) queueBuffer: BufferQueue has been abandoned
10-20 01:49:15.476 E/Surface (  468): queueBuffer: error queuing buffer, -19
10-20 01:49:15.477 E/BufferQueueProducer( 5135): [SurfaceView[com.myapp/com.myapp.MainActivity]#3(BLAST Consumer)3](id:140f00000003,api:4,p:468,c:5135) cancelBuffer: BufferQueue has been abandoned
10-20 01:49:15.478 E/BufferQueueProducer( 5135): [SurfaceView[com.myapp/com.myapp.MainActivity]#3(BLAST Consumer)3](id:140f00000003,api:4,p:468,c:5135) cancelBuffer: BufferQueue has been abandoned

```

mrousavy, 2023-10-20:
i think i can fix this in #2049

810505339, 2023-10-22:
> 我想我可以在#2049中解决此问题

I update ```"react-native-vision-camera": "^3.5.1" ````  ,It seems that the problem still exists .

810505339, 2023-10-23:
> > 我想我可以在#2049中解决此问题
> 
> I update ```"react-native-vision-camera": "^3.5.1" ```` ,It seems that the problem still exists .

I tried adding key and solved the problem   ``` <Camera
      ref={camera}
      key={device.id}```
https://github.com/mrousavy/react-native-vision-camera/issues/1773#issuecomment-1710524501

closed 2023-10-24 by mrousavy

mrousavy, 2023-10-24:
I just fixed this in #2049, and released it to npm (v3.6.0) 🎉 

If you appreciate my work and dedication to fix your issues and make VisionCamera better, please [💖 consider sponsoring me on GitHub 💖](https://github.com/sponsors/mrousavy) to say thanks. Thanks!