margelo/react-native-vision-camera #1990: 🐛 Frame processor crash on Android

xseignard, 2023-10-12:
### What's happening?

After a few seconds (random, from 1s to 20s), the frame processor crashes on Android with a good old:
```
signal 11 (SIGSEGV), code 1 (SEGV_MAPERR), fault addr 0x0000000000000000
Cause: null pointer dereference
```

I post below the [`ndk-stack`](https://developer.android.com/ndk/guides/ndk-stack) symbolicated stack trace (truncated due to github char limits) obtained with `$ANDROID_HOME/ndk/23.1.7779620/ndk-stack -dump my_crash.txt -sym android/app/build/intermediates/merged_native_libs/debug/out/lib/arm64-v8a`

Looking at the logs it seems that `this->frame` is null when calling `get` on `FrameHostObject` [here](https://github.com/mrousavy/react-native-vision-camera/blob/main/package/android/src/main/cpp/frameprocessor/FrameHostObject.cpp#L52)

Any idea how/why?

### Reproduceable Code

```tsx
const measurePupillaryDistanceInPixelsJS = Worklets.createRunInJsFn(measurePupillaryDistanceInPixels);

const frameProcessor = useFrameProcessor((frame: CameraFrame) => {
  "worklet";
  runAtTargetFps(frameProcessorFps, () => {
    "worklet";
    if (!frame.isValid) return;
    const result = scanForEyes(frame); // custom frame processor
    measurePupillaryDistanceInPixelsJS({
      frame,
      face: result?.faces?.[0],
    });
  });
}, []);

//...

<Camera
  style={styles.camera}
  device={device}
  isActive
  frameProcessor={frameProcessor}
  pixelFormat={Platform.OS === "ios" ? "native" : "yuv"}
  enableZoomGesture={false}
  zoom={1}
/>
```

### Relevant log output

```shell
********** Crash dump: **********
Build fingerprint: 'google/redfin/redfin:13/TQ3A.230901.001/10750268:user/release-keys'
#00 0x00000000000418f0 /system/lib64/libmedia_jni.so (Image_getFormat(_JNIEnv*, _jobject*, int)+100) (BuildId: 012e7bc47c226602b2c4185b894c1c3c)
#01 0x000000000034dc30 /apex/com.android.art/lib64/libart.so (art_quick_generic_jni_trampoline+144) (BuildId: a5fcf27f4a71b07dff05c648ad58e3cd)
#02 0x00000000005d0530 /apex/com.android.art/lib64/libart.so (nterp_helper+4016) (BuildId: a5fcf27f4a71b07dff05c648ad58e3cd)
#03 0x000000000033c8c2 /system/framework/framework.jar (android.media.ImageReader$SurfaceImage.getFormat+46)
#04 0x00000000005d04d4 /apex/com.android.art/lib64/libart.so (nterp_helper+3924) (BuildId: a5fcf27f4a71b07dff05c648ad58e3cd)
#05 0x000000000033c9b2 /system/framework/framework.jar (android.media.ImageReader$SurfaceImage.getWidth+6)
#06 0x00000000005d04d4 /apex/com.android.art/lib64/libart.so (nterp_helper+3924) (BuildId: a5fcf27f4a71b07dff05c648ad58e3cd)
#07 0x000000000033f356 /system/framework/framework.jar (android.media.Image.getCropRect+18)
#08 0x00000000003371a4 /apex/com.android.art/lib64/libart.so (art_quick_invoke_stub+612) (BuildId: a5fcf27f4a71b07dff05c648ad58e3cd)
#09 0x0000000000519790 /apex/com.android.art/lib64/libart.so (bool art::interpreter::DoCall<false>(art::ArtMethod*, art::Thread*, art::ShadowFrame&, art::Instruction const*, unsigned short, bool, art::JValue*)+1976) (BuildId: a5fcf27f4a71b07dff05c648ad58e3cd)
#10 0x00000000004a2e3c /apex/com.android.art/lib64/libart.so (void art::interpreter::ExecuteSwitchImplCpp<false>(art::interpreter::SwitchImplContext*)+960) (BuildId: a5fcf27f4a71b07dff05c648ad58e3cd)
#11 0x00000000003503d8 /apex/com.android.art/lib64/libart.so (ExecuteSwitchImplAsm+8) (BuildId: a5fcf27f4a71b07dff05c648ad58e3cd)
#12 0x000000000011e638 [anon:dalvik-classes2.dex extracted in memory from /data/app/~~BMVBv-pBIFrTSW-JQ9axLA==/com.REDACTED-Xs1dulzccjN40HiZwN4c-A==/base.apk!classes2.dex] (com.mrousavy.camera.frameprocessor.Frame.getIsValid+0)
#13 0x000000000036c700 /apex/com.android.art/lib64/libart.so (art::interpreter::Execute(art::Thread*, art::CodeItemDataAccessor const&, art::ShadowFrame&, art::JValue, bool, bool) (.__uniq.112435418011751916792819755956732575238.llvm.3816585244953842104)+232) (BuildId: a5fcf27f4a71b07dff05c648ad58e3cd)
#14 0x000000000036bff8 /apex/com.android.art/lib64/libart.so (artQuickToInterpreterBridge+964) (BuildId: a5fcf27f4a71b07dff05c648ad58e3cd)
#15 0x000000000034dd68 /apex/com.android.art/lib64/libart.so (art_quick_to_interpreter_bridge+88) (BuildId: a5fcf27f4a71b07dff05c648ad58e3cd)
#16 0x00000000003371a4 /apex/com.android.art/lib64/libart.so (art_quick_invoke_stub+612) (BuildId: a5fcf27f4a71b07dff05c648ad58e3cd)
#17 0x00000000003352cc /apex/com.android.art/lib64/libart.so (art::JValue art::InvokeVirtualOrInterfaceWithVarArgs<art::ArtMethod*>(art::ScopedObjectAccessAlreadyRunnable const&, _jobject*, art::ArtMethod*, std::__va_list)+772) (BuildId: a5fcf27f4a71b07dff05c648ad58e3cd)
#18 0x000000000058dd24 /apex/com.android.art/lib64/libart.so (art::JNI<true>::CallBooleanMethodV(_JNIEnv*, _jobject*, _jmethodID*, std::__va_list)+148) (BuildId: a5fcf27f4a71b07dff05c648ad58e3cd)
#19 0x00000000003f3f10 /apex/com.android.art/lib64/libart.so (art::(anonymous namespace)::CheckJNI::CallMethodV(char const*, _JNIEnv*, _jobject*, _jclass*, _jmethodID*, std::__va_list, art::Primitive::Type, art::InvokeType) (.__uniq.99033978352804627313491551960229047428)+720) (BuildId: a5fcf27f4a71b07dff05c648ad58e3cd)
[... 120 lines cut ...]
  "hasTorch": false,
  "position": "front",
  "id": "1"
}
```

### Device

Pixel 5 / Android 13

### VisionCamera Version

3.3.1 / worklets-core 0.2.2 / rn 0.72.5 / expo 49

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [X] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

xseignard, 2023-10-13:
Ok, after a good night of sleep, I know the problem.

If you look at my frame processor:
```
const measurePupillaryDistanceInPixelsJS = Worklets.createRunInJsFn(measurePupillaryDistanceInPixels);

const frameProcessor = useFrameProcessor((frame: CameraFrame) => {
  "worklet";
  runAtTargetFps(frameProcessorFps, () => {
    "worklet";
    if (!frame.isValid) return;
    const result = scanForEyes(frame); // custom frame processor
    measurePupillaryDistanceInPixelsJS({
      frame,
      face: result?.faces?.[0],
    });
  });
}, []);
```

You see I jump threads with the call to `measurePupillaryDistanceInPixelsJS` which wraps a `measurePupillaryDistanceInPixels`

Looking more closely the `measurePupillaryDistanceInPixels` I have the following:

```
const measurePupillaryDistanceInPixels = useCallback(
  ({ frame, face }: { frame: CameraFrame, face?: Face }) => {
    if (!frame.isValid) return;
    // Android is given dimensions in landscape mode
    const portraitWidth = Math.min(frame.width, frame.height);
    const portraitHeight = Math.max(frame.width, frame.height);

    // blablabla
  },
  []
);
```

Well, don't try to access the `frame` props from the JS thread as the `frame` can be null here. I won't dig into the why 🤷 

Instead stay on the worklet thread and use [`useWorklet`](https://github.com/margelo/react-native-worklets-core/blob/main/docs/USAGE.md) instead of doing you computations on the JS thread like I do 🤦 

Hope this helps

zzz08900, 2023-10-16:
Just hit by the same problem on V2 - sort of different setup than OP's though. (I'm not using Expo)
While trying to access `frame` from within useFrameProcessor using js, the app crashes with `SIGSEGV` on release builds (yeah wired enough debug builds works fine).

So I quickly moved all frame-related code to Java side of frame processor plugin and the crash is gone.

I do feel like it's some kind of "intend usage" (the `frame` is only intended to be passed to native side of frame processor and is not for anything else), mixed with some Android-specific wired behavior.

mrousavy, 2023-10-17:
Yea I mean `frame` should be intended to be either passed to native or access it in JS - it's completely fine to access anything in JS there. Please create a crash report

xseignard, 2023-10-20:
Still happening in release mode 🤷 

Actually I never succeedeed to make `react-native-vision-camera@v3` work on android release builds (with proguard enabled)

mrousavy, 2023-10-20:
From the stacktrace it looks like `frame.isValid` is calling `_frame.getIsValid()` on the native C++ side, and the internal `_frame` is `null`...

mrousavy, 2023-10-20:
Does it work if you disable proguard?

mrousavy, 2023-10-31:
This is really weird, as I never assign `this->frame` to `nullptr` and never pass a `nullptr` `frame` to the ctor.

mrousavy, 2023-10-31:
Only situation where I do this is in the JS destructor: https://github.com/mrousavy/react-native-vision-camera/blob/f4617a3918783ff188c8b63f0694b1ad94ce3fae/package/android/src/main/cpp/frameprocessor/FrameHostObject.cpp#L24-L30

But I doubt that the JS destructor is called when you actually still have a reference in JS 🤔

Vipulpmt, 2023-11-17:
Hello 
I am using  **"react-native-vision-camera": "3.6.4"** and **"react-native-worklets-core": "^0.2.4",**
App crashes after 5 minutes during **frame processing**

rodgomesc, 2023-11-17:
can you wait the same 5 minutes with `hideSS(avg)` commented?

Vipulpmt, 2023-11-20:
Yes @rodgomesc I have check with commenting hideSS(avg), still crash

[image]

rodgomesc, 2023-11-21:
that's weird, o can see the `PreviewView Surface` being destroyed after a `Memory warning` something is eating your memory, i had a similar issue some days ago while i was not releasing memory correctly in my c++ functions, please double checking if you are not somehow blocking vision-camera from release it's resources

Vipulpmt, 2023-11-22:
Thanks, @rodgomesc, but I already checked all the way. I also reset the bundle on the run time every 2 minutes app, it will crash after one hour. If I comment  `frameProcessor={frameProcessor}` from <Camera /> it's working fine

cseluzha, 2023-12-13:
> Does it work if you disable proguard?

Yes, it works
in my case:
 "react-native": "0.72.7",
  "react-native-reanimated": "3.6.1",
  "react-native-vision-camera": "^3.6.16",
  "react-native-worklets-core": "^0.2.4",

mrousavy, 2024-01-15:
Hey! Closing as I think this has been recently fixed. Let me know if you still encounter this issue in 3.7.1

closed 2024-01-15 by mrousavy