margelo/react-native-vision-camera #1993: Be able to import vision camera package when running jest unit tests✨

jesse-one-app, 2023-10-12:
### What feature or enhancement are you suggesting?

When I import the vision camera package to a spec file used for unit testing I am getting the error 

`system/camera-module-not-found: Failed to initialize VisionCamera: The native Camera Module (`NativeModules.CameraView`) could not be found.`

this error is coming from the NativeCameraModule import because jest runners do not have native modules 

I would like to be able to mock out and unit test my code that imports the vision camera library 

### What Platforms whould this feature/enhancement affect?

iOS, Android

### Alternatives/Workarounds

I have tried to yarn patch the repository to not enforce the native module checks in imports but have not had much luck so far

### Additional information

- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar feature requests in this repository](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22✨+enhancement%22) and found none.

mrousavy, 2023-10-14:
There's even a separate "Mocking" page in the docs: https://react-native-vision-camera.com/docs/guides/mocking

D4nt4lion, 2023-11-01:
I am facing the same issue with jest since I updated and mocking it does not help. 
I also do not understand why I need to start my emulator as instructed in the mocking doc?
I want to do jest unit tests, not detox tests.
Or am I missing sth. here?

tatiana-lonestone, 2023-11-02:
Hello,
The Mocking page show how mock Camera on dev test, but not with Jest, no ?
I've got the same error :

````
system/camera-module-not-found: Failed to initialize VisionCamera: The native Camera Module (`NativeModules.CameraView`) could not be found
````

This error block my push because we added some needed git hook with jest test.
@mrousavy : Please consider adding this feature. Thank you !

D4nt4lion, 2023-11-03:
@tatiana-lonestone 

I ended up mocking everything in my jest setup file.

```javascript
import { NativeModules as RNNativeModules } from 'react-native';

...
RNNativeModules.CameraView = {
  getConstants: jest.fn(),
}

jest.mock('react-native-vision-camera', () => {
  return {
    useCameraDevices: jest.fn(() => (deviceMock)),
    useCameraFormat: jest.fn(() => (formatMock)),
    useCodeScanner: jest.fn(() => (codeScannerMock)),
  }});
...
```

Not sure if this is actually a bad idea in the case of vision camera, but the tests ran without error again.

mrousavy, 2023-11-07:
> Hello, The Mocking page show how mock Camera on dev test, but not with Jest, no ? I've got the same error :
>
> ```
> system/camera-module-not-found: Failed to initialize VisionCamera: The native Camera [...]

@tatiana-lonestone if you need this feature consider paying me for it so I can develop it for you.

tatiana-lonestone, 2023-11-08:
Oh, my bad. English is not my native language, I think I didn't express myself well.

I only mean this ticket seems to be a feature request indeed, not a bad reading of documentation.
You indeed provide a mocking page for the Camera. But there is no documentation about the jest mocking.

I'm sorry for this misunderstood. 

Thank you for your work on this project :)

mrousavy, 2023-11-08:
Ah gotcha, then I'll reopen! No worries :)

camboYY, 2026-01-31:
++