margelo/react-native-vision-camera #1995: 🔧 Version 2.16.2 doesn't build with react-native-reanimated 3.5.x because the minSdkVersion is 21

yesnein, 2023-10-13:
### How were you trying to build the app?

Building our app failed because the vision camera uses a minSdkVersion of 21 but react-native-reanimated 3.5.x requires 23.

### Full build logs

```tsx
react-native runAndroid
...
...node_modules/react-native-vision-camera/android/CMakeLists.txt debug|arm64-v8a : User has minSdkVersion 21 but library was built for 23 [//react-native-reanimated/reanimated]
...
```

### Project dependencies

```json
react-native: 0.72.5
react-native-vision-camera: 2.16.2
react-native-reanimated: 3.5.4
```

### VisionCamera Version

2.16.2

### Target platforms

Android

### Operating system

Linux

### Can you build the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

GnApollo, 2023-10-14:
I think the minSdkVersion for this package is 26.

you can change it in
**android/build.gradle**

mrousavy, 2023-10-14:
This might get fixewd with https://github.com/mrousavy/react-native-vision-camera/pull/1955

mrousavy, 2024-01-15:
Yea you can change minSdkVersion to 23 or 26.