margelo/react-native-vision-camera #1998: 🔧 V3.x FrameProcessorPlugin : Can't find the -swift.h file

IvanByRool, 2023-10-14:
### How were you trying to build the app?

Hey,

I'm using XCode to build my IOS app with RN 
Xcode : 14.3
RN : 0.72.5
Vison Camera : 3.3.1

I had to re-write my own FrameProcessor after migration from 2.15 to 3.3.1
But I don't understand how to solve my last problems

What I have
1. An OCRFrameProcessor.switft file

```swift
import MLKit

@objc(OCRFrameProcessorPlugin)
public class OCRFrameProcessorPlugin: FrameProcessorPlugin {

  @objc public static func callback(_ frame: Frame!, withArgs _: [Any]!) -> Any! {

...

  }
}
``` 

2. An OCRFrameProcessor.m file

```objective-c
#import <Foundation/Foundation.h>

#import "OCRFrameProcessor.h"
#if defined __has_include && __has_include("OCRFrameProcessor-Swift.h")
#import "OCRFrameProcessor-Swift.h"
#else
#import <OCRFrameProcessor/OCRFrameProcessor-Swift.h>
#endif

@implementation RegisterPlugins

    + (void) load {
        [FrameProcessorPluginRegistry addFrameProcessorPlugin:@"OCR"
                                              withInitializer:^FrameProcessorPlugin*(NSDictionary* options) {
            return [[OCRFrameProcessor alloc] init];
        }];
    }

@end
``` 
**This is where my problems are**
- I don't understand where and what should be the "-swift.h" file
- I've created a OCRFrameProcessor.h file as following but it doesn't help

```objective-c
#import <VisionCamera/FrameProcessorPlugin.h>
#import <VisionCamera/FrameProcessorPluginRegistry.h>

@interface RegisterPlugins : FrameProcessorPlugin

@end
``` 

I'm quite new with this, so any help would be appreciated, I wanted to update to 3.x for my project in order to use new cool features.

Thanks

### Full build logs

```tsx
Showing All Messages
/src/ios/OCRFrameProcessor/OCRFrameProcessor.m:7:9: 'OCRFrameProcessor/OCRFrameProcessor-Swift.h' file not found
```

### Project dependencies

```json
{
    "react": "18.2.0",
    "react-native": "0.72.5",
    "react-native-reanimated": "^3.5.4",
    "react-native-vision-camera": "^3.3.1",
    "react-native-worklets-core": "^0.2.2"
}
```

### VisionCamera Version

3.3.1

### Target platforms

iOS

### Operating system

MacOS

### Can you build the VisionCamera Example app?

No, I cannot build the Example app either

### Additional information

- [ ] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2023-10-14:
Hey - well the `-Swift` file should be `<YourProjectName>-Swift.h`, for VisionCamera it's `VisionCamera-Swift.h`. For the example app it'd be `VisionCameraExample-Swift.h`.

But you can also register it manually, it doesn't need to be in a `+load` method - it can also be in the app's `AppDelegate` :)

IvanByRool, 2023-10-14:
Thanks for your answer @mrousavy, but I'm clearly out of my confort zone with this ;)

I added a CaRoolTruck-Swift.h file, under OCRFrameProcessor folder.
Probably a lot I don't understand, but what is it for ? No sure what to put inside :/

And I don't understand what you meant by AppDelegate.

I don't want you to waste your time teaching my how it works, so do you have a "working" code I can go through in order to learn by myself ?
Once again, thanks for everything,
Ivan

IvanByRool, 2023-10-14:
To give you more context, 
I wrote my FrameProcessor module for OCR reading on an older version with

OCRFrameProcessor.m
```objective-c
#import <VisionCamera/FrameProcessorPlugin.h>

@interface VISION_EXPORT_SWIFT_FRAME_PROCESSOR(OCR, OCRFrameProcessorPlugin)
@end
``` 

And 
OCRFrameProcessor.swift
```swift
import MLKit

@objc(OCRFrameProcessorPlugin)
public class OCRFrameProcessorPlugin: NSObject, FrameProcessorPluginBase {

  @objc public static func callback(_ frame: Frame!, withArgs _: [Any]!) -> Any! {
...

}

``` 

I noticed changes since 3.x, 
this is where I'm struggling. 

thx

mrousavy, 2023-10-14:
AppDelegate is in your example app or your library user's app. `+load` could be an option but not sure hwo that'd work, I dont have an example. example app here just registers the plugin in AppDElegate

dolakzdenek, 2023-10-16:
@IvanByRool Check if the name of your project is correct (you'll find it in root dir of plugin in podspec file as `s.name`). If it is really `OCRFrameProcessor`, not something like ocr-frame-processor, then you should import that as
```
#import <ocr_frame_processor-Swift.h>
```

@mrousavy I noticed that XCode15 has issues with `+load`
`+initialize` doesn't work for Swift plugin (Plugin is not found), I used code from the Example plugin
I guess AppDelegate will probably work for now but Iam looking for better solution

IvanByRool, 2023-10-16:
Thanks @dolakzdenek, 

I fixed it, you were both right here. The -Swift.h file wasn't properly named.

I know end up with the issue " Frame Processor threw an error: Failed to load Frame Processor Plugin!" when using the plugin.
I'm using +initialize like in the example, so based on your last com, I'm assuming something else should help me solve it ? 

Thanks a lot,
ivan

mrousavy, 2023-10-16:
Cool, so the original issue is solved then.

> Frame Processor threw an error: Failed to load Frame Processor Plugin!

Check your native logs. Does it say "Registering Frame Processor Plugin ...." in there? If not, it is not registering it.

closed 2023-10-16 by mrousavy

IvanByRool, 2023-10-16:
Hey,

no it"s not, I have the following

```typescript
2023-10-16 16:41:27.871404+0200 CaRoolTruck-Val[80831:13683086] [javascript] Loading react-native-worklets-core...
2023-10-16 16:41:27.872749+0200 CaRoolTruck-Val[80831:13683086] [javascript] Worklets loaded successfully
2023-10-16 16:41:27.875624+0200 CaRoolTruck-Val[80831:13683086] VisionCameraProxy: Creating Worklet Context...
2023-10-16 16:41:27.876233+0200 CaRoolTruck-Val[80831:13683086] VisionCameraProxy: Worklet Context Created!
2023-10-16 16:41:28.028830+0200 CaRoolTruck-Val[80831:13682780] [native] VisionCamera.didSetProps(_:): Updating 11 prop(s)...
2023-10-16 16:41:28.029231+0200 CaRoolTruck-Val[80831:13683079] [native] VisionCamera.configureCaptureSession(): Configuring Session...
2023-10-16 16:41:28.029239+0200 CaRoolTruck-Val[80831:13683079] [native] VisionCamera.configureCaptureSession(): Initializing Camera with device com.apple.avfoundation.avcapturedevice.built-in_video:0...
2023-10-16 16:41:28.029245+0200 CaRoolTruck-Val[80831:13683079] [native] VisionCamera.configureCaptureSession(): Adding Video input...
2023-10-16 16:41:28.035549+0200 CaRoolTruck-Val[80831:13683079] [native] VisionCamera.configureCaptureSession(): Adding Video Data output...
2023-10-16 16:41:28.039073+0200 CaRoolTruck-Val[80831:13683079] [native] VisionCamera.invokeOnInitialized(): Camera initialized!
2023-10-16 16:41:28.039124+0200 CaRoolTruck-Val[80831:13683079] [native] VisionCamera.configureCaptureSession(): Session successfully configured!
2023-10-16 16:41:28.039861+0200 CaRoolTruck-Val[80831:13683079] [native] VisionCamera.configureFormat(): Configuring Format...
2023-10-16 16:41:28.039866+0200 CaRoolTruck-Val[80831:13683079] [native] VisionCamera.configureDevice(): Configuring Device...
2023-10-16 16:41:28.040101+0200 CaRoolTruck-Val[80831:13683079] [native] VisionCamera.configureDevice(): Device successfully configured!
2023-10-16 16:41:28.040239+0200 CaRoolTruck-Val[80831:13683079] [native] VisionCamera.didSetProps(_:): Starting Session...
2023-10-16 16:41:28.228937+0200 CaRoolTruck-Val[80831:13683079] [native] VisionCamera.didSetProps(_:): Started Session!
2023-10-16 16:41:28.235759+0200 CaRoolTruck-Val[80831:13683086] [javascript] Frame Processor threw an error: Failed to load Frame Processor Plugin!
2023-10-16 16:41:28.649403+0200 CaRoolTruck-Val[80831:13683086] [javascript] Frame Processor threw an error: Failed to load Frame Processor Plugin!
``` 

Any idea ? thx

mrousavy, 2023-10-16:
Way further up in your logs. This gets logged when the app starts.

IvanByRool, 2023-10-16:
Thanks,

no they're not called.

If it helps, this is the code I have for my FrameProcessor (from the builder)
```
#if __has_include(<VisionCamera/FrameProcessorPlugin.h>)
#import <Foundation/Foundation.h>
#import <VisionCamera/FrameProcessorPlugin.h>
#import <VisionCamera/FrameProcessorPluginRegistry.h>
#import <VisionCamera/Frame.h>

#import "caroolTruck-Swift.h"

@interface XyzFrameProcessorPlugin (FrameProcessorPluginLoader)
@end

@implementation XyzFrameProcessorPlugin (FrameProcessorPluginLoader)

+ (void)initialize
{
  [FrameProcessorPluginRegistry addFrameProcessorPlugin:@"xyz"
                                        withInitializer:^FrameProcessorPlugin*(NSDictionary* options) {
    return [[XyzFrameProcessorPlugin alloc] initWithOptions:options];
  }];
}

@end

#endif

``` 

And,
```
#if VISION_CAMERA_ENABLE_FRAME_PROCESSORS
import VisionCamera

@objc(XyzFrameProcessorPlugin)
public class XyzFrameProcessorPlugin: FrameProcessorPlugin {
  @objc public init(withOptions options: [AnyHashable : Any]) {
    super.init()
  }

  @objc override public func callback(_ frame: Frame, withArguments arguments: [AnyHashable : Any]?) -> Any {
    let buffer = frame.buffer
    let orientation = frame.orientation
    // code goes here
    return "yeah"
  }
}
#endif
```

mrousavy, 2023-10-17:
Well if they're not getting called then they're just not registered.

Can you try to use +load instead of +initialize? That will definitely be called

IvanByRool, 2023-10-17:
It's not logged with +load either.

Don't know if that helps but added logs on frameProcessorPlugins above function and getting plugins = {}

```
+ (NSMutableDictionary<NSString*, PluginInitializerFunction>*)frameProcessorPlugins {
  static NSMutableDictionary<NSString*, PluginInitializerFunction>* plugins = nil;
  if (plugins == nil) {
    plugins = [[NSMutableDictionary alloc] init];
  }
  return plugins; --> this is {}
}
```

mrousavy, 2023-10-17:
@IvanByRool is your `+load` method even called? Put a breakpoint in your plugin's `+load` method.

IvanByRool, 2023-10-17:
No it's not.

I followed this doc : https://react-native-vision-camera.com/docs/guides/frame-processors-plugins-ios
Anything you think may be missing there ? 

I have .m & .swift files for the frame processor
I have the plugin called in my JS
I have [react-native-worklets-core]
babel.config.js looks like this
```
module.exports = {
  presets: ['module:metro-react-native-babel-preset'],
  plugins: [
    ['react-native-reanimated/plugin'],
    ['react-native-worklets-core/plugin'],
  ],
};
```

mrousavy, 2023-10-17:
This has something to do with native iOS development.

I don't know why the `+load` method is not called, maybe your .m file is not properly included in your app's xcode project?

You can also add the plugin in your `AppDelegate`'s start method - that'll always be called.

IvanByRool, 2023-10-19:
Thanks @mrousavy I'll try.
Had to pause this for the moment. 

Thanks a lot for your help

IvanByRool, 2023-11-22:
Hey, 

just to let you know, it's now working fine with latest version

thx