margelo/react-native-vision-camera #2003: v3 camera focus() not working on Android 🐛

rkostrab, 2023-10-14:
### What's happening?

This is similar issue as this https://github.com/mrousavy/react-native-vision-camera/issues/1938 but on Android platform. It seem it just turn off autofocus on Android. In V2 it is working well!

### Reproduceable Code

```tsx
const camera = useRef()

...

<Camera
  ref={camera}
  style={styles.container}
  device={device}
  photo
  zoom={zoom}
  isActive={isScreenFocused && !pictureTaken}
  onInitialized={() => setCameraReady(true)}
  orientation='portrait'
  onError={onError}/>

...

[some button press while camera ready] () => camera.current.focus({ x: 300, y: 300 });
```

### Relevant log output

```shell
[no relevant log]
```

### Camera Device

```json
{
  "formats": undefined,
  "sensorOrientation": "landscape-right",
  "hardwareLevel": "full",
  "maxZoom": 20,
  "minZoom": 0.6704425811767578,
  "supportsLowLightBoost": true,
  "neutralZoom": 1,
  "physicalDevices": [
    "wide-angle-camera"
  ],
  "supportsFocus": true,
  "supportsRawCapture": true,
  "isMultiCam": true,
  "name": "BACK (0)",
  "hasFlash": true,
  "hasTorch": true,
  "position": "back",
  "id": "0"
}
```

### Device

Google Pixel 6 Pro

### VisionCamera Version

3.3.1

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2023-10-16:
There's not enough details here. What do you mean it's not working? What's happening instead?

rkostrab, 2023-10-16:
When I call for example `cameraRef.current.focus({ x: XXX, y: YYY })` then camera does not focus object at point XXX,YYY. Instead of that it looks like it just turns off autofocus. No any error log.

I created a full replicable code for expo project (copy&paste into App.js). Try:

```jsx
import "react-native-gesture-handler";
import { useEffect, useRef } from "react";
import { Text, ToastAndroid } from "react-native";
import {
  Camera,
  useCameraDevice,
  useCameraPermission,
} from "react-native-vision-camera";
import {
  TapGestureHandler,
  GestureHandlerRootView,
  State,
} from "react-native-gesture-handler";

export default function App() {
  const camera = useRef();
  const device = useCameraDevice("back");
  const { hasPermission, requestPermission } = useCameraPermission();

  useEffect(() => {
    if (!hasPermission) {
      requestPermission();
    }
  }, [hasPermission]);

  const onFocus = (event) => {
    if (event.nativeEvent.state !== State.ACTIVE) return;
    const point = {
      x: Math.round(event.nativeEvent.x),
      y: Math.round(event.nativeEvent.y),
    };
    try {
      ToastAndroid.show(
        "Focusing (support=" +
          device.supportsFocus +
          ") on point: " +
          JSON.stringify(point),
        1000
      );
      camera.current.focus(point);
    } catch (error) {
      console.error(error.message);
    }
  };

  if (!hasPermission)
    return <Text style={{ flex: 1 }}>{"No persmission"}</Text>;

  return (
    <GestureHandlerRootView style={{ flex: 1 }}>
      <TapGestureHandler onHandlerStateChange={onFocus}>
        <Camera
          ref={camera}
          style={{ flex: 1 }}
          device={device}
          torch={"off"}
          orientation="portrait"
          isActive={hasPermission}
          onError={(error) => console.error(error.message)}
        />
      </TapGestureHandler>
    </GestureHandlerRootView>
  );
}
```

nBanousi, 2023-10-31:
any update on this?

mrousavy, 2023-10-31:
Hey, nope. I did not have any time to investigate this yet as I have many other priorities. If anyone wants this fixed, consider [sponsoring me](https://github.com/sponsors/mrousavy) / paying me my consultancy fee so I can dedicate time to investigate this and fix the bug :)

DM me for details

rsnr, 2023-11-05:
> Hey, nope. I did not have any time to investigate this yet as I have many other priorities. If anyone wants this fixed, consider [sponsoring me](https://github.com/sponsors/mrousavy) / paying me my consultancy fee so I can dedicate time to investigate this and fix the bug :)
> 
> DM me for details

Hi @mrousavy I sponsored a very similar issue on iOS, I wonder if both are due to the same root cause? #2120 Thanks!

mrousavy, 2023-11-05:
@rsnr Hey - might be! I was on vacation last week and today is Sunday, I'll be back on Tuesday! :)

EuSet, 2023-11-23:
is there some new information about this issue?

mrousavy, 2023-11-23:
Not yet, it's on my priority list though! I just fixed a critical synchronization bug here: https://github.com/mrousavy/react-native-vision-camera/pull/2206 and will focus on `focus()` (pun intended) soon!

mrousavy, 2023-11-23:
Thanks @rsnr for sponsoring this btw 🙏

Brianwebb22, 2023-11-29:
I'm currently receiving the following error when I call `camera.current.focus()` on Android. All other camera functions (zoom, flash, takePhoto(), etc.) are working as expected on the same screen. I'm seeing the tap gesture being received and handled properly so I don't think it's related to that. 

Captured from Logcat:
```
com.mrousavy.camera.core.CameraNotReadyError: [session/camera-not-ready] The Camera is not ready yet! Wait for the onInitialized() callback!
at com.mrousavy.camera.core.CameraSession.focus(CameraSession.kt:579)
at com.mrousavy.camera.core.CameraSession.focus(CameraSession.kt:572)
at com.mrousavy.camera.CameraView_FocusKt.focus(CameraView+Focus.kt:8)
at com.mrousavy.camera.CameraViewModule$focus$1.invokeSuspend(CameraViewModule.kt:141)
```

I'd take the error more at face value but `focus()` call only fires if `onInitialized` has already occurred.

mrousavy, 2024-01-15:
Working on focus this week 💪

MilanRadon, 2024-01-22:
@mrousavy how does it look with the focus correction?

mrousavy, 2024-01-23:
Didn't finish it yet. had some other issues to look into first and released vision-camera-resize-plugin, which had higher priority for me.

MilanRadon, 2024-01-23:
@mrousavy thanks for reply. When y mean will be done?

mrousavy, 2024-01-24:
No idea. I don't have it planned for now as it's not a paid feature, we'll see when I have some free time. :)

closed 2024-02-08 by mrousavy

mrousavy, 2024-02-08:
I finally implemented focus now in #2523!

VisionCamera 3.9.0-beta.3 now includes this feature! :)