margelo/react-native-vision-camera #2005: 🐛 Could not find com.google.android.gms:play-services-mlkit-barcode-scanning:18.3.0

LiFuHai1106, 2023-10-16:
### What's happening?

Could not determine the dependencies of task ':react-native-vision-camera:compileDebugAidl'.
> Could not resolve all task dependencies for configuration ':react-native-vision-camera:debugCompileClasspath'.
   > Could not find com.google.android.gms:play-services-mlkit-barcode-scanning:18.3.0.
     Required by:
         project :react-native-vision-camera

### Reproduceable Code

```tsx
<Camera
        style={StyleSheet.absoluteFill}
        device={device}
        isActive={true}
        video={true}
        audio={true}
        ref={camera}
        photo={true}
      />
```

### Relevant log output

```shell
Could not determine the dependencies of task ':react-native-vision-camera:compileDebugAidl'.
> Could not resolve all task dependencies for configuration ':react-native-vision-camera:debugCompileClasspath'.
   > Could not find com.google.android.gms:play-services-mlkit-barcode-scanning:18.3.0.
     Required by:
         project :react-native-vision-camera
```

### Camera Device

```json
1
```

### Device

android

### VisionCamera Version

3.3.1

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

LiFuHai1106, 2023-10-16:
"react": "18.2.0",
    "react-native": "0.72.6",
    "react-native-vision-camera": "3.3.1",
    "react-native-worklets-core": "^0.2.2",

mrousavy, 2023-10-16:
Also you created a bug report instead of a build error. I really don't get why people constantly create the wrong issue templates, is that not clear what to choose on this page?
[image]

LiFuHai1106, 2023-10-17:
> It does work in the example app, and this is what Google recommends. Check this guide: https://developers.google.com/ml-kit/vision/barcode-scanning/android
> 
> This is likely not a fault in VisionCamera.
> 
> Can you try to replace
> 
> ```
> implementation 'com.google.android.gms:play-services-mlkit-barcode-scanning:18.3.0'
> ```
> 
> with
> 
> ```
> implementation 'com.google.mlkit:barcode-scanning:17.2.0'
> ```
> 
> then do a Gradle Sync and then rebuild your app to see if that works? That line is in `node_modules/react-native-vision-camera/android/build.gradle`.

> It does work in the example app, and this is what Google recommends. Check this guide: https://developers.google.com/ml-kit/vision/barcode-scanning/android
> 
> This is likely not a fault in VisionCamera.
> 
> Can you try to replace
> 
> ```
> implementation 'com.google.android.gms:play-services-mlkit-barcode-scanning:18.3.0'
> ```
> 
> with
> 
> ```
> implementation 'com.google.mlkit:barcode-scanning:17.2.0'
> ```
> 
> then do a Gradle Sync and then rebuild your app to see if that works? That line is in `node_modules/react-native-vision-camera/android/build.gradle`.

Could not determine the dependencies of task ':app:compileDebugJavaWithJavac'.
> Could not resolve all task dependencies for configuration ':app:debugCompileClasspath'.
   > Could not find com.google.mlkit:barcode-scanning:17.2.0

LiFuHai1106, 2023-10-17:
Could not determine the dependencies of task ':app:compileDebugJavaWithJavac'.
> Could not resolve all task dependencies for configuration ':app:debugCompileClasspath'.
   > Could not find com.google.mlkit:barcode-scanning:17.2.0

closed 2023-10-17 by mrousavy