margelo/react-native-vision-camera #2007: 🐛 ERROR  frame-processor/unavailable

creativekdev, 2023-10-16:
### What's happening?

ERROR  frame-processor/unavailable
I can not use frame processor on 
  "dependencies": {
    "react": "18.2.0",
    "react-native": "0.72.3",
    "react-native-biometrics": "^3.0.1",
    "react-native-fs": "^2.20.0",
    "react-native-image-picker": "^7.0.1",
    "react-native-reanimated": "^3.5.4",
    "react-native-touch-id": "^4.4.1",
    "react-native-vector-icons": "^10.0.0",
    "react-native-vision-camera": "^3.3.1",
    "vision-camera-face-detector": "^0.1.8"
  },

### Reproduceable Code

```tsx
const frameProcessor = useFrameProcessor(frame => {
    'worklet';
    const scannedFaces = scanFaces(frame);
    console.log(scannedFaces);
    runOnJS(setFaces)(scannedFaces);
  }, []);

    <Camera
      style={StyleSheet.absoluteFill}
      device={device}
      isActive={true}
      frameProcessor={frameProcessor}
      frameProcessorFps={5}
    />
```

### Relevant log output

```shell
> Task :vision-camera-face-detector:compileDebugJavaWithJavac FAILED
```

### Camera Device

```json
const optionalConfigObject = {
    title: 'Please Authenticate', // Android
    imageColor: '#000', // Android
    imageErrorColor: '#ff0000', // Android
    sensorDescription: 'Slightly Touch sensor', // Android
    sensorErrorDescription: 'Failed', // Android
    cancelText: 'Cancel', // Android
    fallbackLabel: 'Show Passcode', // iOS (if empty, then label is hidden)
    unifiedErrors: false, // use unified error messages (default false)
    passcodeFallback: false, // iOS
  };
```

### Device

Android Emulator

### VisionCamera Version

3.3.1

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

maslovay, 2023-10-16:
Same issue

mrousavy, 2023-10-16:
> I think the current vision camera is not compatible with the vision-camera-face-detector

The Face Detector library didn't upgrade to V3 yet. 

> I need to use face recognition module on react native, seems react-native-vision-camera not support face recognition.

It would, if you write a native plugin. Just a few lines of native code.

Also, you didn't install Worklets Core. Read the docs.

closed 2023-10-16 by mrousavy

ombogdanJoinToIt, 2023-10-19:
frame-processor/unavailable:
  const frameProcessor = useFrameProcessor((frame) => {
    'worklet';
    const detectedBarcodes = scanBarcodes(frame, [BarcodeFormat.QR_CODE], { checkInverted: true });
    runOnJS(setBarcodes)(detectedBarcodes);
  }, []);