margelo/react-native-vision-camera #2008: 🐛 AVCaptureDevice.Position has unknown state

Nilotpal35, 2023-10-16:
### What's happening?

**Iphone 15 pro and pro max Camera component from React-native-vision-camera is getting break.** 

**Error showing :** 
[image]
 
**package.json file looks like this :** 
[image]

**But after doing some head around i have tired to change the react-native-vision-camera version to 2.16,2 and react-native-reanimated to 2.10.0** 
[image]

**but this time some new error is coming that looks like this in my simulator :** 
[image]

**and crashing on all the iphone devices.**

FYI : _in my code react-native-worklets-core is not added_
i am not sure as is this breaking the code or something else

---------------**_Happy to get suggestions from developers around the world_**-------------------------

### Reproduceable Code

```tsx
import { runOnJS } from 'react-native-reanimated';
import { Camera, useCameraDevices, useFrameProcessor } from 'react-native-vision-camera';
import { BarcodeFormat, scanBarcodes } from 'vision-camera-code-scanner';

<Camera
    style={StyleSheet.absoluteFill}
    device={device}
    isActive={!isScanning}
    frameProcessor={frameProcessor}
    frameProcessorFps={1.0}
    zoom={device.neutralZoom}
    onLayout={(event) =>
     {
       var {x, y, width, height} = event.nativeEvent.layout;
       if (cameraDimensions &&
       (cameraDimensions.width != width ||
        cameraDimensions.height != height))
        {
          const barcodeAspectRatio = height > width ? 0.5: 0.35
          const widthPadding = height > width ? 30: 120
          const barcodeHeight = (width - widthPadding * 2) * barcodeAspectRatio;
          setCameraDimensions({
                 width: width,
                 height: height,
                 widthPadding: widthPadding,
                 heightPadding: (height - barcodeHeight) * 0.5
                 });
               }
             }
            }
  >
     {isActivityIndicatorVisible ? (
        <View style={styles.indicatorView}>
          <ActivityIndicator size="large" />
        </View>
      ) : (
        <View style={styles.innerCameraView(cameraDimensions)} />
         )}
   </Camera>
```

### Relevant log output

```shell
ERROR:frame-processor/unavailable: Frame Processors are not enabled. See https://react-native-vision-camera.com/docs/guides/troubleshooting\n\nThis error is located at:\n    in NavigationContainer (created by SceneView)\n    in SceneView (created by DrawerView)\n    in RCTView (created by View)\n    in View (created by AnimatedComponent)\n    in AnimatedComponent\n    in AnimatedComponentWrapper (created by DrawerLayout)\n    in RCTView (created by View)\n    in View (created by DrawerLayout)\n    in DrawerLayout (created by DrawerView)\n    in DrawerView (created by Navigator)\n    in Navigator (created by NavigationContainer)\n    in NavigationContainer (created by screen)\n    in screen (created by SceneView)\n    in SceneView (created by DrawerView)\n    in RCTView (created by View)\n    in View (created by AnimatedComponent)\n    in AnimatedComponent\n    in AnimatedComponentWrapper (created by DrawerLayout)\n    in RCTView (created by View)\n    in View (created by DrawerLayout)\n    in DrawerLayout (created by DrawerView)\n    in DrawerView (created by Navigator)\n    in Navigator (created by NavigationContainer)\n    in NavigationContainer (created by Main)\n    in Main (created by screen)\n    in Provider (created by screen)\n    in RNCSafeAreaProvider (created by SafeAreaProvider)\n    in SafeAreaProvider (created by screen)\n    in screen (created by SceneView)\n    in SceneView (created by SwitchView)\n    in SwitchView (created by Navigator)\n    in Navigator (created by NavigationContainer)\n    in NavigationContainer (created by App)\n    in App (created by ExpoRoot)\n    in ExpoRoot\n    in RCTView (created by View)\n    in View (created by AppContainer)\n    in RCTView (created by View)\n    in View (created by AppContainer)\n    in AppContainer\n    in main(RootComponent)', undefined
frame-processor/unavailable: Frame Processors are not enabled. See https://react-native-vision-camera.com/docs/guides/troubleshooting
 
This error is located at:
    in NavigationContainer (created by SceneView)
    in SceneView (created by DrawerView)
    in RCTView (created by View)
    in View (created by AnimatedComponent)
    in AnimatedComponent
    in AnimatedComponentWrapper (created by DrawerLayout)
    in RCTView (created by View)
    in View (created by DrawerLayout)
    in DrawerLayout (created by DrawerView)
    in DrawerView (created by Navigator)
    in Navigator (created by NavigationContainer)
    in NavigationContainer (created by screen)
    in screen (created by SceneView)
    in SceneView (created by DrawerView)
    in RCTView (created by View)
    in View (created by AnimatedComponent)
    in AnimatedComponent
    in AnimatedComponentWrapper (created by DrawerLayout)
    in RCTView (created by View)
    in View (created by DrawerLayout)
    in DrawerLayout (created by DrawerView)
    in DrawerView (created by Navigator)
    in Navigator (created by NavigationContainer)
    in NavigationContainer (created by Main)
    in Main (created by screen)
    in Provider (created by screen)
    in RNCSafeAreaProvider (created by SafeAreaProvider)
    in SafeAreaProvider (created by screen)
    in screen (created by SceneView)
    in SceneView (created by SwitchView)
    in SwitchView (created by Navigator)
    in Navigator (created by NavigationContainer)
    in NavigationContainer (created by App)
    in App (created by ExpoRoot)
    in ExpoRoot
    in RCTView (created by View)
    in View (created by AppContainer)
    in RCTView (created by View)
    in View (created by AppContainer)
    in AppContainer
    in main(RootComponent)
```

### Camera Device

```json
Unknown as I have don't have the new iPhone and this is a crash report
```

### Device

iPhone 15 pro  & Pro Max

### VisionCamera Version

2.16.2

### Can you reproduce this issue in the VisionCamera Example app?

No, I cannot reproduce the issue in the Example app

### Additional information

- [X] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

DarkMatter513, 2023-10-16:
did this work for you without worklet cores?

mrousavy, 2023-10-16:
The "AVCaptureDevice.Position has unknown state" error has already been fixed in latest VisionCamera (both V3 and V2).

closed 2023-10-16 by mrousavy

Nilotpal35, 2023-10-16:
Is there anything else like "react-native-worklets-core" or some other library i need to make changes;

Right now i am using runOnJS  IIFE to implement this :
const frameProcessor = useFrameProcessor((frame) => {
    'worklet';
    const detectedBarcodes = scanBarcodes(
      frame,
      [BarcodeFormat.CODE_128, BarcodeFormat.EAN_13, BarcodeFormat.EAN_8, BarcodeFormat.CODE_39, BarcodeFormat.DATA_MATRIX],
      { checkInverted: true }
    );
    if (detectedBarcodes && detectedBarcodes.length > 0 && !isScanning && timerId === 0) {
      detectedBarcodes.forEach( nextBarcode =>
        {
          const bufferX = cameraDimensions.width > cameraDimensions.height ? 0 : 150;
          const bufferY = cameraDimensions.width > cameraDimensions.height ? 150 : 0;
          const barcodePoint1 = { x: nextBarcode.cornerPoints[0].x / frame.width, y: nextBarcode.cornerPoints[0].y / frame.height };
          const barcodePoint2 = { x: nextBarcode.cornerPoints[1].x / frame.width, y: nextBarcode.cornerPoints[1].y / frame.height };
          const barcodePoint3 = { x: nextBarcode.cornerPoints[2].x / frame.width, y: nextBarcode.cornerPoints[2].y / frame.height };
          const barcodePoint4 = { x: nextBarcode.cornerPoints[3].x / frame.width, y: nextBarcode.cornerPoints[3].y / frame.height };
 
          const scannerMinX = cameraDimensions.widthPadding / (cameraDimensions.width - bufferX);
          const scannerMaxX = 1.0 - (cameraDimensions.widthPadding / (cameraDimensions.width - bufferX));
          const scannerMinY = cameraDimensions.heightPadding / (cameraDimensions.height - bufferY);
          const scannerMaxY = 1.0 - (cameraDimensions.heightPadding / (cameraDimensions.height - bufferY));
 
          if (scannerMinX < barcodePoint1.x &&
              scannerMinX < barcodePoint2.x &&
              scannerMinX < barcodePoint3.x &&
              scannerMinX < barcodePoint4.x &&
              scannerMaxX > barcodePoint1.x &&
              scannerMaxX > barcodePoint2.x &&
              scannerMaxX > barcodePoint3.x &&
              scannerMaxX > barcodePoint4.x &&
              scannerMinY < barcodePoint1.y &&
              scannerMinY < barcodePoint2.y &&
              scannerMinY < barcodePoint3.y &&
              scannerMinY < barcodePoint4.y &&
              scannerMaxY > barcodePoint1.y &&
              scannerMaxY > barcodePoint2.y &&
              scannerMaxY > barcodePoint3.y &&
              scannerMaxY > barcodePoint4.y
            )
          {
            console.log(`BARCODE SCAN IN THE LANE.`);
            runOnJS(notifyBarcodeScan)({
              data: nextBarcode.displayValue,
              format: nextBarcode.format
            });
            return;
          }
        }
      );
     
    }`
`

thongquach, 2023-10-25:
I encountered `frame-processor/unavailable: Frame Processors are not enabled` on both Android and iOS in latest v2 version `2.16.5`.
[image]