margelo/react-native-vision-camera #2011: ✨ Expose ISO/Exposure/White-Balance/Focus controls

mrousavy, 2023-10-16:
### What feature or enhancement are you suggesting?

VisionCamera currently exposes a lot of features to the JS side. For example, `zoom` can be used to smoothly zoom in/out using the hardware Camera APIs by binding it to a Reanimated Shared Value.

Additionally, for advanced apps it would be a pretty dope feature to also expose **ISO**, **Exposure**, **White-Balance** and **Focus** as controllable props directly.

For example:

```tsx
<Camera
  iso={5}
  exposure={7}
  whiteBalance={9}
  focus={0.3}
    />
```

The following features need to be added:

- **ISO** prop
  - Expose ISO in format to get possible range/values (Already done; `minISO` and `maxISO`)
  - Add prop to `<Camera>`, default value is `"auto"`
  - If set to a number, override ISO on the native side (in "side-props")
  - If set to null, return to auto ISO again
  - Range-check possible values and throw errors if trying to set an unsupported ISO value
- **Exposure** prop
  - Expose Exposure in format to get possible range/values
  - Add prop to `<Camera>`, default value is `"auto"`
  - If set to a number, override exposure (CONTROL_AE) on the native side (in "side-props")
  - If set to null, return to auto exposure again
  - Range-check possible values and throw errors if trying to set an unsupported value
- **White-Balance** prop
  - Expose White-Balance in format to get possible range/values
  - Add prop to `<Camera>`, default value is `"auto"`
  - If set to a number, override White-Balance (CONTROL_AWB) on the native side (in "side-props")
  - If set to null, return to auto White-Balance again
  - Range-check possible values and throw errors if trying to set an unsupported value
- **Focus** prop
  - Expose Focus in format to get possible range/values
  - Add prop to `<Camera>`, default value is `"auto"`
  - If set to a number, override Focus (CONTROL_AF) on the native side (in "side-props")
  - If set to null, return to auto Focus again
  - Range-check possible values and throw errors if trying to set an unsupported value

### What Platforms whould this feature/enhancement affect?

iOS, Android

### Alternatives/Workarounds

None

### Additional information

- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar feature requests in this repository](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22✨+enhancement%22) and found none.

yiday, 2023-10-20:
I really need this feature, thanks

mrousavy, 2023-10-20:
cool

mrousavy, 2023-11-18:
I just implemented `exposure` in https://github.com/mrousavy/react-native-vision-camera/pull/2173! 🥳 

The main difference here is that `exposure` is a bias, and will auto-control both exposure-duration and ISO.
So I don't think that we need manual control for ISO.

I also don't think that we need controls for white-balance, that should be left on auto imo.

closed 2023-11-19 by mrousavy

meTech1021, 2025-12-10:
@mrousavy 
I have a question.
Right now, I’d like to capture the current exposure value at a specific moment and then lock it for a while.
I checked the available functions, but it seems there’s no built-in API in React Native Vision Camera to read the current exposure level.
Could you please let me know if this API is planned, and whether there’s any alternative solution we can use in the meantime?
Thank you.