margelo/react-native-vision-camera #2013: 🔧 undefined symbol: RNWorklet::JsiWorkletContext::getWorkletRuntime()

Danushka96, 2023-10-16:
### How were you trying to build the app?

I updated Vision Camera 3.4.0 due to the fps dropping issue in Android in the previous version. Previously it was built without issue. But after update I got this build issue

Works with `VisionCamera_disableFrameProcessors = true` in gradle-wrapper.properties

Can I disable it if I need to use CodeScanner feature?

### Full build logs

```tsx
error: undefined symbol: RNWorklet::JsiWorkletContext::getWorkletRuntime()
  >>> referenced by JFrameProcessor.cpp:37 (/Users/danushkaherath/BeepMerchant/node_modules/react-native-vision-camera/android/src/main/cpp/frameprocessor/java-bindings/JFrameProcessor.cpp:37)
  >>>               CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessor/java-bindings/JFrameProcessor.cpp.o:(vision::JFrameProcessor::callWithFrameHostObject(std::__ndk1::shared_ptr<vision::FrameHostObject> const&) const)
  >>> referenced by JVisionCameraProxy.cpp:57 (/Users/danushkaherath/BeepMerchant/node_modules/react-native-vision-camera/android/src/main/cpp/frameprocessor/java-bindings/JVisionCameraProxy.cpp:57)
  >>>               CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessor/java-bindings/JVisionCameraProxy.cpp.o:(vision::JVisionCameraProxy::~JVisionCameraProxy())
  
  ld: error: undefined symbol: RNWorklet::JsiWorkletContext::invokeOnJsThread(std::__ndk1::function<void (facebook::jsi::Runtime&)>&&)
  >>> referenced by JFrameProcessor.cpp:50 (/Users/danushkaherath/BeepMerchant/node_modules/react-native-vision-camera/android/src/main/cpp/frameprocessor/java-bindings/JFrameProcessor.cpp:50)
  >>>               CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessor/java-bindings/JFrameProcessor.cpp.o:(vision::JFrameProcessor::callWithFrameHostObject(std::__ndk1::shared_ptr<vision::FrameHostObject> const&) const)
  >>> referenced by WKTJsiWorklet.h:368 (/Users/danushkaherath/BeepMerchant/node_modules/react-native-worklets-core/android/build/headers/rnworklets/react-native-worklets-core/WKTJsiWorklet.h:368)
  >>>               CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessor/java-bindings/JFrameProcessor.cpp.o:(RNWorklet::WorkletInvoker::~WorkletInvoker())
  
  ld: error: undefined symbol: RNWorklet::JsiWorkletContext::runtimeMappings
  >>> referenced by __tree:0 (/Users/danushkaherath/Library/Android/sdk/ndk/23.1.7779620/toolchains/llvm/prebuilt/darwin-x86_64/sysroot/usr/include/c++/v1/__tree:0)
  >>>               CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessor/java-bindings/JFrameProcessor.cpp.o:(RNWorklet::WorkletInvoker::call(facebook::jsi::Runtime&, facebook::jsi::Value const&, facebook::jsi::Value const*, unsigned long))
  >>> referenced by __tree:0 (/Users/danushkaherath/Library/Android/sdk/ndk/23.1.7779620/toolchains/llvm/prebuilt/darwin-x86_64/sysroot/usr/include/c++/v1/__tree:0)
  >>>               CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessor/java-bindings/JFrameProcessor.cpp.o:(RNWorklet::WorkletInvoker::call(facebook::jsi::Runtime&, facebook::jsi::Value const&, facebook::jsi::Value const*, unsigned long))
  
  ld: error: undefined symbol: RNWorklet::JsiWorkletContext::invokeOnWorkletThread(std::__ndk1::function<void (RNWorklet::JsiWorkletContext*, facebook::jsi::Runtime&)>&&)
  >>> referenced by WKTJsiWorklet.h:371 (/Users/danushkaherath/BeepMerchant/node_modules/react-native-worklets-core/android/build/headers/rnworklets/react-native-worklets-core/WKTJsiWorklet.h:371)
  >>>               CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessor/java-bindings/JFrameProcessor.cpp.o:(RNWorklet::WorkletInvoker::~WorkletInvoker())
  
  ld: error: undefined symbol: RNWorklet::JsiWorkletContext::defaultInstance
  >>> referenced by memory:3887 (/Users/danushkaherath/Library/Android/sdk/ndk/23.1.7779620/toolchains/llvm/prebuilt/darwin-x86_64/sysroot/usr/include/c++/v1/memory:3887)
  >>>               CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessor/java-bindings/JFrameProcessor.cpp.o:(RNWorklet::JsiWorkletContext::getDefaultInstance())
  >>> referenced by memory:3887 (/Users/danushkaherath/Library/Android/sdk/ndk/23.1.7779620/toolchains/llvm/prebuilt/darwin-x86_64/sysroot/usr/include/c++/v1/memory:3887)
  >>>               CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessor/java-bindings/JFrameProcessor.cpp.o:(RNWorklet::JsiWorkletContext::getDefaultInstance())
  
  ld: error: undefined symbol: RNWorklet::JsiHostObject::JsiHostObject()
  >>> referenced by WKTJsiWorkletContext.h:31 (/Users/danushkaherath/BeepMerchant/node_modules/react-native-worklets-core/android/build/headers/rnworklets/react-native-worklets-core/WKTJsiWorkletContext.h:31)
  >>>               CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessor/java-bindings/JFrameProcessor.cpp.o:(RNWorklet::JsiWorkletContext::getDefaultInstance())
  >>> referenced by WKTJsiWorklet.h:77 (/Users/danushkaherath/BeepMerchant/node_modules/react-native-worklets-core/android/build/headers/rnworklets/react-native-worklets-core/WKTJsiWorklet.h:77)
  >>>               CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessor/java-bindings/JVisionCameraProxy.cpp.o:(RNWorklet::JsiWorklet::JsiWorklet(facebook::jsi::Runtime&, facebook::jsi::Value const&))
  
  ld: error: undefined symbol: vtable for RNWorklet::JsiWorkletContext
  >>> referenced by WKTJsiWorkletContext.h:31 (/Users/danushkaherath/BeepMerchant/node_modules/react-native-worklets-core/android/build/headers/rnworklets/react-native-worklets-core/WKTJsiWorkletContext.h:31)
  >>>               CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessor/java-bindings/JFrameProcessor.cpp.o:(RNWorklet::JsiWorkletContext::getDefaultInstance())
  >>> referenced by WKTJsiWorkletContext.h:31 (/Users/danushkaherath/BeepMerchant/node_modules/react-native-worklets-core/android/build/headers/rnworklets/react-native-worklets-core/WKTJsiWorkletContext.h:31)
  >>>               CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessor/java-bindings/JFrameProcessor.cpp.o:(RNWorklet::JsiWorkletContext::getDefaultInstance())
  >>> the vtable symbol may be undefined because the class is missing its key function (see https://lld.llvm.org/missingkeyfunction)
  
  ld: error: undefined symbol: RNWorklet::JsiWorkletContext::~JsiWorkletContext()
  >>> referenced by memory:2252 (/Users/danushkaherath/Library/Android/sdk/ndk/23.1.7779620/toolchains/llvm/prebuilt/darwin-x86_64/sysroot/usr/include/c++/v1/memory:2252)
  >>>               CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessor/java-bindings/JFrameProcessor.cpp.o:(std::__ndk1::__shared_ptr_emplace<RNWorklet::JsiWorkletContext, std::__ndk1::allocator<RNWorklet::JsiWorkletContext> >::~__shared_ptr_emplace())
  >>> referenced by memory:2252 (/Users/danushkaherath/Library/Android/sdk/ndk/23.1.7779620/toolchains/llvm/prebuilt/darwin-x86_64/sysroot/usr/include/c++/v1/memory:2252)
  >>>               CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessor/java-bindings/JFrameProcessor.cpp.o:(std::__ndk1::__shared_ptr_emplace<RNWorklet::JsiWorkletContext, std::__ndk1::allocator<RNWorklet::JsiWorkletContext> >::~__shared_ptr_emplace())
  
  ld: error: undefined symbol: RNWorklet::JsiWorkletContext::JsiWorkletContext(std::__ndk1::basic_string<char, std::__ndk1::char_traits<char>, std::__ndk1::allocator<char> > const&, facebook::jsi::Runtime*, std::__ndk1::function<void (std::__ndk1::function<void ()>&&)>, std::__ndk1::function<void (std::__ndk1::function<void ()>&&)>)
  >>> referenced by memory:2278 (/Users/danushkaherath/Library/Android/sdk/ndk/23.1.7779620/toolchains/llvm/prebuilt/darwin-x86_64/sysroot/usr/include/c++/v1/memory:2278)
  >>>               CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessor/java-bindings/JVisionCameraProxy.cpp.o:(vision::JVisionCameraProxy::JVisionCameraProxy(facebook::jni::alias_ref<facebook::jni::detail::JTypeFor<facebook::jni::HybridClass<vision::JVisionCameraProxy, facebook::jni::detail::BaseHybridClass>::JavaPart, facebook::jni::JObject, void>::_javaobject*> const&, facebook::jsi::Runtime*, std::__ndk1::shared_ptr<facebook::react::CallInvoker> const&, facebook::jni::basic_strong_ref<facebook::jni::detail::JTypeFor<facebook::jni::HybridClass<vision::JVisionCameraScheduler, facebook::jni::detail::BaseHybridClass>::JavaPart, facebook::jni::JObject, void>::_javaobject*, facebook::jni::GlobalReferenceAllocator> const&))
  
  ld: error: undefined symbol: RNWorklet::JsiHostObject::~JsiHostObject()
[... cut ...]
### Additional information

- [ ] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

Fabianurrutia, 2023-10-16:
check this: https://github.com/mrousavy/react-native-vision-camera/issues/1770#issuecomment-1707550362 maybe can help

mrousavy, 2023-10-17:
> Can I disable it if I need to use CodeScanner feature?

Yes, FPs can be disabled and the CodeScanner will still work.

If you dont install react-native-worklets-core, it should work right?

Danushka96, 2023-10-18:
Yes, Working without issue when removing react-native-worklets-core from the project.

ldstein, 2023-11-16:
Hi, just updated to v3.6.6 and noticed in the changelog that this issue should be fixed by 
https://github.com/mrousavy/react-native-vision-camera/commit/75fd9248996caeccccec65da8f6c950c8d440e47

After a a clean install of my project, I get the `undefined symbol: RNWorklet::JsiWorkletContext::getWorkletRuntime()` error when building via `npm run react-native run-android`.

However, building via Android Studio (Giraffe | 2022.3.1 Patch 2) succeeds. Then subsequent builds via `npm run react-native run-android` will also succeed. 

`gradlew clean` will make the build error return and another single build via Android Studio fixes it.

I went back through my project's history use of RNVC v3 and this issue has always been there. I never noticed it before as I typically kick off builds via Android Studio.

I have no idea why Android Studio builds would behave differently to `npm run react-native run-android`.

Zoxive, 2023-12-09:
Anytime i change other expo libraries dependancies that require a new dev-client, the android compile fails for the above error (`undefined symbol: RNWorklet::JsiWorkletContext::getWorkletRuntime() error when building via npm run react-native run-android`) . I have to delete the /android/.gradle folder (as noted by `@ldstein`). I havent had to do this for any other library ive played with thus far.

Either im doing something wrong (only been playing with react-native for a few months so, very possible), or there still is something fishy going on.

mrousavy, 2023-12-12:
Maybe some caching issue. I still don't know why this happens, I am focusing on higher priority issues at the moment. But if someone fixes this, I'll happily accept PRs here! ❤️

mrousavy, 2024-01-15:
It builds fine for me, both in VisionCamera and in a new RN app.

closed 2024-01-15 by mrousavy

DevMinhThu, 2024-03-05:
Hello, I cleared the cache according to the instructions but the error still occurs, please help me. Thankyou very much. @mrousavy

Voxar, 2024-09-19:
Hi there. I got this error after running a ./gradlew clean, if that helps anyone. 

"react-native-vision-camera": "^4.4.1"

jedt, 2024-11-04:
If deleting the `./android/.gradle` did not solve the issue, try deleting the `./node_modules` as well and then install the packages again with `npm install`

sumitsyngenta, 2024-12-16:
Solution for RN 0.76:
 1. Upgrade react-native-vision-camera
 2. delete android/.gradle 
 3. delete android/build
 5. Run the app and enjoy coding.

tremblerz, 2025-01-02:
I upgraded to RN 0.76 and tried various combinations of cache removal but what worked in the end is as follows -
1. `rm -rf node_modules/ android/.gradle android/build`
2. `yarn install`
3. Restart Android Studio
4. Build for Android

babldev, 2025-01-24:
I'm hitting this today as well. Trying to pinpoint the cause but workaround was:

```sh
rm -rf android/.gradle/ node_modules/react-native-worklets-core/android/build/ node_modules/react-native-vision-camera/android/build/ node_modules/
npm install
npm run android
```

bgyhnddr, 2025-05-08:
> I upgraded to RN 0.76 and tried various combinations of cache removal but what worked in the end is as follows -
> 
> 1. `rm -rf node_modules/ android/.gradle android/build`
> 2. `yarn install`
> 3. Restart Android Studio
> 4. Build for Android

after this it is ok, but once you do ./gradlew clean and than problem show again, you need to redo 1~4

kashkashio, 2025-05-28:
very annoying to clean all cache and rebuild everytime you change a flavor or do different apk builds, no idea how to solve this forever? its very flaky and makes the android development not a nice experience

it is a caching issue for sure, maybe we can add a step that cleans what ever it is on every build?