margelo/react-native-vision-camera #2014: ❓[CodeScanner][Android] What is the best way to determine frame boundaries?

spsaucier, 2023-10-16:
### Question

Using the new `codeScanner`, each code's `frame` includes `height`, `width`, `x`, and `y`.

What is the best way to know/determine the frame boundaries, so that we can tell that, say, the `x` value is at 50% of the frame's width?

It does not seem to always be the same as `Dimensions.get('screen').width` and `.height`.
I'm effectively trying to get `regionOfInterest` working on our side, since that doesn't seem to be working as intended in the package.

### What I tried

`Dimensions.get('screen').width`/`height` appears to work as expected on iOS (that is, it aligns with `x` and `y` exactly, but in my Android testing, the `x` or `y` values can exceed those.

This is from when the code was at the bottom-left of the view (per portrait mode) in Android:
```
{
    "width": 89,
    "y": 486,
    "height": 87,
    "x": 1110,
    "DimensionsHeight": 914,
    "DimensionsWidth": 432
  }
```
Android is setting `device.sensorOrientation === "landscape-right"` (aside: this is inconsistent with iOS and doesn't seem to be configurable), so `x` should always be lower than `DimensionsHeight - width`, and `y` should always be lower than `DimensionsWidth - height` right? `x` and `y`'s values are too high for `Dimensions.get('window')`.

### VisionCamera Version

3.4.0

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

mrousavy, 2023-10-16:
It's relative to the currently active Camera format. You need to select a format using `useCameraFormat`. I will document this more

spsaucier, 2023-10-18:
Given your response, here is some code which converts the code's frame into constant-portrait coordinate values for its corner points:

```ts
const pxDiff = Math.max(PixelRatio.get() - 1, 1);

const portraitVideoHeight = Math.max(
  format.videoWidth,
  format.videoHeight
);
const portraitVideoWidth = Math.min(
  format.videoWidth,
  format.videoHeight
);

const getPortraitCornerPoints = (
  code: Code,
  device: CameraDevice | undefined,
  portraitVideoWidth: number
) => {
  if (!code.frame) return undefined;

  return {
    tl:
      device?.sensorOrientation === 'portrait'
        ? [code.frame.x * pxDiff, code.frame.y * pxDiff]
        : device?.sensorOrientation === 'landscape-left' ||
          device?.sensorOrientation === 'landscape-right'
        ? [
            portraitVideoWidth -
              (code.frame.y * pxDiff + code.frame.height * pxDiff),
            code.frame.x * pxDiff,
          ]
        : [code.frame.x * pxDiff, code.frame.y * pxDiff],
    tr:
      device?.sensorOrientation === 'portrait'
        ? [
            code.frame.x * pxDiff + code.frame.width * pxDiff,
            code.frame.y * pxDiff,
          ]
        : device?.sensorOrientation === 'landscape-left' ||
          device?.sensorOrientation === 'landscape-right'
        ? [portraitVideoWidth - code.frame.y * pxDiff, code.frame.x * pxDiff]
        : [code.frame.x * pxDiff, code.frame.y * pxDiff],
    br:
      device?.sensorOrientation === 'portrait'
        ? [
            code.frame.x * pxDiff + code.frame.width * pxDiff,
            code.frame.y * pxDiff + code.frame.height * pxDiff,
          ]
        : device?.sensorOrientation === 'landscape-left' ||
          device?.sensorOrientation === 'landscape-right'
        ? [
            portraitVideoWidth - code.frame.y * pxDiff,
            code.frame.x * pxDiff + code.frame.width * pxDiff,
          ]
        : [code.frame.x * pxDiff, code.frame.y * pxDiff],
    bl:
      device?.sensorOrientation === 'portrait'
        ? [
            code.frame.x * pxDiff,
            code.frame.y * pxDiff + code.frame.height * pxDiff,
          ]
        : device?.sensorOrientation === 'landscape-left' ||
          device?.sensorOrientation === 'landscape-right'
        ? [
            portraitVideoWidth -
              (code.frame.y * pxDiff + code.frame.height * pxDiff),
            code.frame.x * pxDiff + code.frame.width * pxDiff,
          ]
        : [code.frame.x * pxDiff, code.frame.y * pxDiff],
  };
};
```

NOTE: `landscape-left` is currently the same as `landscape-right`. Context:

In `3.4.1`, iOS codeScanner is:
* No longer in `sensorOrientation` `'portrait'` as it has been
* Reporting `device.sensorOrientation === 'landscape-left'`
* It is _actually_ in `'landscape-right'` orientation -- that is, `x === y === 0` is true when the code is in the top-righthand corner (when the home button is at the bottom [portrait orientation]).

mrousavy, 2023-10-19:
Oh wow, that's a lot of code 😅 

> It is actually in 'landscape-right' orientation

Wait, you mean iPhone Cameras are actually landscape-right instead of landscape-left?

spsaucier, 2023-10-19:
That's right - they are being shown as landscape-left but read the barcodes the same as Android's landscape-right.

mrousavy, 2023-10-19:
Gotcha, I'll rename this then. thanks!

mrousavy, 2023-10-19:
Related: https://github.com/mrousavy/react-native-vision-camera/issues/2046

mrousavy, 2024-01-15:
This has been implemented a few weeks ago! :) https://react-native-vision-camera.com/docs/api/interfaces/CodeScannerFrame

closed 2024-01-15 by mrousavy