margelo/react-native-vision-camera #2015: 🔧 AVCaptureConnection has no member isVideoRotationAngleSupported

fobos531, 2023-10-16:
### How were you trying to build the app?

I was trying to build the latest release (3.4.0) on EAS. The android build works fine, but the iOS one seems to fail at `AVCaptureOutput+mirror.swift` and `CameraDevicesManager.swift`

[image]

### Full build logs

```tsx
See entire build log in the attached file (had to upload since the log is 22 MB): https://dropover.cloud/5dbe38
```

### Project dependencies

```json
"dependencies": {
    "@config-plugins/react-native-webrtc": "^7.0.0",
    "@expo-google-fonts/inter": "^0.2.3",
    "@expo/react-native-action-sheet": "^4.0.1",
    "@gorhom/bottom-sheet": "4.5.0",
    "@gorhom/portal": "^1.0.14",
    "@hookform/resolvers": "3.1.0",
    "@intercom/intercom-react-native": "^4.0.1",
    "@lukemorales/query-key-factory": "^0.6.1",
    "@microsoft/signalr": "^7.0.12",
    "@notifee/react-native": "^7.8.0",
    "@pietile-native-kit/keyboard-aware-scrollview": "^1.5.0",
    "@pr/api": "*",
    "@pr/types": "*",
    "@react-native-community/datetimepicker": "7.2.0",
    "@react-native-community/hooks": "^3.0.0",
    "@react-native-community/netinfo": "9.3.10",
    "@react-native-firebase/app": "^18.4.0",
    "@react-native-firebase/messaging": "^18.4.0",
    "@react-native-menu/menu": "^0.8.0",
    "@react-native-picker/picker": "^2.4.10",
    "@react-navigation/bottom-tabs": "^6.5.7",
    "@react-navigation/elements": "^1.3.17",
    "@react-navigation/material-top-tabs": "^6.6.2",
    "@react-navigation/native": "^6.1.6",
    "@react-navigation/native-stack": "^6.9.12",
    "@reduxjs/toolkit": "^1.9.5",
    "@sentry/react-native": "5.5.0",
    "@shopify/flash-list": "^1.4.3",
    "@shopify/react-native-skia": "^0.1.213",
    "@stream-io/react-native-webrtc": "^104.0.1",
    "@stream-io/video-react-native-sdk": "0.0.27",
    "@tanstack/react-query": "^4.32.6",
    "@tanstack/react-query-persist-client": "^4.32.6",
    "@vvo/tzdb": "^6.108.0",
    "app": "*",
    "axios": "^1.4.0",
    "babel-plugin-inline-dotenv": "^1.7.0",
    "burnt": "^0.11.4",
    "config-plugin-react-native-intercom": "^1.10.1",
    "dayjs": "^1.11.9",
    "expo": "49.0.13",
    "expo-application": "~5.3.0",
    "expo-asset": "8.10.1",
    "expo-av": "~13.4.1",
    "expo-blur": "~12.4.1",
    "expo-brightness": "~11.4.0",
    "expo-build-properties": "~0.8.3",
    "expo-camera": "~13.4.4",
    "expo-clipboard": "~4.3.1",
    "expo-constants": "~14.4.2",
    "expo-dev-client": "~2.4.11",
    "expo-device": "~5.4.0",
    "expo-document-picker": "~11.5.4",
    "expo-font": "~11.4.0",
    "expo-haptics": "~12.4.0",
    "expo-image": "~1.3.4",
    "expo-image-manipulator": "~11.3.0",
    "expo-image-picker": "~14.3.2",
    "expo-insights": "~0.2.0",
    "expo-linear-gradient": "~12.3.0",
    "expo-linking": "~5.0.2",
    "expo-local-authentication": "~13.4.1",
    "expo-localization": "~14.3.0",
    "expo-location": "~16.1.0",
    "expo-media-library": "~15.4.1",
    "expo-notifications": "~0.20.1",
    "expo-sharing": "~11.5.0",
    "expo-splash-screen": "~0.20.5",
    "expo-status-bar": "~1.6.0",
    "expo-system-ui": "~2.4.0",
    "expo-updates": "~0.18.14",
    "expo-web-browser": "~12.3.2",
    "i18next": "^23.5.1",
    "jotai": "^2.2.3",
    "just-clone": "^6.2.0",
    "jwt-decode": "^3.1.2",
    "moti": "^0.25.3",
    "nanoid": "^4.0.2",
    "nativewind": "^2.0.11",
    "react": "18.2.0",
    "react-hook-form": "^7.45.2",
    "react-i18next": "^13.2.2",
    "react-native": "0.72.5",
    "react-native-avoid-softinput": "^3.1.5",
    "react-native-check-version": "^1.1.1",
    "react-native-collapsible": "^1.6.1",
    "react-native-draggable-flatlist": "^4.0.1",
    "react-native-gesture-handler": "~2.12.0",
    "react-native-image-picker": "5.0.2",
    "react-native-incall-manager": "^4.1.0",
    "react-native-ios-context-menu": "^1.15.3",
    "react-native-keyboard-controller": "^1.5.5",
    "react-native-maps": "2.0.0-beta.14",
    "react-native-mmkv": "^2.8.0",
    "react-native-modal-datetime-picker": "^15.0.1",
    "react-native-modalfy": "^3.5.2",
    "react-native-pager-view": "^6.2.0",
    "react-native-paper": "5.0.0-rc.6",
    "react-native-permissions": "^3.8.0",
    "react-native-qrcode-svg": "^6.2.0",
    "react-native-reanimated": "~3.3.0",
    "react-native-render-html": "^6.3.4",
    "react-native-responsive-screen": "^1.4.2",
    "react-native-safe-area-context": "4.6.3",
    "react-native-screens": "~3.22.0",
    "react-native-square-reader-sdk": "https://github.com/fobos531/react-native-square-reader-sdk",
    "react-native-svg": "13.9.0",
    "react-native-tab-view": "^3.5.1",
    "react-native-toast-message": "2.1.5",
    "react-native-vision-camera": "^3.4.0",
    "react-native-webview": "13.2.2",
    "react-redux": "^8.1.2",
    "react-use-signalr": "^0.0.3",
    "redux": "^4.2.1",
    "rn-iphone-helper": "^2.0.3",
    "sentry-expo": "~7.0.0",
    "stream-chat": "8.12.1",
    "stream-chat-expo": "5.18.0",
    "universal-tooltip": "^1.0.0",
    "use-debounce": "^9.0.4",
    "victory-native": "^36.6.11",
    "yup": "^1.2.0",
    "yup-phone-lite": "^2.0.1",
    "zeego": "^1.6.2"
  },
```

### VisionCamera Version

3.4.0

### Target platforms

iOS

### Operating system

MacOS

### Can you build the VisionCamera Example app?

No, I cannot build the Example app either

### Additional information

- [X] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

fobos531, 2023-10-16:
My bad, I was building with XCode 14.3.1, not Xcode 15.0 as outlined in the release notes.

closed 2023-10-16 by fobos531

tmoran-stenoa, 2023-10-16:
Had the same issue. It would be good to add that XCode 15 is required to the docs, I couldn't find any references of this elsewhere.

Thank you @fobos531 for pointing me to the release notes, you've saved me from a big headache!

bfar97, 2023-10-17:
Thanks for pointing the solution out, had the same issue!

taekeun-two, 2023-10-19:
https://expo.dev/changelog/2023/09-28-new-xcode-ios#added-xcode-15-image-on-eas-build

For building using XCode15 in eas, the change should be applied.

```json
{
  "build": {
    "production": {
      "ios": {
        "image": "latest"
      }
    }
  }
}
```