margelo/react-native-vision-camera #2017: 🐛  Problems loading custom frame processors

rafaelmotta, 2023-10-16:
### What's happening?

Since I updated to xcode 15 / ios 17, I started facing some issues on my app which was working great. I can't use anymore my custom frame processors - they always fail when loading. 

I could even reproduce the same problem using the example code [here](https://github.com/mrousavy/react-native-vision-camera/tree/main/package/example), but using "example_kotlin_swift_plugin". The "examplePlugin" works great.

Frame Processor threw an error: Failed to load Frame Processor Plugin "example_kotlin_swift_plugin"!

### Reproduceable Code

```tsx
import * as React from 'react'
import { useRef, useState, useCallback, useMemo } from 'react'
import { StyleSheet, Text, View } from 'react-native'
import { PinchGestureHandler, PinchGestureHandlerGestureEvent, TapGestureHandler } from 'react-native-gesture-handler'
import { CameraRuntimeError, PhotoFile, useCameraDevice, useCameraFormat, useFrameProcessor, VideoFile } from 'react-native-vision-camera'
import { Camera } from 'react-native-vision-camera'
import { CONTENT_SPACING, MAX_ZOOM_FACTOR, SAFE_AREA_PADDING, SCREEN_HEIGHT, SCREEN_WIDTH } from './Constants'
import Reanimated, { Extrapolate, interpolate, useAnimatedGestureHandler, useAnimatedProps, useSharedValue } from 'react-native-reanimated'
import { useEffect } from 'react'
import { useIsForeground } from './hooks/useIsForeground'
import { StatusBarBlurBackground } from './views/StatusBarBlurBackground'
import { CaptureButton } from './views/CaptureButton'
import { PressableOpacity } from 'react-native-pressable-opacity'
import MaterialIcon from 'react-native-vector-icons/MaterialCommunityIcons'
import IonIcon from 'react-native-vector-icons/Ionicons'
import type { Routes } from './Routes'
import type { NativeStackScreenProps } from '@react-navigation/native-stack'
import { useIsFocused } from '@react-navigation/core'
import { exampleKotlinSwiftPlugin } from './frame-processors/ExampleKotlinSwiftPlugin'
import { usePreferredCameraDevice } from './hooks/usePreferredCameraDevice'

const ReanimatedCamera = Reanimated.createAnimatedComponent(Camera)
Reanimated.addWhitelistedNativeProps({
  zoom: true,
})

const SCALE_FULL_ZOOM = 3
const BUTTON_SIZE = 40

type Props = NativeStackScreenProps<Routes, 'CameraPage'>
export function CameraPage({ navigation }: Props): React.ReactElement {
  const camera = useRef<Camera>(null)
  const [isCameraInitialized, setIsCameraInitialized] = useState(false)
  const [hasMicrophonePermission, setHasMicrophonePermission] = useState(false)
  const zoom = useSharedValue(0)
  const isPressingButton = useSharedValue(false)

  // check if camera page is active
  const isFocussed = useIsFocused()
  const isForeground = useIsForeground()
  const isActive = isFocussed && isForeground

  const [cameraPosition, setCameraPosition] = useState<'front' | 'back'>('back')
  const [enableHdr, setEnableHdr] = useState(false)
  const [flash, setFlash] = useState<'off' | 'on'>('off')
  const [enableNightMode, setEnableNightMode] = useState(false)

  // camera device settings
  const [preferredDevice] = usePreferredCameraDevice()
  let device = useCameraDevice(cameraPosition)

  if (preferredDevice != null && preferredDevice.position === cameraPosition) {
    // override default device with the one selected by the user in settings
    device = preferredDevice
  }

  const [targetFps, setTargetFps] = useState(60)

  const screenAspectRatio = SCREEN_HEIGHT / SCREEN_WIDTH
[... 361 lines cut ...]
  "supportsFocus": true,
  "supportsLowLightBoost": false,
  "supportsRawCapture": false
}
```

### Device

iPhone 12 iOS 17.0.3

### VisionCamera Version

3.3.1

### Can you reproduce this issue in the VisionCamera Example app?

Yes, I can reproduce the same issue in the Example app here

### Additional information

- [ ] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2023-10-16:
Wait so they do not get registered anymore?

How do you register the FP Plugin in native?

rafaelmotta, 2023-10-16:
Exactly. As I mentioned, even the `example` project I could reproduce the same problem.

But this is my code for context:

```swift
#import <Foundation/Foundation.h>
#import <VisionCamera/FrameProcessorPlugin.h>
#import <VisionCamera/FrameProcessorPluginRegistry.h>
#import <VisionCamera/Frame.h>

#if __has_include("My_App_Clip-Swift.h")
#import "My_App_Clip-Swift.h"
#else
#import "My-Swift.h"
#endif

@interface FaceDetectionPlugin (FrameProcessorPluginLoader)
@end

@implementation FaceDetectionPlugin (FrameProcessorPluginLoader)

+ (void)initialize {
  [FrameProcessorPluginRegistry addFrameProcessorPlugin:@"detectFace"
                                        withInitializer:^FrameProcessorPlugin* _Nonnull(NSDictionary* _Nullable options) {
    return [[FaceDetectionPlugin alloc] init];
  }];
}

@end
```

and

```swift
import VisionCamera
import CoreMotion
import AVFoundation
import CoreImage
import CoreVideo
import Vision

@objc(FaceDetectionPlugin)
public class FaceDetectionPlugin: FrameProcessorPlugin {
  @objc public init(withOptions options: [AnyHashable : Any]) {
    super.init()
  }
  
  public static func detectFace(_ pixelBuffer: CVPixelBuffer) -> VNFaceObservation? {
    let imageRequestHandler = VNImageRequestHandler(cvPixelBuffer: pixelBuffer, orientation: .up, options: [:])
    let faceDetectionRequest = VNDetectFaceLandmarksRequest()
    do {
      try imageRequestHandler.perform([faceDetectionRequest])
    } catch {
      print("Failed to perform the request: \(error)")
      return nil
    }
    guard let results = faceDetectionRequest.results else { return nil }
    if results.count == 1, let faceObservation = results[0] as? VNFaceObservation {
      return faceObservation
    }
    return nil
  }
  
  public static func isFaceInCenter(_ observation: VNFaceObservation, frameSize: CGSize) -> Bool {
    let centerFaceX = observation.boundingBox.midX * frameSize.width
    let centerFaceY = observation.boundingBox.midY * frameSize.height
    let frameCenter = CGPoint(x: frameSize.width / 2.0, y: frameSize.height / 2.0)
    let tolerance: CGFloat = 0.2
    let diffX = abs(centerFaceX - frameCenter.x)
    let diffY = abs(centerFaceY - frameCenter.y)
    return diffX <= frameSize.width * tolerance && diffY <= frameSize.height * tolerance
  }
  
  @available(iOS 12.0, *)
  public static func isFaceStraight(_ faceObservation: VNFaceObservation) -> Bool {
    guard let yaw = faceObservation.yaw, let roll = faceObservation.roll else { return false }
    let yawThreshold = CGFloat(0.15)
    let rollThreshold = CGFloat(0.15)
    // Check if the absolute values for yaw and roll are within the defined thresholds
    // If they are, the face is considered straight
    return abs(yaw.floatValue) < Float(yawThreshold) && abs(roll.floatValue) < Float(rollThreshold)
  }

  @objc override public func callback(_ frame: Frame, withArguments arguments: [AnyHashable : Any]?) -> Any {
    let buffer = frame.buffer
    
    guard let pixelBuffer = CMSampleBufferGetImageBuffer(frame.buffer) else {
      return [
        "hasFace": false,
        "isFaceInCenter": false,
        "isFaceStraight": false
      ]
    }
    
    guard let arguments = arguments,
         let frameOptions = arguments as? [String: Any],
         let frameSizeWidth = frameOptions["width"] as? CGFloat,
         let frameSizeHeight = frameOptions["height"] as? CGFloat else {
       return [
           "hasFace": false,
           "isFaceInCenter": false,
           "isFaceStraight": false
       ]
   }
    
    let frameSize = CGSize(width: frameSizeWidth, height: frameSizeHeight)
    if let faceObservation = FaceDetectionPlugin.detectFace(pixelBuffer) {
      let isFaceInCenter = FaceDetectionPlugin.isFaceInCenter(faceObservation, frameSize: frameSize)
      var isFaceStraight = false
      if #available(iOS 12.0, *) {
        isFaceStraight = FaceDetectionPlugin.isFaceStraight(faceObservation)
      }

      return [
        "hasFace": true,
        "isFaceInCenter": isFaceInCenter,
        "isFaceStraight": isFaceStraight
      ]
    }
    
    return [
      "hasFace": false,
      "isFaceInCenter": false,
      "isFaceStraight": false
    ]
  }
}

```

iBotPeaches, 2023-10-16:
I was typing a response to this saying that I replicated this issue, then built the sample and also replicated it there (`__example_plugin`) failing, but not the other.

Then I realized I was doing all of this on the v2 branch and this is v3. Completely different situation, but oddly the same type of issue where FPs stopped working. I'll submit my own issue to not mix things.

fedpre, 2023-10-17:
The problem is `+ (void)initialize {` . Change it back to `load` and it will start working again. I had the same issue and I fixed it this way. Which is odd as `load` shouldn't work with Xcode 15 but it is working for me

mrousavy, 2023-10-17:
Does `+load` work and get called properly? I think this is the best solution.

fedpre, 2023-10-17:
@mrousavy It does on my project (Xcode 15 and 0.72.5 RN) and before the frame processor never got initialized. I wonder why I am not getting the `+load` error though..

mrousavy, 2023-10-17:
Perfect, so `+load` is the recommended approach now. Just like before.

I think that was just a bug in Xcode 14.5 or Xcode 15 beta that `+load` didnt work for a while. Now it works perfectly

closed 2023-10-17 by mrousavy

rafaelmotta, 2023-10-17:
Gotcha. Can you help me figure out what I'm doing wrong? I can still reproduce the same problem using the `examples` project + `ExampleSwiftFrameProcessorPlugin`, even after updating `initialize` to `load`:

[image]

Now I'm getting this:

```
objc[58598]: Swift class extensions and categories on Swift classes are not allowed to have +load methods
```

[image]

fedpre, 2023-10-17:
@rafaelmotta What version of Xcode do you have? And RN?

rafaelmotta, 2023-10-17:
I'm using:

- macOS Sonoma Version 14.0 (23A344)
- Xcode Version 15.0 (15A240d)

And:

```json
react-native": "^0.72.3"
```

Tried using `vision-camera` 3.3.1, 3.4.0, but I got all the same issues.

mrousavy, 2023-10-17:
Is this the same Swift class? You need a separate ObjC class that registers the Swift class (FPPlugin).

rafaelmotta, 2023-10-17:
> Is this the same Swift class? You need a separate ObjC class that registers the Swift class (FPPlugin).

not sure if I followed 100%, but yes, I have an Obj-c with registers the plugin and the Swift class with the actual implementation, like [here](https://github.com/mrousavy/react-native-vision-camera/issues/2017#issuecomment-1765177338)

rafaelmotta, 2023-10-18:
I still couldn't fix this in v3, even after changing `initialize` => `load`.  However, after the fix https://github.com/mrousavy/react-native-vision-camera/pull/2025, I could make it work, so I'm reverting to v2 for now