margelo/react-native-vision-camera #2020: 🐛 Camera stuck on first takePhoto call on ADV Pixel 5 API 31.32

staiyojp, 2023-10-17:
### What's happening?

I am using react-native-vision-camera 3.4.0 library.
I have problems taking photos with the takePhoto function.
It seems that after the first call of the function takePhoto will hang and return no results, on subsequent calls an error is returned "[unknown/unknown] CameraDevice was already closed]".

### Reproduceable Code

```tsx
const cameraRef = useRef(null);

const device = useCameraDevice('back')

const takePicture = useCallback(() => {
  const options = {
    qualityPrioritization: 'quality',
    flash: 'auto',
  };
  cameraRef.current?.takePhoto(options)
    .then((res) => {
      console.log('==res==', res);
    }).catch((err) => {
      console.log('==Error Camera Take Photo==', err);
    })
}, [cameraRef.current])

return <>
  {
    !device ? <NoCameraErrorView />
    :
    <Camera
      ref={ref => (cameraRef.current = ref)}
      style={StyleSheet.absoluteFill}
      device={device}
      isActive={true}
      photo={true} />
    }
</>
```

### Relevant log output

```shell
[Error: [unknown/unknown] CameraDevice was already closed]
```

### Camera Device

```json
{
  id: undefined,
  devices: [
    ultra-wide-angle-camera,
    wide-angle-camera
  ],
  formats: null,
  hasFlash: false,
  position:back
}
```

### Device

Android Pixel 5 API 31, Android Pixel 5 API 32

### VisionCamera Version

3.4.0

### Can you reproduce this issue in the VisionCamera Example app?

Yes, I can reproduce the same issue in the Example app here

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

trongdaim2m, 2023-10-17:
I also had the same problem with Pixel 4 API 31, Pixel 4 API 32.
Pixel 5 API 30 and Pixel 5 API 33 it works fine.

staiyojp, 2023-10-17:
I also had the same problem with Pixel 4 API 31, Pixel 4 API 32.
Pixel 5 API 30 and Pixel 5 API 33 it works fine.

mrousavy, 2023-10-17:
Hm, that's weird. Honestly there's not much I can do without having a Pixel 4 in my hands.

Here's what you can do:

1. Download Google's **official** Camera example apps, run the one you want (probably Camera2Basic) and see if the problem also exists there.
  
    **If it does exist there as well**; It's Google's fault. They build poor APIs that don't work. We need to find a workaround. Try this:
        1. Try using [CameraX](https://developer.android.com/training/camerax). If the issue does not exist there; browse the CameraX codebase and see how they do it. Then let me know so I can implement the same fix!
        2. Try using [CameraView](https://github.com/natario1/CameraView). If the issue does not exist there; browse the CameraView codebase and see how they do it.Then let me know so I can implement the same fix!
        3. Try using [camerakit-android](https://github.com/CameraKit/camerakit-android). If the issue does not exist there; browse the camerakit-android codebase and see how they do it. Then let me know so I can implement the same fix!

    **If it does not exist there**; It's VisionCamera's fault. We probably do something that we shouldn't do.
    This is VisionCamera's takePhoto function on Android: https://github.com/mrousavy/react-native-vision-camera/blob/3b1d8353a5b723e6ae488d23ad79e9f567f9ade8/package/android/src/main/java/com/mrousavy/camera/core/CameraSession.kt#L214-L255

    You can try to remove a bunch of stuff from the Photo Capture request - maybe I am trying to set a field that is just not supported (e.g. video stabilization). You can try to remove a bunch of stuff from this function: https://github.com/mrousavy/react-native-vision-camera/blob/3b1d8353a5b723e6ae488d23ad79e9f567f9ade8/package/android/src/main/java/com/mrousavy/camera/extensions/CameraDevice%2BcreatePhotoCaptureRequest.kt#L35-L92

    ...and then see if something here can make this work. Maybe remove zoom, maybe remove video stabilization, etc.

2. Browse StackOverflow to find similar issues and try the suggestions (or let me know)
3. Pay me my hourly consultancy fee so I will do everything up there and fix the problem for you.

mrousavy, 2023-10-17:
Btw., wtf? Why is your `id` undefined?

[image]

Which `format` are you using? Can you try to use a small format to see if that works?

```tsx
const format = useCameraFormat(device, [
  { photoResolution: { width: 0, height: 0 } }
])

...

<Camera ... format={format} />
```

mrousavy, 2023-10-17:
Also, honestly this is just rude.

[image]

I included precise following steps to include the logs (**which I need to help you**) in here

[image]

..and you simply chose to ignore it, because idk it was too much effort? The logs are important! That's what logs are for!

staiyojp, 2023-10-17:
I am really sorry.
Because logging is quite long, I've put it on GGdocs.
Link here: https://docs.google.com/document/d/1LVN-imGKa-fHykdvVkOrdUj7ouOKDT-QUty_kCifJds/edit
Thank you for your feedback.

mrousavy, 2023-10-18:
Is that a Simulator? The logs say that some camera function is not implemented..

[image]

mrousavy, 2023-10-18:
This might actually be an Android/Google bug: https://buganizer.corp.google.com/issues/209821054.

See this bug report in flutter https://github.com/flutter/flutter/issues/102644

staiyojp, 2023-10-19:
Okay, I understand. Thank you very much

closed 2023-10-19 by staiyojp