margelo/react-native-vision-camera #2023: 🐛 Camera is a black screen

basharmalkawi, 2023-10-17:
### What's happening?

I get this error and the camera is a black screen

### Reproduceable Code

```tsx
react native code:
import React, { useState } from 'react';
import { Text, View, StyleSheet, TouchableOpacity } from 'react-native';
import { Camera, runAsync, useCameraDevice, useCameraDevices, useFrameProcessor } from 'react-native-vision-camera';
import { documentRecognition } from './src/Helpers';

 const App = () => {

   const [camera, setCamera] = useState(false)
  const frameProcessor = useFrameProcessor((frame) => {
    console.log('---before---')
    const faces = documentRecognition(frame)
    console.log(`Faces in Frame: ${faces}`)
  }, [])

  const devices = useCameraDevice('back')

  const device = devices

  if (device == null) {
    return <View />;
  }
  return (
    <View style={{flex: 1}}>
      <TouchableOpacity title={'Open Camera'} onPress={() =>setCamera(true)} style={{backgroundColor: 'red', marginTop: 100, padding: 100, zIndex: 2}}>
        <Text>Open</Text>
      </TouchableOpacity>
     {camera && <Camera
        frameProcessor={frameProcessor}
        style={StyleSheet.absoluteFill}
        isActive={camera}
        device={device}
        torch={'on'}
        //pixelFormat={'yuv'}
      />}
    </View>
  )
}

export default App;

android native code :
package  com.camerav3.frameprocessor;

import androidx.annotation.NonNull;
import androidx.annotation.Nullable;
import com.mrousavy.camera.frameprocessor.Frame;
import com.mrousavy.camera.frameprocessor.FrameProcessorPlugin;

import java.util.Map;

public class documentRecognitionFrameProcessorPlugin extends FrameProcessorPlugin {
    @Nullable
    @Override
    public Object callback(@NonNull Frame frame, @Nullable Map<String, Object> arguments) {
        // code goes here
        return null;
    }
}
```

### Relevant log output

```shell
Failed to configure session: [system/frame-processors-unavailable] Frame Processors are unavailable! Reason: Frame Processors require API 29 or higher. (Q)
 com.mrousavy.camera.core.FrameProcessorsUnavailableError: [system/frame-processors-unavailable] Frame Processors are unavailable! Reason: Frame Processors require API 29 or higher. (Q)
  at com.mrousavy.camera.core.VideoPipeline.<init>(VideoPipeline.kt:88)
  at com.mrousavy.camera.core.outputs.CameraOutputs.<init>(CameraOutputs.kt:158)
  at com.mrousavy.camera.core.CameraSession.configureSession(CameraSession.kt:132)
  at com.mrousavy.camera.CameraView.configureSession(CameraView.kt:246)
  at com.mrousavy.camera.CameraView.access$configureSession(CameraView.kt:48)
  at com.mrousavy.camera.CameraView$setupPreviewView$previewView$1$1.invokeSuspend(CameraView.kt:154)
  at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:33)
  at kotlinx.coroutines.DispatchedTask.run(DispatchedTask.kt:106)
  at android.os.Handler.handleCallback(Handler.java:790)
  at android.os.Handler.dispatchMessage(Handler.java:99)
  at android.os.Looper.loop(Looper.java:214)
  at android.os.HandlerThread.run(HandlerThread.java:65)
```

### Camera Device

```json
"dependencies": {
    "react": "18.2.0",
    "react-native": "0.72.6",
    "react-native-reanimated": "^3.5.4",
    "react-native-vision-camera": "^3.4.0",
    "react-native-worklets-core": "^0.2.2"
  },
```

### Device

Samsung s23 , OPPO A5s

### VisionCamera Version

3.4.0

### Can you reproduce this issue in the VisionCamera Example app?

No, I cannot reproduce the issue in the Example app

### Additional information

- [ ] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2023-10-17:
> Reason: Frame Processors require API 29 or higher. (Q)

?

closed 2023-10-17 by mrousavy

basharmalkawi, 2023-10-17:
> > Reason: Frame Processors require API 29 or higher. (Q)
> 
> ?

I get this error

mrousavy, 2023-10-17:
yes but what do you not understand about that? They require API 29 or higher. Upgrade your phone

basharmalkawi, 2023-10-17:
@mrousavy , 
 thanks a lot

basharmalkawi, 2023-10-17:
@mrousavy  
do you have any idea about this 
FATAL EXCEPTION: mrousavy/VisionCamera.video
 Process: com.tahaluf.idify, PID: 21705
com.facebook.jni.CppException: Compiling JS failed: 1:1:invalid empty parentheses '( )' Buffer size 3 starts with: 280a29 and has protection mode(s): rw-p
   at com.mrousavy.camera.frameprocessor.FrameProcessor.call(Native Method)
at com.mrousavy.camera.core.VideoPipeline._init_$lambda$0(VideoPipeline.kt:105)
at com.mrousavy.camera.core.VideoPipeline.$r8$lambda$EMriwS_FKhtjTFtppM84Z5V8tiI(Unknown Source:0)
    at com.mrousavy.camera.core.VideoPipeline$$ExternalSyntheticLambda0.onImageAvailable(Unknown Source:2)
 at android.media.ImageReader$ListenerHandler.handleMessage(ImageReader.java:798)
                                                                                                    	at android.os.Handler.dispatchMessage(Handler.java:110)
                                                                                                    	at android.os.Looper.loop(Looper.java:219)
                                                                                                    	at android.os.HandlerThread.run(HandlerThread.java:67)

celik75, 2023-10-17:
> com.facebook.jni.CppException: Compiling JS failed: 1:1:invalid empty parentheses '( )' Buffer size 3 starts with: 280a29 and has protection mode(s): rw-p

Fix:
https://github.com/mrousavy/react-native-vision-camera/issues/1776#issuecomment-1757204827

mrousavy, 2023-10-19:
In VisionCamera 3.5.0 you can now also use Frame Processors on API lower than <29