margelo/react-native-vision-camera #2026: Black screen on scanner after update to 3.4.0 (iOS) 🐛

maksimLobachevskiy, 2023-10-17:
### What's happening?

Stack trace in the xCode says that code type is not supported, I tried all code types.
When I go scan screen it is just black.

### Reproduceable Code

```tsx
const deviceBack = useCameraDevice('back');
 
const codeScanner = useCodeScanner({
    codeTypes: [
      'qr',
      'pdf-417',
      'aztec',
      'data-matrix',
    ],
    onCodeScanned: codes => {   console.log(codes) },
  });

 <Camera
            device={deviceBack}
            codeScanner={codeScanner}
            format={format}
          />
```

### Relevant log output

```shell
2023-10-17 16:17:43.308655+0200 OneStopPro[4812:807738] [native] VisionCamera.configureDevice(configuration:): Successfully configured Input Device!
2023-10-17 16:17:43.308663+0200 OneStopPro[4812:807738] [native] VisionCamera.configureOutputs(configuration:): Configuring Outputs...
2023-10-17 16:17:43.308672+0200 OneStopPro[4812:807738] [native] VisionCamera.configureOutputs(configuration:): Adding Code Scanner output...
2023-10-17 16:17:43.308723+0200 OneStopPro[4812:807738] [native] VisionCamera.onError(_:): Invoking onError(): The codeType "ean-8" is not supported by the Code Scanner!
2023-10-17 16:17:49.747913+0200 OneStopPro[4812:807540] [native] VisionCamera.didSetProps(_:): Updating 1 props: [codeScannerOptions]
2023-10-17 16:17:49.747943+0200 OneStopPro[4812:807540] [native] VisionCamera.configure(_:): Updating Session Configuration...
2023-10-17 16:17:49.748102+0200 OneStopPro[4812:807773] [native] VisionCamera.configure(_:): Beginning CameraSession configuration...
2023-10-17 16:17:49.748115+0200 OneStopPro[4812:807785] [native] VisionCamera.configure(_:): Beginning AudioSession configuration...
2023-10-17 16:17:49.748141+0200 OneStopPro[4812:807773] [native] VisionCamera.configureDevice(configuration:): Configuring Input Device...
2023-10-17 16:17:49.748145+0200 OneStopPro[4812:807785] [native] VisionCamera.configureAudioSession(configuration:): Configuring Audio Session...
2023-10-17 16:17:49.748275+0200 OneStopPro[4812:807785] [native] VisionCamera.configure(_:): Committed AudioSession configuration!
2023-10-17 16:17:49.748712+0200 OneStopPro[4812:807773] [native] VisionCamera.configureDevice(configuration:): Configuring Camera com.apple.avfoundation.avcapturedevice.built-in_video:1...
2023-10-17 16:17:49.749657+0200 OneStopPro[4812:807773] [native] VisionCamera.configureDevice(configuration:): Successfully configured Input Device!
2023-10-17 16:17:49.749668+0200 OneStopPro[4812:807773] [native] VisionCamera.configureOutputs(configuration:): Configuring Outputs...
2023-10-17 16:17:49.749678+0200 OneStopPro[4812:807773] [native] VisionCamera.configureOutputs(configuration:): Adding Code Scanner output...
2023-10-17 16:17:49.749735+0200 OneStopPro[4812:807773] [native] VisionCamera.onError(_:): Invoking onError(): The codeType "qr" is not supported by the Code Scanner!
```

### Camera Device

```json
{
  "maxZoom": 121.875,
  "hardwareLevel": "full",
  "minZoom": 1,
  "id": "com.apple.avfoundation.avcapturedevice.built-in_video:0",
  "position": "back",
  "isMultiCam": false,
  "name": "Back Camera",
  "hasFlash": true,
  "supportsFocus": true,
  "physicalDevices": [
    "wide-angle-camera"
  ],
  "supportsRawCapture": false,
  "neutralZoom": 1,
  "sensorOrientation": "landscape-left",
  "supportsLowLightBoost": false,
  "hasTorch": true
}
```

### Device

iPhone 13 Pro (iOS 16.7.1)

### VisionCamera Version

3.4.0

### Can you reproduce this issue in the VisionCamera Example app?

Yes, I can reproduce the same issue in the Example app here

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

stefan-schweiger, 2023-10-17:
Getting the same behavior, android works without a problem. This only seems to affect iOS (17.1 in my case).

maksimLobachevskiy, 2023-10-17:
> Getting the same behavior, android works without a problem. This only seems to affect iOS (17.1 in my case).

Yes after update Android works good.

mrousavy, 2023-10-17:
Duplicate of https://github.com/mrousavy/react-native-vision-camera/issues/2016. I already fixed this but didn't release it yet

closed 2023-10-17 by mrousavy

phamkythanh, 2023-10-19:
Hi @mrousavy upgraded to 3.4.1 but I still get black screen when trying QR scan on iOS 17.0.3. Do you have any idea?

```
<Camera
                        style={StyleSheet.absoluteFill}
                        device={device}
                        isActive={isActive}
                        codeScanner={codeScanner}
                        format={format}
                    />
```
[image]

mrousavy, 2023-10-19:
No because you're not sharing any logs with me.

stefan-schweiger, 2023-10-19:
@phamkythanh try to remove all properties which are not necessary and see if it works (so basically anything besides `isActive`, `device` and maybe `style`). On my iPhone 15 Pro the scanner is working fine with iOS 17.1 and 3.4.1, but when I had the black screen issue in the past it was actually the fault of another property being incorrectly used.

phamkythanh, 2023-10-19:
Ah sorry my bad. I did init a fresh new RN then everything working fine. It
was my fault when rendering Camera in the existing project. Sorry again for
wrong alarm!

mrousavy, 2023-10-19:
> when I had the black screen issue in the past it was actually the fault of another property being incorrectly used.

@stefan-schweiger can you maybe create issues for that separately? VisionCamera should throw precise errors accordingly if something is used wrong, and we need to catch that.