margelo/react-native-vision-camera #2028: Does vision-camera need kotlin-android-extensions❓

hibinoA, 2023-10-18:
### Question

Does vision-camera need kotlin-android-extensions? Can I remove the dependency on kotlin-android-extensions with patch-package?

### What I tried

I'm facing the issue as same as below one.
[Support for kotlin-android-extensions Gradle plugin version 1.6.20 and higher](https://github.com/mrousavy/react-native-vision-camera/issues/1352)
Of course I can resolve it with using `kotlinVersion = "1.6.20"`. However I don't want to change `kotlinVersion` if possible. Therefore I tried to removed the dependency on `kotlin-android-extensions`.

[image]

And it works fine.
I can find that this solution is also used by others.
https://github.com/mrousavy/react-native-vision-camera/issues/1524#issuecomment-1470041637
https://github.com/mrousavy/react-native-vision-camera/issues/1814#issuecomment-1728925822

Also vision-camera does not use View Binding and Parcelable which are provided by kotlin-android-extensions as far as I can see. 

### VisionCamera Version

all

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

mrousavy, 2023-10-18:
Hey! If it doesn't use kotlin-android-extensions anymore then we can safely remove it, can you submit a PR here removing it? Thanks!

mrousavy, 2023-10-18:
Here you go: https://github.com/mrousavy/react-native-vision-camera/pull/2034

hibinoA, 2023-10-19:
Thanks for your quick addressing:rocket: