margelo/react-native-vision-camera #2029: Camera Alignment issues❓

shre6701, 2023-10-18:
### Question

I took the photo from the react vision camera but the photo is rotated 90 degrees irrespective if it is portrait or landscape 

How do we exactly align the images on our app?

### What I tried

camera.takePhoto(); method to take the photo

### VisionCamera Version

3.4.0

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

zzz08900, 2023-10-18:
Take a look at this:
https://github.com/mrousavy/react-native-vision-camera/issues/1891

If you need it, think about funding it.

mrousavy, 2023-10-18:
Exactly what @zzz08900 said.

closed 2023-10-18 by mrousavy