margelo/react-native-vision-camera #2036: 🐛 Tried to add a Frame Processor Plugin with a name that already exists!

pepperav, 2023-10-18:
### What's happening?

I wanted to create a Frame Plugin on Android following guide line but I get always the error in the title

[image]

**XyzFrameProcessorPlugin.java**
```
package com.giuseppe.ravida.exponativeexamplev331;

import androidx.annotation.NonNull;
import androidx.annotation.Nullable;
import com.mrousavy.camera.frameprocessor.Frame;
import com.mrousavy.camera.frameprocessor.FrameProcessorPlugin;

import java.util.Map;

public class XyzFrameProcessorPlugin extends FrameProcessorPlugin {
  @Nullable
  @Override
  public Object callback(@NonNull Frame frame, @Nullable Map<String, Object> arguments) {
    // code goes here
    return "Hello World";
  }
}

```

**XyzFrameProcessorPluginPackage.java**
```
package com.giuseppe.ravida.exponativeexamplev331;

import androidx.annotation.NonNull;
import com.facebook.react.ReactPackage;
import com.facebook.react.bridge.NativeModule;
import com.facebook.react.bridge.ReactApplicationContext;
import com.facebook.react.uimanager.ViewManager;
import com.mrousavy.camera.frameprocessor.FrameProcessorPlugin;
import com.mrousavy.camera.frameprocessor.FrameProcessorPluginRegistry;

import java.util.*;

public class XyzFrameProcessorPluginPackage implements ReactPackage {
  XyzFrameProcessorPluginPackage() {
    FrameProcessorPluginRegistry.addFrameProcessorPlugin("xyz", options -> new XyzFrameProcessorPlugin());
  }

  @NonNull
  @Override
  public List<NativeModule> createNativeModules(@NonNull ReactApplicationContext reactContext) {
    return Collections.emptyList();
  }

  @NonNull
  @Override
  public List<ViewManager> createViewManagers(@NonNull ReactApplicationContext reactContext) {
    return Collections.emptyList();
  }
}
```

**MainApplication.java**
```
package com.giuseppe.ravida.exponativeexamplev331;

import android.app.Application;
import android.content.res.Configuration;
import androidx.annotation.NonNull;

import com.facebook.react.PackageList;
[... 209 lines cut ...]
  "hasTorch": true,
  "position": "back",
  "id": "0"
}
```

### Device

Xiaomi Redmi Note 7 - Android 10

### VisionCamera Version

3.4.1

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [X] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2023-10-19:
Can you try this and see if that works?

```diff
 public class XyzFrameProcessorPluginPackage implements ReactPackage {
-  XyzFrameProcessorPluginPackage() {
-    FrameProcessorPluginRegistry.addFrameProcessorPlugin("xyz", options -> new XyzFrameProcessorPlugin());
-  }
+ static {
+   FrameProcessorPluginRegistry.addFrameProcessorPlugin("xyz", options -> new XyzFrameProcessorPlugin());
+ }
...
```

mateusz1913, 2023-10-19:
I am not Java expert, but will it still require to initialize that ReactPackage somewhere in the user codebase?

mateusz1913, 2023-10-19:
Ah ok, it should be only run once, got it

mrousavy, 2023-10-19:
Nope, the only reason we need to create a `ReactPackage` is because the react-native CLI only includes those files in the `android/` project if it can find a `ReactPackage`.

So this is strictly a "fake" package so the plugin is part of the sources, and we just want to register our plugin there

mateusz1913, 2023-10-19:
If it works, then sure, I'll adjust it in plugin builder

closed 2023-10-19 by mrousavy

pepperav, 2023-10-19:
> Can you try this and see if that works?
>
> ```diff
>  public class XyzFrameProcessorPluginPackage implements ReactPackage {
> -  XyzFrameProcessorPluginPackage() {
> - [...]

Yes it works now! Thank you @mrousavy 🙌

mrousavy, 2023-10-19:
awesome!