margelo/react-native-vision-camera #2042: 🐛 [V3] VisionCameraProxy.initFrameProcessorPlugin does not exist

ShchipskiiYura, 2023-10-19:
### What's happening?

I try create FrameProcessor plugin, read instruction on site, but get error

Property 'initFrameProcessorPlugin' does not exist on type 'TVisionCameraProxy'. Did you mean 'getFrameProcessorPlugin'?ts(2551)
FrameProcessorPlugins.d.ts(21, 5): 'getFrameProcessorPlugin' is declared here.

### Reproduceable Code

```tsx
import { VisionCameraProxy, type Frame } from 'react-native-vision-camera';

const _frameToBase64 = VisionCameraProxy.initFrameProcessorPlugin('frameToBase64');
```

### Relevant log output

```shell
Property 'initFrameProcessorPlugin' does not exist on type 'TVisionCameraProxy'. Did you mean 'getFrameProcessorPlugin'?ts(2551)
FrameProcessorPlugins.d.ts(21, 5): 'getFrameProcessorPlugin' is declared here.
```

### Camera Device

```json
any
```

### Device

iPhone 11 Pro

### VisionCamera Version

3.4.1

### Can you reproduce this issue in the VisionCamera Example app?

Yes, I can reproduce the same issue in the Example app here

### Additional information

- [ ] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2023-10-19:
Ah yea sorry, I already updated the docs for 3.4.2. It is called `getFrameProcessorPlugin` now, but will be called `init...` in the next release.

closed 2023-10-19 by mrousavy

ShchipskiiYura, 2023-10-19:
When will be next release?

mrousavy, 2023-10-19:
idk, maybe today maybe tomorrow