margelo/react-native-vision-camera #2043: 🔧

lhagva98, 2023-10-19:
### How were you trying to build the app?

i tried to build with react native vision camera. but there. are build error at line codes in xcode

 if #available(iOS 17.0, *),
       let userPreferred = AVCaptureDevice.userPreferredCamera {  this is ->  #'userPreferredCamera' is unavailable in iOS
      preferredDevice = userPreferred.toDictionary()
    } else {
      preferredDevice = devices.first
    }

### Full build logs

```tsx
yarn ios --simulator "iPhone 14"
```

### Project dependencies

```json
"dependencies": {
    "@hookform/resolvers": "^3.3.1",
    "@react-native-async-storage/async-storage": "^1.19.3",
    "@react-native-clipboard/clipboard": "^1.12.1",
    "@react-navigation/native": "^6.1.8",
    "@react-navigation/stack": "^6.3.18",
    "@tanstack/react-query": "^5.0.0-rc.6",
    "axios": "^1.5.1",
    "lodash": "^4.17.21",
    "react": "18.2.0",
    "react-hook-form": "^7.47.0",
    "react-native": "0.72.5",
    "react-native-fast-image": "^8.6.3",
    "react-native-gesture-handler": "^2.13.1",
    "react-native-keyboard-aware-scroll-view": "^0.9.5",
    "react-native-linear-gradient": "^2.8.3",
    "react-native-mask-input": "^1.2.3",
    "react-native-modalize": "^2.1.1",
    "react-native-paper": "^5.10.6",
    "react-native-permissions": "^3.9.3",
    "react-native-safe-area-context": "^4.7.2",
    "react-native-screens": "^3.25.0",
    "react-native-svg": "^13.14.0",
    "react-native-vector-icons": "^10.0.0",
    "react-native-vision-camera": "^3.4.1",
    "zod": "^3.22.4",
    "zustand": "^4.4.3"
  },
```

### VisionCamera Version

3.4.1

### Target platforms

iOS

### Operating system

MacOS

### Can you build the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2023-10-19:
Fixed in 3.5.0.

closed 2023-10-19 by mrousavy

HairyRabbit, 2024-07-21:
same issue here, 4.5.0 & RN 0.73.8
would I missing some config

mrousavy, 2024-07-21:
upgrade ur xcode & macos

HairyRabbit, 2024-07-21:
ooops looks like i have to buy a new mac device 🥲