margelo/react-native-vision-camera #2044: IOS build error in version 3.5.0🔧

hamdij0maa, 2023-10-19:
### How were you trying to build the app?

building on iOS simulator

### Full build logs

```tsx
Function: createItemModels(for:itemModelSource:)
Thread:   <_NSMainThread: 0x6000018ac2c0>{number = 1, name = main}
Please file a bug at https://feedbackassistant.apple.com with this warning message and any useful information you can provide.
2023-10-19 12:51:10.793 xcodebuild[20392:88943] [MT] DVTAssertions: Warning in /System/Volumes/Data/SWE/Apps/DT/BuildRoots/BuildRoot11/ActiveBuildRoot/Library/Caches/com.apple.xbs/Sources/IDEFrameworks/IDEFrameworks-22267/IDEFoundation/Provisioning/Capabilities Infrastructure/IDECapabilityQuerySelection.swift:103
Details:  createItemModels creation requirements should not create capability item model for a capability item model that already exists.
Function: createItemModels(for:itemModelSource:)
Thread:   <_NSMainThread: 0x6000018ac2c0>{number = 1, name = main}
Please file a bug at https://feedbackassistant.apple.com with this warning message and any useful information you can provide.
** BUILD FAILED **
```

### Project dependencies

```json
{

"react-native-vision-camera": "^3.5.0",
"react-native-worklets-core": "^0.2.2",
   "react-native-reanimated": "^3.5.4",
}
```

### VisionCamera Version

3.5.0

### Target platforms

iOS

### Operating system

MacOS

### Can you build the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [X] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

ShchipskiiYura, 2023-10-19:
I catch another error 

```Cannot assign to value: 'preferredDevice' is a 'let' constant```

[image]

balmasich, 2023-10-19:
> I catch another error
> 
> `Cannot assign to value: 'preferredDevice' is a 'let' constant`
> 
> [image]

change
`let preferredDevice = devices.first` 
to
`var preferredDevice = devices.first`