margelo/react-native-vision-camera #2045: 🐛  Frame Processor threw an error: Failed to load Frame Processor Plugin "functionName"

ShchipskiiYura, 2023-10-19:
### What's happening?

I create own custom Frame Processor plugin for convert frame to base64, and when I load camera, I catch error

```
 LOG  plugin undefined
 ERROR  Frame Processor threw an error: Failed to load Frame Processor Plugin "frameToBase64"!
```

### Reproduceable Code

```tsx
import { frameToBase64 } from 'vision-camera-trustee-base64'
...

const frameProcessor = useFrameProcessor((frame) => {
  'worklet'
  const imageAsBase64 = frameToBase64(frame)
}, [])

```
```

### Relevant log output

```shell
LOG  plugin undefined
ERROR  Frame Processor threw an error: Failed to load Frame Processor Plugin "frameToBase64"!
```

### Camera Device

```json
camera
```

### Device

iPhone 11 Pro

### VisionCamera Version

3.5.0

### Can you reproduce this issue in the VisionCamera Example app?

Yes, I can reproduce the same issue in the Example app here

### Additional information

- [ ] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2023-10-19:
Hey! This might be due to the breaking change in 3.5.0, the function is now called `initFrameProcessorPlugin` instead of `getFrameProcessorPlugin`. But the old one should still work (it should just warn), so not sure where that error is coming from.

I googled vision-camera-trustee-base64 and I found this one: https://github.com/yasintorun/vision-camera-base64/blob/main/src/index.tsx - that's for VisionCamera V2, not for V3. That package needs to be updated so it works in V3

ShchipskiiYura, 2023-10-19:
> Hey! This might be due to the breaking change in 3.5.0, the function is now called `initFrameProcessorPlugin` instead of `getFrameProcessorPlugin`. But the old one should still work (it should just [...]

I try update this package for my project

Okay, I try again with ```getFrameProcessorPlugin```

mrousavy, 2023-10-19:
`initFrameProcessorPlugin` not `getFrameProcessorPlugin`

closed 2023-10-19 by mrousavy

edritech93, 2023-10-26:
I have same issue,

index.tsx
```
const plugin = VisionCameraProxy.initFrameProcessorPlugin('scanFace');

export function scanFaces(frame: Frame): Face[] {
  'worklet';
  if (plugin == null) {
    throw new Error('Failed to load Frame Processor Plugin!');
  }
  return plugin.call(frame) as unknown as Face[];
}
```

VisionCameraFaceDetector.m
```
#import <VisionCamera/FrameProcessorPlugin.h>
#import <VisionCamera/FrameProcessorPluginRegistry.h>

#if __has_include("VisionCameraFaceDetector/VisionCameraFaceDetector-Swift.h")
#import "VisionCameraFaceDetector/VisionCameraFaceDetector-Swift.h"
#else
#import "VisionCameraFaceDetector-Swift.h"
#endif

VISION_EXPORT_SWIFT_FRAME_PROCESSOR(VisionCameraFaceDetectorPlugin, scanFace)
```

I use "react-native-vision-camera": "^3.6.4",