margelo/react-native-vision-camera #2047: iOS: App Crash while using camera and frame processor in production🐛

C3000, 2023-10-19:
### What's happening?

 have encountered an issue where my app crashes when installed via TestFlight, but it works fine when I install it directly from Xcode. The crash occurs as soon as I open the view that contains the camera.

### Reproduceable Code

```tsx
const device = useCameraDevice('back');

const frameProcessor = useFrameProcessor((frame) => {
        'worklet'
        const code = '1234';
        if (code && code.format == '13') {
            onBarCodeInternal(code.code);
        }
    }, []);

<Camera
                        frameProcessor={frameProcessor}
                        style={StyleSheet.absoluteFill}
                        device={device}
                        isActive={true}
                    />
```

### Relevant log output

```shell
Incident Identifier: 32616ED6-D5AA-4C47-99B0-021DC29957BC
Hardware Model:      iPhone16,2
Process:             myMzv [13886]
Path:                /private/var/containers/Bundle/Application/CA62AA82-8878-450B-9B49-C8B5D3137539/myMzv.app/myMzv
Identifier:          de.mzv.myMzv-App
Version:             1.5 (30)
AppStoreTools:       15A240a
AppVariant:          1:iPhone16,2:17
Beta:                YES
Code Type:           ARM-64 (Native)
Role:                Foreground
Parent Process:      launchd [1]
Coalition:           de.mzv.myMzv-App [2259]

Date/Time:           2023-10-19 06:12:58.7823 +0200
Launch Time:         2023-10-19 06:12:39.5620 +0200
OS Version:          iPhone OS 17.0.3 (21A360)
Release Type:        User
Baseband Version:    1.00.03
Report Version:      104

Exception Type:  EXC_CRASH (SIGABRT)
Exception Codes: 0x0000000000000000, 0x0000000000000000
Termination Reason: SIGNAL 6 Abort trap: 6
Terminating Process: myMzv [13886]

Triggered by Thread:  5

Thread 0 name:
Thread 0:
0   libsystem_kernel.dylib        	0x00000001e8f201d8 mach_msg2_trap + 8
1   libsystem_kernel.dylib        	0x00000001e8f1ff70 mach_msg2_internal + 80 (mach_msg.c:201)
2   libsystem_kernel.dylib        	0x00000001e8f1fe88 mach_msg_overwrite + 436 (mach_msg.c:0)
3   libsystem_kernel.dylib        	0x00000001e8f1fcc8 mach_msg + 24 (mach_msg.c:323)
4   CoreFoundation                	0x00000001a23774bc __CFRunLoopServiceMachPort + 160 (CFRunLoop.c:2624)
5   CoreFoundation                	0x00000001a23753b4 __CFRunLoopRun + 1208 (CFRunLoop.c:3007)
6   CoreFoundation                	0x00000001a2374e18 CFRunLoopRunSpecific + 608 (CFRunLoop.c:3420)
7   GraphicsServices              	0x00000001e4e395ec GSEventRunModal + 164 (GSEvent.c:2196)
8   UIKitCore                     	0x00000001a47832fc -[UIApplication _run] + 888 (UIApplication.m:3690)
9   UIKitCore                     	0x00000001a4782938 UIApplicationMain + 340 (UIApplication.m:5275)
10  myMzv                         	0x0000000102e44278 main + 80 (main.m:7)
11  dyld                          	0x00000001c4b5fd44 start + 2104 (dyldMain.cpp:1269)
[... 230 lines cut ...]
  "neutralZoom": 1,
  "supportsRawCapture": false,
  "isMultiCam": false
}
```

### Device

iPhone XR - 17.0.3 / iPhone 15 Max Pro - 17.0.3

### VisionCamera Version

3.5.0

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [ ] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2023-10-19:
Hey - can you try two things for me?

1. Try just removing the `useFrameProcessor` and `frameProcessor={..}` from your code and see if that crashes
2. Try completely removing the Frame Processor **and** the react-native-worklets-core dependency, and see if that crashes

Thanks.

C3000, 2023-10-19:
However, when I comment out the frameProcessor code or remove the react-native-worklets-core package, the app functions properly.

`const frameProcessor = useFrameProcessor((frame) => {
    'worklet'
    const code = scanCode(frame)
    if (code && (code.format == '13+2' || code.format == '13+5')) {
        onBarCodeInternal(code.code);
    }
}, []);

<Camera
    // frameProcessor={frameProcessor}
    style={styles.preview}
    device={device}
    isActive={true}
/>`

tristanheilman, 2023-10-23:
I believe this same issue is occurring with some of our production users as well. Reported issue by users on `iPhone 15 Max Pro - 17.0.3`

_Utilizing_
react-native `0.68.5`
react-native-reaniamted `2.15.0`
react-native-vision-camera `2.15.2`

Also the component causing crash is utilizing the `useScanBarcodes` hook instead of `useFrameProcessor`

I'm trying to get logs shared from users experiencing this but that has been a struggle. I will share my error logs once/if I can get them.

bglgwyng, 2023-10-24:
I think I'm facing a similar problem. It's happening in production only.
[Here](https://gist.github.com/bglgwyng/9e319f1283234293ef2b11930bd43768) is the crash log from Sentry.
I haven't observed this behavior in development mode and don't know how to reproduce it.

jamiecjm, 2023-10-26:
I have the same issue but in development

C3000, 2023-11-02:
I was on vacation for the past few days, and I only now could turn my attention back to the issue. After updating both react-native-camera-vision and react-native-worklets-core to the latest version, the FrameProcessor is now functioning in the production mode. As far as I'm concerned, the bug is fixed. Thank you for your work.

closed 2023-11-02 by C3000

mrousavy, 2023-11-07:
Awesome, glad to hear everything's working fine. Thanks!