margelo/react-native-vision-camera #2050: 🐛 Torch flashes for a moment after update to 3.4.0-3.5.1 (iOS)

metrix-hu, 2023-10-19:
### What's happening?

I tried to preview image using torch='on' and at the beginning it flashes for a moment, and after a sec it turns on properly.

### Reproduceable Code

```tsx
const camera = useRef<Camera>(null);
const device = useCameraDevice(props.facing);
const format = useCameraFormat(device, Templates.FrameProcessingYUV);

<Camera
  ref={camera}
  style={styles.container}
  device={device}
  isActive={isActive}
  enableZoomGesture={true}
  zoom={props.zoom ?? 1}
  torch={'on'}
  format={format}
  pixelFormat='yuv'
  photo={true}
/>
```

### Relevant log output

```shell
No relevant outputs.
```

### Camera Device

```json
{
  "maxZoom": 16,
  "hasFlash": true,
  "name": "Back Camera",
  "position": "back",
  "hasTorch": true,
  "supportsRawCapture": false,
  "supportsFocus": true,
  "physicalDevices": [
    "wide-angle-camera"
  ],
  "neutralZoom": 1,
  "minZoom": 1,
  "hardwareLevel": "full",
  "id": "com.apple.avfoundation.avcapturedevice.built-in_video:0",
  "supportsLowLightBoost": false,
  "sensorOrientation": "landscape-left",
  "isMultiCam": false
}
```

### Device

iPhone XR (iOS 16.6)

### VisionCamera Version

3.5.1

### Can you reproduce this issue in the VisionCamera Example app?

Yes, I can reproduce the same issue in the Example app here

### Additional information

- [ ] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

metrix-hu, 2023-10-19:
One more thing I noticed:

It only works like this when the value of torch is constant
And this way the torch stays on even after deactivating the camera
if I add it like this:

torch={props.torch && isActive ? 'on' : 'off'}

then it works perfectly

metrix-hu, 2023-10-19:
https://github.com/mrousavy/react-native-vision-camera/assets/12185646/da42e678-895f-4b70-8ea1-8eecc98195a9

mrousavy, 2023-10-20:
thanks I'll look into this soon!

mrousavy, 2024-01-15:
Hey! I think this is fixed in the latest VisionCamera version :)

closed 2024-01-15 by mrousavy