margelo/react-native-vision-camera #2054: 🐛 App crash when open camera

malkawi00, 2023-10-20:
### What's happening?

App crash after open camera , I tried this solution put it not work https://github.com/mrousavy/react-native-vision-camera/issues/1776#issuecomment-1757204827

### Reproduceable Code

```tsx
import React, { useState } from 'react';
import { Text, View, StyleSheet, TouchableOpacity } from 'react-native';
import { Camera, runAsync, useCameraDevice, useCameraDevices, useFrameProcessor } from 'react-native-vision-camera';
import { documentRecognition } from './src/Helpers';

 const App = () => {

   const [camera, setCamera] = useState(false)
  const frameProcessor = useFrameProcessor((frame) => {
    console.log("I'm running synchronously at 60 FPS!")
    const faces = documentRecognition(frame)
    console.log("I'm running asynchronously, possibly at a lower FPS rate!")
   
  }, [])

  const devices = useCameraDevice('back')

  const device = devices

  if (device == null) {
    return <View />;
  }
  return (
    <View style={{flex: 1}}>
      <TouchableOpacity title={'Open Camera'} onPress={() =>setCamera(true)} style={{backgroundColor: 'red', marginTop: 100, padding: 100, zIndex: 2}}>
        <Text>Open</Text>
      </TouchableOpacity>
     {camera && <Camera
        frameProcessor={frameProcessor}
        style={StyleSheet.absoluteFill}
        isActive={camera}
        device={device}
        torch={'on'}
        pixelFormat={'yuv'}
      />}
    </View>
  )
}

export default App;

=========
native code
package  com.camerav3.frameprocessor;

import androidx.annotation.NonNull;
import androidx.annotation.Nullable;
import com.mrousavy.camera.frameprocessor.Frame;
import com.mrousavy.camera.frameprocessor.FrameProcessorPlugin;

import java.util.Map;

public class documentRecognitionFrameProcessorPlugin extends FrameProcessorPlugin {
    @Nullable
    @Override
    public Object callback(@NonNull Frame frame, @Nullable Map<String, Object> arguments) {
        // code goes here
        return "tset";
    }
}
```

### Relevant log output

```shell
FATAL EXCEPTION: mrousavy/VisionCamera.video
Process: com.tahaluf.idify, PID: 21705
com.facebook.jni.CppException: Compiling JS failed: 1:1:invalid empty parentheses '( )' Buffer size 3 starts with: 280a29 and has protection mode(s): rw-p
at com.mrousavy.camera.frameprocessor.FrameProcessor.call(Native Method)
at com.mrousavy.camera.core.VideoPipeline.init$lambda$0(VideoPipeline.kt:105)
at com.mrousavy.camera.core.VideoPipeline.$r8$lambda$EMriwS_FKhtjTFtppM84Z5V8tiI(Unknown Source:0)
at com.mrousavy.camera.core.VideoPipeline$$ExternalSyntheticLambda0.onImageAvailable(Unknown Source:2)
at android.media.ImageReader$ListenerHandler.handleMessage(ImageReader.java:798)
at android.os.Handler.dispatchMessage(Handler.java:110)
at android.os.Looper.loop(Looper.java:219)
at android.os.HandlerThread.run(HandlerThread.java:67)
```

### Camera Device

```json
"dependencies": {
    "react": "18.2.0",
    "react-native": "0.72.6",
    "react-native-reanimated": "^3.5.4",
    "react-native-vision-camera": "^3.4.0",
    "react-native-worklets-core": "^0.2.0"
  },
```

### Device

Samsung s23

### VisionCamera Version

3.4.0

### Can you reproduce this issue in the VisionCamera Example app?

No, I cannot reproduce the issue in the Example app

### Additional information

- [ ] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

luciancaetano, 2023-10-20:
Same here, app just creshes when use frame processor

celik75, 2023-10-20:
can you update react-native-vision-camera 3.5.1 and test again?

malkawi00, 2023-10-20:
@celik75  still the same error 
FATAL EXCEPTION: mrousavy/VisionCamera.video
 Process: com.tahaluf.idify, PID: 8810
com.facebook.jni.CppException: Compiling JS failed: 1:1:invalid empty parentheses '( )' Buffer size 3 starts with: 280a29

rafaelmotta, 2023-10-20:
I could fix this problem by updating react-native-worklets to the latest version (which is not published on npm yet):

```
"react-native-worklets-core": "https://github.com/margelo/react-native-worklets-core.git#800cf1b2cc65704cf5dfbc725aef1ac1dff32481",
```

fahmiabdulyakub, 2023-10-21:
> I could fix this problem by updating react-native-worklets to the latest version (which is not published on npm yet):
> 
> ```
> "react-native-worklets-core": "https://github.com/margelo/react-native-worklets-core.git#800cf1b2cc65704cf5dfbc725aef1ac1dff32481",
> ```

can you please share your patch @rafaelmotta  ?

malkawi00, 2023-10-23:
hello @rafaelmotta ,
 what the version number ?

xulihang, 2023-10-24:
Still facing this error after using the worklets-core from GitHub:

```
FATAL EXCEPTION: mrousavy/VisionCamera.video
                                                                                                    Process: com.visioncameradynamsoftbarcodereaderexample, PID: 10080
                                                                                                    com.facebook.jni.CppException: Compiling JS failed: 1:1:invalid empty parentheses '( )' Buffer size 3 starts with: 280a29
                                                                                                    	at com.mrousavy.camera.frameprocessor.FrameProcessor.call(Native Method)
                                                                                                    	at com.mrousavy.camera.core.VideoPipeline._init_$lambda$0(VideoPipeline.kt:113)
                                                                                                    	at com.mrousavy.camera.core.VideoPipeline.$r8$lambda$EMriwS_FKhtjTFtppM84Z5V8tiI(Unknown Source:0)
                                                                                                    	at com.mrousavy.camera.core.VideoPipeline$$ExternalSyntheticLambda0.onImageAvailable(Unknown Source:2)
                                                                                                    	at android.media.ImageReader$1.run(ImageReader.java:837)
                                                                                                    	at android.os.Handler.handleCallback(Handler.java:938)
                                                                                                    	at android.os.Handler.dispatchMessage(Handler.java:99)
                                                                                                    	at android.os.Looper.loopOnce(Looper.java:233)
                                                                                                    	at android.os.Looper.loop(Looper.java:344)
                                                                                                    	at android.os.HandlerThread.run(HandlerThread.java:67)
```

xulihang, 2023-10-24:
> Still facing this error after using the worklets-core from GitHub:
>
> ```
> FATAL EXCEPTION: [...]

This error goes away. I am facing a new `system/frame-processors-unavailable` error: https://github.com/mrousavy/react-native-vision-camera/issues/2066

Gruak, 2023-10-24:
Looking at your feedback and my observations after doing some debugging my theory is that [react-native-worklets-core](https://github.com/margelo/react-native-worklets-core): 0.2.0 is compatible with [react-native-reanimated](https://github.com/software-mansion/react-native-reanimated): < 3.5 and [react-native-worklets-core](https://github.com/margelo/react-native-worklets-core):  >= 0.2.1 is compatible with [react-native-reanimated](https://github.com/software-mansion/react-native-reanimated): >= 3.5, althought I might be wrong.

At least I can say that it works for me in these configurations.

mrousavy, 2023-10-24:
Yup, you need the latest versions. We will try to make Worklets Core compatible with more versions of Reanimated, we'll need to talk to the Reanimated team to figure out a good solutoin.

closed 2023-10-24 by mrousavy

paulobunga, 2023-11-20:
Just experienced this issue right now. It only happens when I add frame processor. When I remove it everything works well. However I need the frame processor.

Using 
```json
"react-native-vision-camera": "3.6.6",
"react-native-worklets-core": "^0.2.4",