margelo/react-native-vision-camera #2055: When change the orientation of app the camera rotated to 90deg more 🐛

HarshaUdaraSandaru, 2023-10-20:
### What's happening?

I need to get capture proper orientated image but it not work always. look like every orientation not working.  

But when I used 2.15.2 version there is no problem at all but there is a some performance issue because  I think it would be a new ios updates

[image]
[image]

### Reproduceable Code

```tsx
let device = useCameraDevice("back", [
    {
      physicalDevices: ["ultra-wide-angle-camera"],
    }
  ]); 

 const format = useCameraFormat(device, [
    { fps: 240 },
    { photoHDR: true },
    { photoResolution: "max" }
  ]);

<Modal visible supportedOrientations={["portrait", "landscape"]}>
<ReanimatedCamera
            ref={camera}
            fps={fps}
            isMultiCam
            supportsFocus
            style={styles.mainCamera}
            device={device}
            enableDepthData
            photo={true}
            enablePortraitEffectsMatteDelivery
            lowLightBoost={device.supportsLowLightBoost}
            isActive={true}
            enableZoomGesture={false}
            orientation="portrait"
            focus={touchCode}
            animatedProps={animatedProps}
            enableHighQualityPhotos={true}
            format={format}
          >
            <View style={styles.capture_view}>
              <TouchableOpacity
                onPress={() => takePicture(camera)}
                style={styles.capture}
              />
            </View>
          </ReanimatedCamera>
</Modal>
```

### Relevant log output

```shell
height: 1080
thumbnail: null
isRawPhoto: false
width: 1920
isMirrored: false
path: "/private/var/mobile/Containers/Data/Application/B685F597-8B3F-45BD-BFF7-E08EC2DCF369/tmp/ReactNative/967984AE-F0BF-49B0-B921-44CF936D9DBB.jpeg"
orientation: "landscape-right"
metadata: {DPIWidth: 72, Orientation: 6, {Exif}: {…}, DPIHeight: 72, {TIFF}: {…}, …}

VisionCamera.configureDevice(configuration:): Configuring Camera com.apple.avfoundation.avcapturedevice.built-in_video:0...
VisionCamera.configure(_:): Committed AudioSession configuration!
VisionCamera.configureDevice(configuration:): Successfully configured Input Device!
VisionCamera.configureOutputs(configuration:): Configuring Outputs...
VisionCamera.configureOutputs(configuration:): Adding Photo output...
VisionCamera.configureOutputs(configuration:): Successfully configured all outputs!
VisionCamera.onSessionInitialized(): Camera initialized!
VisionCamera.withSessionLock(_:): Committed CameraSession configuration!
VisionCamera.withDeviceLock(_:): Beginning CaptureDevice configuration...
VisionCamera.configureFormat(configuration:device:): Configuring Format (1920x1080 | 1920x1080@240.0 (ISO: 18.0..720.0, Pixel Formats: [VisionCamera.PixelFormat.yuv, VisionCamera.PixelFormat.yuv, VisionCamera.PixelFormat.rgb, VisionCamera.PixelFormat.yuv, VisionCamera.PixelFormat.yuv, VisionCamera.PixelFormat.rgb]))...
VisionCamera.configureFormat(configuration:device:): Successfully configured Format!
VisionCamera.withDeviceLock(_:): Committed CaptureDevice configuration!
VisionCamera.withDeviceLock(_:): Beginning CaptureDevice configuration...
VisionCamera.withDeviceLock(_:): Committed CaptureDevice configuration!
BUG IN CLIENT OF UIKIT: Setting UIDevice.orientation is not supported. Please use UIWindowScene.requestGeometryUpdate(_:)
```

### Camera Device

```json
isMultiCam: false
formats: (48) [{…}, {…}, {…}, {…}, {…}, {…}, {…}, {…}, {…}, {…}, {…}, {…}, {…}, {…}, {…}, {…}, {…}, {…}, {…}, {…}, {…}, {…}, {…}, {…}, {…}, {…}, {…}, {…}, {…}, {…}, {…}, {…}, {…}, {…}, {…}, {…}, {…}, {…}, {…}, {…}, {…}, {…}, {…}, {…}, {…}, {…}, {…}, {…}]
name: "Back Camera"
hasFlash: true
id: "com.apple.avfoundation.avcapturedevice.built-in_video:0"
minZoom: 1
neutralZoom: 1
supportsFocus: true
supportsRawCapture: false
physicalDevices: ["wide-angle-camera"]
maxZoom: 67.5
supportsLowLightBoost: false
sensorOrientation: "landscape-right"
hardwareLevel: "full"
hasTorch: true
position: "back"
```

### Device

iPad pro 11-inch (3rd generation)

### VisionCamera Version

3.5.0 and 3.5.1

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2023-10-20:
Orientation is not yet fully implemented. This is the first pinned issue in this repo. https://github.com/mrousavy/react-native-vision-camera/issues/1891

Also;

[image]
🙃

closed 2023-10-20 by mrousavy

HarshaUdaraSandaru, 2023-10-20:
Then Why are u guys closing the ticket?

mrousavy, 2023-10-20:
"u guys"? I'm just one guy my friend.

Closing it because it is a **duplicate**

HarshaUdaraSandaru, 2023-10-20:
Okay. How do I get the update after u implemented the orientation properly? Does this get much more time to fix?

HarshaUdaraSandaru, 2023-10-20:
https://github.com/mrousavy/react-native-vision-camera/issues/1891 
Found. 
I also try to change the swift code. but it did not give any results. But when comes to the old version (2.15.2) which I used once it works properly. 
Did you compare that one? 
I'm not diving a lot more into that but maybe there is something can find my friend.