margelo/react-native-vision-camera #2056: ❓ Reload the JS app while using Frame Processors

ldstein, 2023-10-20:
### Question

Reloading the JS bundle during development crashes the app I'm debugging.

It appears to be caused by the JS bundle re-initializing a FrameProcessor with the same id:

Javascript:
```js
const plugin_detectQr = VisionCameraProxy.getFrameProcessorPlugin('detectQr');
```

Logcat:
```
FATAL EXCEPTION: create_react_context
Process: com.meshmobile, PID: 21044
java.lang.AssertionError: Tried to add a Frame Processor Plugin with a name that already exists! Either choose unique names, or remove the unused plugin, at com.mrousavy.camera.frameprocessor.FrameProcessorPluginRegistry.addFrameProcessorPlugin(FrameProcessorPluginRegistry.java:17)
```

Is there any way to avoid this crash while reloading during development?

*Edit* - Just discovered PR #2038. Was this intended to solve this sort of issue?

```
const plugin_detectQr = VisionCameraProxy.getFrameProcessorPlugin(pluginId) || VisionCameraProxy.initFrameProcessorPlugin(pluginId);
```

### What I tried

- "reload" in metro
- "reload" by shaking the phone and  selecting "reload" from the menu

### VisionCamera Version

3.4.0

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

mrousavy, 2023-10-20:
Hey! This probably should've been a bug report instead of a question.

Anyways, I added a new way to register Frame Processor Plugins in 3.5.0 - those native plugins need to adapt that new API.

Check out the docs:

[image]
[image]

Basically they are only registered statically once. That fixes your error.

closed 2023-10-20 by mrousavy

ldstein, 2023-10-20:
Thanks @mrousavy , the static register did the trick.

mrousavy, 2023-10-20:
great!

xulihang, 2023-10-24:
I need to pass `reactContext` or an instance of a native module to a frame processor plugin. Is this still possible using the static register?

mrousavy, 2023-10-24:
No, not as of today. I could however pass context along in the initializer func if that helps you @xulihang ?

mrousavy, 2023-10-24:
See this PR @xulihang https://github.com/mrousavy/react-native-vision-camera/pull/2067

xulihang, 2023-10-24:
This is okay. I need to use `context` to access the assets folder.

As for passing the native module, I can try another approach.