margelo/react-native-vision-camera #2058: Android devices multi-cams support❓

lazarevand1997, 2023-10-20:
### Question

From 3.0.0 lib supports Multi-cams for android. Could you please tell what devices better to use to test this? 

### What I tried

I tried One Plus 7T, it has a few cameras, but there is only one single back camera device.

### VisionCamera Version

3.4.1

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

zzz08900, 2023-10-21:
`but there is only one single back camera device.`

I feel like there's something wrong with your code. How did you get your `device` prop for camera view?

lazarevand1997, 2023-10-23:
I use `getAvailableCameraDevices` to get all devices. And now it returns 2 devices, one front and one single back.

mrousavy, 2023-10-24:
Hi.

VisionCamera has a very solid Devices API that is capable of detecting all Camera devices on your phone that are available for use in your app. Even USB Cameras. If there's a device missing in that list, it is pretty likely that this device is simply not available for use, and the vendor (Samsung, Huawei, One Plus, ...) just locked this and doesn't allow you to use this specific Camera device. In other words; if VisionCamera can't detect the Camera, there's nothing you can do about it. It's not even available in a native Android app (except for their stock Samsung/Huawei/... Camera app).

To confirm this, download the official Google example for Camera apps; [camera-samples/Camera2Video](https://github.com/android/camera-samples/tree/main/Camera2Video) and run it on your phone. If the Camera device is not in there either, then again, nothing you can do.

I'm sure that for specific devices such as telephoto or ultra-wide, they usually don't allow you to use those devices directly, but only in combination with other devices (e.g. a "dual-camera" consisting of a normal wide-angle and the ultra-wide-angle), because they only work like that. 

Try logging all devices you get:

```js
const devices = Camera.getAvailableCameraDevices()
devices.forEach((d) => {
  d.formats = null
})
console.log(JSON.stringify(devices, null, 2))
```

And if there's only one device on the back there then that's the only device third party apps can use. That's OnePlus' fault then.

mrousavy, 2023-10-24:
See [Multi-camera API](https://developer.android.com/training/camera2/multi-camera) and [Multi-Camera Hardware Support](https://source.android.com/docs/core/camera/multi-camera) documentation for more information.

closed 2023-10-24 by mrousavy

mrousavy, 2023-10-24:
Actually maybe this fixes this: https://github.com/mrousavy/react-native-vision-camera/pull/2073

mrousavy, 2023-10-24:
Try 3.6.3 and let me know if the multi-cam now shows up for you. If you appreciate my efforts & dedication to VisionCamera, please [💖 consider sponsoring me on GitHub 💖](https://github.com/sponsors/mrousavy) to say thanks. :)

Asadaial, 2024-11-29:
@mrousavy  how can we use front and back cameras at a time when the back camera is open and we want to show our face as well with front camera? I am unable to use it in that way and no documentation is helpful in this