margelo/react-native-vision-camera #2059: 🐛 [V3][iOS]Frame Processor threw an error: Failed to load Frame Processor Plugin "functionName"

ShchipskiiYura, 2023-10-20:
### What's happening?

I'm testing [react-native-face-detector](https://github.com/rodgomesc/vision-camera-face-detector) plugin on V3.
I can convert the plugin on new V3 version, but there is an error ```ERROR  Frame Processor threw an error: Failed to load Frame Processor Plugin```

### Reproduceable Code

```tsx
.m

#if __has_include(<VisionCamera/FrameProcessorPlugin.h>)
#import <VisionCamera/FrameProcessorPlugin.h>
#import <VisionCamera/FrameProcessorPluginRegistry.h>

#import "VisionCameraFaceDetector-Swift.h"

VISION_EXPORT_SWIFT_FRAME_PROCESSOR(VisionCameraFaceDetector, detectFaces)

#endif

```VisionCameraFaceDetector.swift```

import Foundation
import VisionCamera
import MLKitFaceDetection
import MLKitVision
import CoreML

import UIKit
import AVFoundation

@objc(VisionCameraFaceDetector)
public class VisionCameraFaceDetector: FrameProcessorPlugin {
    public override init(options: [AnyHashable : Any]! = [:]) {
      super.init(options: options)
    }
...

public override func callback(_ frame: Frame, withArguments arguments: [AnyHashable: Any]?) -> Any {
return array
}
}

```VisionCameraFaceDetector-Bridging-Header.h```

#if VISION_CAMERA_ENABLE_FRAME_PROCESSORS
#import <VisionCamera/FrameProcessorPlugin.h>
#import <VisionCamera/Frame.h>
#endif

```index.tsx```

import { VisionCameraProxy, type Frame } from 'react-native-vision-camera';

const plugin = VisionCameraProxy.initFrameProcessorPlugin('detectFaces');

export function scanFaces(frame: Frame, options?: FaceDetectionOptions): Face {
  'worklet';
  console.log('plugin', plugin);
  // if (plugin == null) {
  //   throw new Error('Failed to load Frame Processor Plugin "detectFaces"!');
  // }
  // @ts-ignore
  return plugin?.call(frame, options);
}
```

### Relevant log output

```shell
LOG  plugin undefined
ERROR  Frame Processor threw an error: Failed to load Frame Processor Plugi
```

### Camera Device

```json
any
```

### Device

iPhone 11 Pro

### VisionCamera Version

3.5.1

### Can you reproduce this issue in the VisionCamera Example app?

Yes, I can reproduce the same issue in the Example app here

### Additional information

- [ ] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

erickcrus, 2023-10-20:
same here with `vision-camera-code-scanner` plugin:
[image]

mrousavy, 2023-10-24:
Hey.

I really don't get why people don't do this, but in the issue template it tells you the exact steps on how you can add the native logs here. Without the native logs I can't help you.

closed 2023-10-24 by mrousavy

ShchipskiiYura, 2023-11-04:
> Hey.
> 
> I really don't get why people don't do this, but in the issue template it tells you the exact steps on how you can add the native logs here. Without the native logs I can't help you.

I don't have other native logs, only these

2023-11-04 13:28:56.258147+0200 project[9185:440604] VisionCameraProxy: Creating Worklet Context...
2023-11-04 13:28:56.259055+0200 project[9185:440604] VisionCameraProxy: Worklet Context Created!
2023-11-04 13:30:45.960142+0200 project[9185:440277] [native] VisionCamera.didSetProps(_:): Updating 14 props: [isActive, right, orientation, onViewReady, cameraId, position, left, onError, onInitialized, enableBufferCompression, onCodeScanned, bottom, enableFrameProcessor, top]
2023-11-04 13:30:45.960178+0200 project[9185:440277] [native] VisionCamera.configure(_:): Updating Session Configuration...
2023-11-04 13:30:45.960326+0200 project[9185:441509] [native] VisionCamera.configure(_:): Beginning AudioSession configuration...
2023-11-04 13:30:45.960326+0200 project[9185:441554] [native] VisionCamera.withSessionLock(_:): Beginning CameraSession configuration...
2023-11-04 13:30:45.960385+0200 project[9185:441554] [native] VisionCamera.configureDevice(configuration:): Configuring Input Device...
2023-11-04 13:30:45.960521+0200 project[9185:441554] [native] VisionCamera.configureDevice(configuration:): Configuring Camera com.apple.avfoundation.avcapturedevice.built-in_video:1...
2023-11-04 13:30:45.960761+0200 project[9185:441509] [native] VisionCamera.configureAudioSession(configuration:): Configuring Audio Session...
2023-11-04 13:30:45.962093+0200 project[9185:441509] [native] VisionCamera.configure(_:): Committed AudioSession configuration!
2023-11-04 13:30:45.985650+0200 project[9185:441554] [native] VisionCamera.configureDevice(configuration:): Successfully configured Input Device!
2023-11-04 13:30:45.985669+0200 project[9185:441554] [native] VisionCamera.configureOutputs(configuration:): Configuring Outputs...
2023-11-04 13:30:45.985682+0200 project[9185:441554] [native] VisionCamera.configureOutputs(configuration:): Adding Video Data output...
2023-11-04 13:30:45.988451+0200 project[9185:441554] [native] VisionCamera.configureOutputs(configuration:): Successfully configured all outputs!
2023-11-04 13:30:45.991218+0200 project[9185:441554] [native] VisionCamera.withSessionLock(_:): Committed CameraSession configuration!
2023-11-04 13:30:45.991256+0200 project[9185:441554] [native] VisionCamera.withDeviceLock(_:): Beginning CaptureDevice configuration...
2023-11-04 13:30:45.991601+0200 project[9185:441554] [native] VisionCamera.withDeviceLock(_:): Committed CaptureDevice configuration!
2023-11-04 13:30:46.381609+0200 project[9185:441554] [native] VisionCamera.withDeviceLock(_:): Beginning CaptureDevice configuration...
2023-11-04 13:30:46.381682+0200 project[9185:441554] [native] VisionCamera.withDeviceLock(_:): Committed CaptureDevice configuration!
2023-11-04 13:30:46.381694+0200 project[9185:441554] [native] VisionCamera.onSessionInitialized(): Camera initialized!

2023-11-04 13:30:46.396802+0200 project[9185:440604] [javascript] Frame Processor threw an error: Failed to load Frame Processor Plugin 'scanFaces'!

Fannolo, 2023-12-04:
what I find to work is to change the implementation file of the plugin to 

Instead of 
```objc
VISION_EXPORT_SWIFT_FRAME_PROCESSOR(VisionCameraFaceDetector, detectFaces)
```
to
```objc
@interface VisionCameraFaceDetector (FrameProcessorLoader)
@end

@implementation VisionCameraFaceDetector (FrameProcessorLoader)
+ (void)load
{
    [FrameProcessorPluginRegistry addFrameProcessorPlugin:@"scanFaces" withInitializer:^FrameProcessorPlugin* (NSDictionary* options) {
        {
            return [[VisionCameraFaceDetector alloc] initWithOptions:nil];
        }
    }];
}
@end
```

iBotPeaches, 2023-12-06:
> So that's using `+ load` again. Not sure if that is actually supported very well... As of right now, this is what the macro does:

Yeah those using `+load` and having success must be staying on some older Xcodes. As when you upgrade Xcode - that no longer works. Creates some real confusion in these issue reports when folks just post - "this random blob works for me, with no context"

Only thing that blob above looks like its missing is the protocol segment. Haven't bite the bullet yet to v3, so not sure the contract class to align with that snippet - https://github.com/mrousavy/react-native-vision-camera/pull/2053/files#diff-338f5b4148229eb64e3d9d9e0be159597f466b70f44983274e1456063d91085bR52