margelo/react-native-vision-camera #2061: ❓ Vision camera is view difference than default camera

Kimdnel, 2023-10-21:
### Question

[image]

hi there

I am using react-native-vision-camera v3 with react-native 0.72.6. While comparing the default camera view and vision-camera view it slightly zoomed than the default camera. both same 1x

my code
```
const TestCam = () => {
  const device = useCameraDevice('back');

  if (device == null) return null;

  return (
      <Camera
        style={styles.camera}
        device={device}
        resizeMode={"contain"}
        isActive
        enableZoomGesture />
  );
};

export default TestCam;

const styles = StyleSheet.create({
  camera: {
     ...StyleSheet.absoluteFill
  },
});
```

my question is .. how can i fix it?
I wanna show the same screen as the default camera.

thank you

### VisionCamera Version

v3

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

zzz08900, 2023-10-21:
https://github.com/mrousavy/react-native-vision-camera/issues/1981

Kimdnel, 2023-10-21:
> Hmm, what does your `device` look like? And what's the phone model that those screenshot were taken?

/

It's iphone 13 pro max

ldstein, 2023-10-23:
Have you tried making your view the exact same width and height of the default camera view?

Kimdnel, 2023-10-23:
> Have you tried making your camera view the exact same width and height of the default camera view?

yes i did

Kimdnel, 2023-10-24:
I solved this problem.

mrousavy, 2023-10-24:
1. Check the format docs, there's multiple formats in a device. Select the one that you need.
2. If you "solved this problem", please post a solution for others and close the issue. No need to pollute the issues tab here

closed 2023-10-24 by mrousavy

un1234, 2023-12-11:
> I solved this problem.

Can you please tell me the solution?

Kimdnel, 2023-12-12:
> > I solved this problem.
> 
> Can you please tell me the solution?

Use the "useCameraFormat" api