margelo/react-native-vision-camera #2064: 🐛 iOS itf barcode not working

lucca180, 2023-10-23:
### What's happening?

iOS barcode scanner is not working with 2 of 5 Interleaved (itf) format.

Android works fine.

[image]

### Reproduceable Code

```tsx
const device = useCameraDevice('back');

  const codeScanner = useCodeScanner({
    codeTypes: ['code-128', 'itf'],
    onCodeScanned: codes => {
      console.log(`Scanned ${codes.length} codes!`);
    },
  });
```

### Relevant log output

```shell
N/A
```

### Camera Device

```json
{
  "minZoom": 1,
  "sensorOrientation": "landscape-right",
  "maxZoom": 16,
  "supportsLowLightBoost": false,
  "isMultiCam": false,
  "name": "Back Camera",
  "hasFlash": true,
  "supportsRawCapture": false,
  "supportsFocus": true,
  "neutralZoom": 1,
  "physicalDevices": [
    "wide-angle-camera"
  ],
  "hardwareLevel": "full",
  "position": "back",
  "id": "com.apple.avfoundation.avcapturedevice.built-in_video:0",
  "hasTorch": true
}
```

### Device

iPhone 8

### VisionCamera Version

3.5.1

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

iliapnmrv, 2023-10-23:
duplicate of [bug](https://github.com/mrousavy/react-native-vision-camera/issues/2060)

lucca180, 2023-10-27:
So, on Android, ift reads both interleaved2of5 and itf14 formats. On iOS, per [docs](https://developer.apple.com/documentation/avfoundation/avmetadataobject/objecttype), these are two diff variables, that's why setting 'ift' doesn't work for interleaved2of5 format.

klabusta, 2023-12-21:
I've had the same problem with some of the ITF labels I'm using. I've replaced (i.e. patched) the itf14 with interleaved2of5 in AVMetadataObject.ObjectType+descriptor.swift and they worked. Although that seems like a bad patch, as itf14 wouldn't work anymore.

Any better solution for this?

mrousavy, 2024-01-15:
Hey - I'm using Apple's custom QR code scanner. If it cannot scan that, there's nothing I can do. You can instead write a custom frame processor plugin (or use a community plugin) that uses a different QR model, maybe MLKit or ZXing. Dynamsoft has one as well (cc @xulihang)

Btw; can you try rotating the Camera to see if that changes anything? Maybe the orientation on the scanner is wrong (e.g. landscape left by default), then it is in fact my fault.

closed 2024-01-15 by mrousavy

klabusta, 2024-01-15:
It can scan both: itf14 and interleaved2of5. But you only map itf14 to "ITF" not both.

mrousavy, 2024-01-15:
Ah I see now.
So instead of patching, why didn't you create a PR to fix this for everyone?

This could be a separate branch in that file to make it work :)

klabusta, 2024-01-15:
My patch isn't fixing it. It's just using interleaved2of5 instead of itf14. But I suppose mapping both to "ITF" is a bigger change. I wouldn't know how.

mrousavy, 2024-01-15:
My idea was to add "interleaved2of5" as an additional branch, not to use both under the itf alias.

The downside of that is that the interleaved2of5 type is not supported on Android/MLKit afaik, so it wouldnt be cross platform

mrousavy, 2024-01-15:
https://stackoverflow.com/questions/51135855/ml-kit-barcode-detector-not-recognizing-itf-barcodes-with-4-or-less-digits <-- from 2018

PR #3252 (2024-10-29): fix: Fix mapping of itf and add new itf-14 type in iOS
## What
I can't scan interleaved2of5 (ITF) barcodes in iOS. 

Currently the CodeType `itf` is mapped:
* on android to `Barcode.FORMAT_ITF`, which is the ["barcode format constant for ITF (Interleaved Two-of-Five)"](https://developers.google.com/android/reference/com/google/mlkit/vision/barcode/common/Barcode#FORMAT_ITF)
* on iOS to `AVMetadataObject.ObjectType.itf14`, which is ["a constant that identifies the ITF14 symbology"](https://developer.apple.com/documentation/avfoundation/avmetadataobject/objecttype/1618831-itf14)

ITF-14 is a var [... cut]