margelo/react-native-vision-camera #2065: 🐛 useCameraDevice sometimes doesn't return ultra-wide-angle-camera

stevengoldberg, 2023-10-24:
### What's happening?

When I pass an array containing `'ultra-wide-angle-camera'` along with any other type of device to `useCameraDevice`, the ultra-wide-angle camera doesn't get selected. It only gets selected when I *only* choose `'ultra-wide-angle-camera`'.

If I get all available devices with `useCameraDevices`, I get "Back Triple Camera" in the array.

This worked correctly in one of the earlier 3.0 versions.

### Reproduceable Code

```tsx
const device = useCameraDevice(cameraPosition, { physicalDevices: [
                  'ultra-wide-angle-camera',
                  'wide-angle-camera',
                  'telephoto-camera',
              ] })
console.log(device.physicalDevices) // ['wide-angle-camera', 'telephoto-camera']

// alternatively:

const device = useCameraDevice(cameraPosition, { physicalDevices:  ['ultra-wide-angle-camera'] })

console.log(device.physicalDevices) // ['ultra-wide-angle-camera']
```

### Relevant log output

```shell
N/A
```

### Camera Device

```json
{
  "sensorOrientation": "landscape-left",
  "isMultiCam": true,
  "id": "com.apple.avfoundation.avcapturedevice.built-in_video:3",
  "position": "back",
  "supportsRawCapture": false,
  "neutralZoom": 1,
  "supportsFocus": true,
  "physicalDevices": [
    "wide-angle-camera",
    "telephoto-camera"
  ],
  "hardwareLevel": "full",
  "supportsLowLightBoost": false,
  "minZoom": 1,
  "maxZoom": 189,
  "hasFlash": true,
  "name": "Back Dual Camera",
  "hasTorch": true
}
```

### Device

iPhone 14 Pro

### VisionCamera Version

3.4.0

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [X] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2023-10-24:
Hey, yes this is intended behaviour.

If you pass multiple cameras in there, your intention is probably a multi-cam. It tries to find a device that has the requested devices, or the closest one if not found.

Normally, a wide-angle camera device is preferred, unless you explicitly omit it from that array.

What exactly is the problem here in the scenario? If you want the ultra-wide, just pass ultra-wide as an argument and it works fine.

The code for the filtering is here: https://github.com/mrousavy/react-native-vision-camera/blob/86b3b9e1d73ebee0a1938135616935c7e479b116/package/src/devices/getCameraDevice.ts#L47-L79

Try changing stuff around here and let me know if there's an improvement we can bring into the repo.

stevengoldberg, 2023-10-24:
My intent is to select the triple camera. Previously I was able to do this by passing an array with all three cameras to the filter. Now when I do that, it selects the dual cam instead and I’m unable to zoom to 0.5x.

mrousavy, 2023-10-24:
Does it work if you remove lines 55 to 62 in `getCameraDevice.ts`? 

https://github.com/mrousavy/react-native-vision-camera/blob/86b3b9e1d73ebee0a1938135616935c7e479b116/package/src/devices/getCameraDevice.ts#L55-L62

stevengoldberg, 2023-10-24:
Yep, if I remove those lines I get the triple camera as expected.

mrousavy, 2023-10-24:
Fixed this in https://github.com/mrousavy/react-native-vision-camera/pull/2074 I think :)

mrousavy, 2023-10-24:
Try 3.6.3 and let me know if that fixed it for you. If you appreciate my efforts & dedication to VisionCamera, please [💖 consider sponsoring me on GitHub 💖](https://github.com/sponsors/mrousavy) to say thanks. :)

closed 2023-10-24 by mrousavy

stevengoldberg, 2023-10-24:
That fixes it, thanks!

dorakadri, 2024-12-30:
hello i have a similar problem for example in galaxy Samsung Galaxy a73 5g  it has wide range which is 0,5 but i got the min zoom its 1