margelo/react-native-vision-camera #2076: ❓ How to get cornerPoints of scanned QR

yukta357, 2023-10-24:
### Question

So far, I've been using ExpoBarcodeScanner, which [returns cornerPoints with each result](https://docs.expo.dev/versions/latest/sdk/bar-code-scanner/#barcodescannerresult).

I see that Vision Camera returns a frames object, which contains origin coordinates and box dimensions. However, for our use case, this is insufficient. We need all 4 corner points in order to calculate the QRs position in 3D space.

So how would one get corner points with Vision Camera?

Use a frame-processor and let it run for only a single frame whenever a qr is scanned? Or switch to a frame processor entirely, replacing native scanning on ios? Or am I missing something perhaps? :p

### VisionCamera Version

v3

### Additional information

- [X] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

mrousavy, 2023-10-25:
Hey!

So the CodeScanner API is a simple, separate lightweight output for the Camera.

The Apple docs suggest that there are two values, [`.bounds`](https://developer.apple.com/documentation/avfoundation/avmetadataobject/1386043-bounds) and [`.corners`](https://developer.apple.com/documentation/avfoundation/avmetadatamachinereadablecodeobject/2926566-corners).

Currently I am using `bounds`. Is `corners` what you need?

Same for Android: [`boundingBox`](https://developers.google.com/android/reference/com/google/mlkit/vision/barcode/common/Barcode#getBoundingBox()) and [`cornerPoints`](https://developers.google.com/android/reference/com/google/mlkit/vision/barcode/common/Barcode#getCornerPoints())

yukta357, 2023-10-25:
Yes, indeed, corners is what I need.

mrousavy, 2023-10-25:
Cool, I think this takes me roughly 3 hours to implement. If you want to sponsor the development of this project, consider [sponsoring me on GitHub](https://github.com/sponsors/mrousavy), usually those rates are $100/h 😄

only1chi, 2023-10-25:
@mrousavy 
I have the same problem and need corners points of the QR code like @yukta357.
Can it be derived from the bounds object you provide?

metrix-hu, 2023-10-31:
@mrousavy @yukta357 I will try to do a PR on this.

yukta357, 2023-10-31:
Wow, that's great @metrix-hu!

PR #2117 (2023-11-09): Feat: Implement corner points for scanned codes
## What

  This PR adds corner points to code scanner results, with an additional second parameter of the scanned frame in the codescanner callback which defines a widht, height proparty for scaling.

## Changes
   * This PR makes sure you can access scanned codes corner points in the corners array of the Code object. Every point has an x,y coordinate which uses the same coordinate system as the frame does.
   * This PR also adds a second param of the code scanner callback which contains width, height values for the scanned image frame.

## Teste [... cut]
closed 2023-11-09 by mrousavy