margelo/react-native-vision-camera #2078: 🐛 With 3.6.3, Barcode scanner does recognize barcodes on Android as it did with 3.5.1

zjkuang, 2023-10-24:
### What's happening?

I upgraded this dependency from v2 to v3 recently. With 3.5.1 I had an issue that the scanner camera view was not positioned properly on Android. (With 3.5.1 the scanner could scan a barcode very sensitively.) Later I updated it to 3.6.3 and the UI layout problem is gone on Android, but the scanner could not recognized the barcode anymore.

### Reproduceable Code

```tsx
...
export const SmartCameraView = ({
  onCancel,
  onImageCaptured,
  title,
  errorMessage,
  types = [
    'aztec',
    'codabar',
    'code-128',
    'code-39',
    'code-93',
    'data-matrix',
    'ean-13',
    'ean-8',
    'itf',
    'pdf-417',
    'qr',
    'upc-e',
  ], // Default to All barcode types.
  mode,
}: SmartCameraViewProps) => {
  const devices = useCameraDevices();
  const device = devices.find((d) => d.position === 'back');
  const camera = React.useRef<Camera>(null);
  const isFocused = useIsFocused();
  const isForeground = useIsForeground();
  const isActive = isFocused && isForeground;
  const isBarcodeScan = mode?.scan === ScanType.Barcode;
  const isVinScan = mode?.scan === ScanType.Vin;
  const isScanEnabled = isBarcodeScan || isVinScan;

  const [barcodes, setBarcodes] = React.useState<Barcode[]>();
  const setBarcodesSafely = (detectedBarcodes: Barcode[]) => {
    if (isActive && mounted.value && isScanEnabled) {
      setBarcodes(detectedBarcodes);
    }
  };

  const codeScanner = useCodeScanner({
    codeTypes: types,
    onCodeScanned(codes) {
      if (codes.length) {
        runOnJS(setBarcodesSafely)(codes);
      }
    },
  });

  React.useEffect(() => {
    if (isActive && barcodes?.length && barcodes?.[0].value && isScanEnabled) {
      log.debug('Got Barcodes: ', barcodes);
      onImageCaptured(barcodes?.[0]);
    }
  }, [barcodes, onImageCaptured, isActive, isScanEnabled]);

  const CameraContainer = () => {
    return device ? (
      <Camera
        style={styles.boundingBoxFrame}
        ref={camera}
        device={device}
        isActive={isActive}
        enableZoomGesture={Boolean(mode?.photo)}
        zoom={device?.neutralZoom ?? 1}
        photo={Boolean(mode?.photo)}
        codeScanner={codeScanner}
      />
    ) : (
      <></>
    );
  };

  return (
    <SharedCameraViewContainer
      devices={devices}
      device={device}
      camera={camera}
      frameView={
        <BoundingBoxFrame cameraContainer={CameraContainer} isBarcodeScan={isBarcodeScan} />
      }
      onCancel={onCancel}
      onImageCaptured={onImageCaptured}
      title={title}
      types={types}
      errorMessage={errorMessage}
      mode={mode}
    />
  );
};
```

### Relevant log output

```shell
N/A
```

### Camera Device

```json
A Google Pixel 7 back camera
```

### Device

Google Pixel 7 (Android 13)

### VisionCamera Version

3.6.3

### Can you reproduce this issue in the VisionCamera Example app?

Yes, I can reproduce the same issue in the Example app here

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

grillorafael, 2023-10-24:
I'm having the same issue

notmarinho, 2023-10-25:
Same here, works fine on iOS but it does not recognize on Android.

tronginc, 2023-10-25:
same here on android

mrousavy, 2023-10-25:
[image]

🙃

grillorafael, 2023-10-25:
[log-android.txt](https://github.com/mrousavy/react-native-vision-camera/files/13164558/log-android.txt)
This is a log file I captured using 3.6.3 on a Galaxy Tab S9 FE.

I did a Grep for Camera to see if I captured something more meaningful.

mrousavy, 2023-10-25:
Thanks @grillorafael. Fixed this in #2082

closed 2023-10-25 by mrousavy

mrousavy, 2023-10-25:
Just released that quick fix in 3.6.4. If you appreciate my dedication and effort towards making VisionCamera better, please [💖 consider sponsoring me on GitHub 💖](https://github.com/sponsors/mrousavy/) to say thanks. :)

quocbaooo, 2023-10-26:
> Great and awesome! Just tested on 3.6.4 and it worked like magic. @mrousavy @grillorafael

can you give me reference source code pls

zjkuang, 2023-10-26:
> > Great and awesome! Just tested on 3.6.4 and it worked like magic. @mrousavy @grillorafael
> 
> can you give me reference source code pls

@quocbaooo 
```
  const devices = useCameraDevices();
  const device = devices.find((d) => d.position === 'back');

  const codeScanner = useCodeScanner({
    codeTypes: types,
    onCodeScanned(codes) {
      if (codes.length) {
        // process barcodes
      }
    },
  });

  const CameraContainer = () => {
    return device ? (
      <Camera
        style={styles.boundingBoxFrame}
        ref={camera}
        device={device}
        isActive={isActive}
        enableZoomGesture={Boolean(mode?.photo)}
        zoom={device?.neutralZoom ?? 1}
        photo={Boolean(mode?.photo)}
        codeScanner={codeScanner}
      />
    ) : (
      <></>
    );
  };
```

VampAK1864, 2023-10-26:
> @quocbaooo
>
> ```
>   const devices = useCameraDevices();
>   const device = devices.find((d) => d.position === 'back');
>
>   const codeScanner = useCodeScanner({
>     codeTypes: types, [...]

What are your imports?

ngenerapps, 2023-10-26:
@zjkuang I have an issue with the "useCodeScanner".
TypeError: 0, _$$_REQUIRE(_dependencyMap[11(...)e-vision-camera").useCodeScanner is not a function (it is 
undefined)
Do you have any ideas on how to fix it?
I really appreciate any help you can provide.

zjkuang, 2023-10-30:
> @zjkuang I have an issue with the "useCodeScanner".
> TypeError: 0, _$$_REQUIRE(_dependencyMap[11(...)e-vision-camera").useCodeScanner is not a function (it is
> undefined)
> Do you have any ideas on how to fix it?
> I really appreciate any help you can provide.

Can you paste your code snippet here? @ngenerapps

ngenerapps, 2023-10-30:
@zjkuang 
import React, {useState, useEffect, Component} from 'react';
import {
  SafeAreaView,
  RefreshControl,
  ScrollView,
  StatusBar,
  StyleSheet,
  PermissionsAndroid,
  Pressable,
  Text,
  useColorScheme,
  View,
  TouchableOpacity,
  Image,
  TextInput,
  Linking,
  Button,
  AnimationEffect,
  ReadableStreamReader,
  ActivityIndicator,
} from 'react-native';
import QRCodeScanner from 'react-native-qrcode-scanner';
// import {ViewPropTypes} from 'deprecated-react-native-prop-types';
import {RNCamera} from 'react-native-camera';
import {
  Camera,
  useCameraDevices,
  useCodeScanner,
} from 'react-native-vision-camera';
import {useScanBarcodes, BarcodeFormat} from 'vision-camera-code-scanner';
import PropTypes from 'prop-types';
import Axios from 'axios';
import Modal from 'react-native-modal';
import RNRestart from 'react-native-restart';
import Geolocation from '@react-native-community/geolocation';
import WifiManager, {getCurrentWifiSSID} from 'react-native-wifi-reborn';
import {useNetInfo} from '@react-native-community/netinfo';

export default function Home({navigation}) {
  const devices = useCameraDevices();
  const device = devices.back;
  const types = ['qr'];

  useEffect(() => {
    checkPermission();
  }, []);

  const checkPermission = async () => {
    try {
      const newCameraPermission = await Camera.requestCameraPermission();
      console.log('Camera Permission:', newCameraPermission);
    } catch (error) {
      console.error('Camera Permission Error:', error);
    }
  };

 // const [qrCode, setQRCode] = useState('');

 // const onCodeScanned = event => {
 //   console.log('Scanned QR Code:', event.nativeEvent.codeStringValue);
 //   setQRCode(event.nativeEvent.codeStringValue);
 // };

  const codeScanner = useCodeScanner({
    codeTypes: types,
    onCodeScanned(codes) {
      if (codes.length) {
        // process barcodes
      }
    },
  });

  if (device == null) {
    return (
      <View style={{flex: 1, justifyContent: 'center', alignItems: 'center'}}>
        <ActivityIndicator />
      </View>
    );
  }

  return (
    <SafeAreaView
      style={{
        flex: 1,
        justifyContent: 'center',
        alignItems: 'center',
        backgroundColor: '#000',
      }}>
      <Camera
        style={styles.boundingBoxFrame}
        ref={Camera}
        device={device}
        isActive={isActive}
        enableZoomGesture={Boolean(mode?.photo)}
        zoom={device?.neutralZoom ?? 1}
        photo={Boolean(mode?.photo)}
        codeScanner={codeScanner}
      />
      <View
        style={{
          position: 'absolute',
          bottom: 0,
          padding: 16,
          backgroundColor: 'rgba(0,0,0,0.5)',
        }}>
        <Text style={{color: 'white'}}>
          {qrCode ? `Scanned QR Code: ${qrCode}` : 'Scanning QR code...'}
        </Text>
      </View>
    </SafeAreaView>
  );
}
const styles = StyleSheet.create({
  input: {
    height: 40,
    margin: 12,
    borderWidth: 1.5,
    padding: 5,
    borderRadius: 12,
    fontSize: 16,
    width: '85%',
  },
  modal: {
    justifyContent: 'center',
    alignItems: 'center',
  },
  modalTitle: {
    color: '#555',
    fontSize: 22,
    fontWeight: '600',
  },
  modalText: {
    fontSize: 18,
    color: '#555',
    marginTop: 14,
    textAlign: 'center',
    marginBottom: 10,
  },
  button: {
    backgroundColor: '#000',
    paddingVertical: 12,
    paddingHorizontal: 16,
    width: '100%',
    alignItems: 'center',
    marginTop: 10,
  },
  buttonText: {
    color: '#fff',
    fontSize: 20,
  },
  scrollView: {
    backgroundColor: '#FFF',
  },
});

ngenerapps, 2023-10-30:
@zjkuang 
`import React, {useState, useEffect, Component} from 'react';
import {
  SafeAreaView,
  RefreshControl,
  ScrollView,
  StatusBar,
  StyleSheet,
  PermissionsAndroid,
  Pressable,
  Text,
  useColorScheme,
  View,
  TouchableOpacity,
  Image,
  TextInput,
  Linking,
  Button,
  AnimationEffect,
  ReadableStreamReader,
  ActivityIndicator,
} from 'react-native';
import QRCodeScanner from 'react-native-qrcode-scanner';
// import {ViewPropTypes} from 'deprecated-react-native-prop-types';
import {RNCamera} from 'react-native-camera';
import {
  Camera,
  useCameraDevices,
  useCodeScanner,
} from 'react-native-vision-camera';
import {useScanBarcodes, BarcodeFormat} from 'vision-camera-code-scanner';
import PropTypes from 'prop-types';
import Axios from 'axios';
import Modal from 'react-native-modal';
import RNRestart from 'react-native-restart';
import Geolocation from '@react-native-community/geolocation';
import WifiManager, {getCurrentWifiSSID} from 'react-native-wifi-reborn';
import {useNetInfo} from '@react-native-community/netinfo';

export default function Home({navigation}) {
  const devices = useCameraDevices();
  const device = devices.back;
  const types = ['qr'];

  useEffect(() => {
    checkPermission();
  }, []);

  const checkPermission = async () => {
    try {
      const newCameraPermission = await Camera.requestCameraPermission();
      console.log('Camera Permission:', newCameraPermission);
    } catch (error) {
      console.error('Camera Permission Error:', error);
    }
  };

 // const [qrCode, setQRCode] = useState('');

 // const onCodeScanned = event => {
 //   console.log('Scanned QR Code:', event.nativeEvent.codeStringValue);
 //   setQRCode(event.nativeEvent.codeStringValue);
 // };

  const codeScanner = useCodeScanner({
    codeTypes: types,
    onCodeScanned(codes) {
      if (codes.length) {
        // process barcodes
      }
    },
  });

  if (device == null) {
    return (
      <View style={{flex: 1, justifyContent: 'center', alignItems: 'center'}}>
        <ActivityIndicator />
      </View>
    );
  }

  return (
    <SafeAreaView
      style={{
        flex: 1,
        justifyContent: 'center',
        alignItems: 'center',
        backgroundColor: '#000',
      }}>
      <Camera
        style={styles.boundingBoxFrame}
        ref={Camera}
        device={device}
        isActive={isActive}
        enableZoomGesture={Boolean(mode?.photo)}
        zoom={device?.neutralZoom ?? 1}
        photo={Boolean(mode?.photo)}
        codeScanner={codeScanner}
      />
      <View
        style={{
          position: 'absolute',
          bottom: 0,
          padding: 16,
          backgroundColor: 'rgba(0,0,0,0.5)',
        }}>
        <Text style={{color: 'white'}}>
          {qrCode ? `Scanned QR Code: ${qrCode}` : 'Scanning QR code...'}
        </Text>
      </View>
    </SafeAreaView>
  );
}
const styles = StyleSheet.create({
  input: {
    height: 40,
    margin: 12,
    borderWidth: 1.5,
    padding: 5,
    borderRadius: 12,
    fontSize: 16,
    width: '85%',
  },
  modal: {
    justifyContent: 'center',
    alignItems: 'center',
  },
  modalTitle: {
    color: '#555',
    fontSize: 22,
    fontWeight: '600',
  },
  modalText: {
    fontSize: 18,
    color: '#555',
    marginTop: 14,
    textAlign: 'center',
    marginBottom: 10,
  },
  button: {
    backgroundColor: '#000',
    paddingVertical: 12,
    paddingHorizontal: 16,
    width: '100%',
    alignItems: 'center',
    marginTop: 10,
  },
  buttonText: {
    color: '#fff',
    fontSize: 20,
  },
  scrollView: {
    backgroundColor: '#FFF',
  },
});`

Oliveirax, 2023-10-31:
Not sure It is related, but I cannot read barcodes in iOS, in fact, I cannot read anything in iOS, if I include "codabar" in the CodeTypes passed to the codescanner. It outputs an error to the console: 

`2023-10-30 19:37:35.198619+0000 Anova Go Debug[437:19553] [native] VisionCamera.didSetProps(_:): Updating 15 prop(s)...
2023-10-30 19:37:35.199275+0000 Anova Go Debug[437:20003] [native] VisionCamera.configureCaptureSession(): Configuring Session...
2023-10-30 19:37:35.199306+0000 Anova Go Debug[437:20003] [native] VisionCamera.configureCaptureSession(): Initializing Camera with device com.apple.avfoundation.avcapturedevice.built-in_video:0...
2023-10-30 19:37:35.199328+0000 Anova Go Debug[437:20003] [native] VisionCamera.configureCaptureSession(): Adding Video input...
2023-10-30 19:37:35.209041+0000 Anova Go Debug[437:20003] [native] VisionCamera.invokeOnError(_:cause:): Invoking onError(): The value "{
    codeTypes =     (
        "code-128",
        "code-39",
        "code-93",
        codabar,
        "ean-13",
        "ean-8",
        itf,
        "upc-e",
        qr,
        "pdf-417",
        aztec,
        "data-matrix"
    );
    onCodeScanned = "<null>";
}" could not be parsed to type codeScanner!`

Not including it in the list, makes it possible to read other codes with success. I wonder if "codabar" is a typo or something like that.

using: 
react-native 0.72.5 
react-native-vision-camera 3.3.1

mrousavy, 2023-10-31:
Codabar is only available on iOS 15.4

Oliveirax, 2023-10-31:
> Codabar is only available on iOS 15.4

So, in iOS versions prior to 15.4 we can't read barcodes using this library? And must take care not to initialize the CodeReader with this code or else it won't even run? Or is there a workaround?

mrousavy, 2023-10-31:
> So, in iOS versions prior to 15.4 we can't read barcodes using this library?

Wrong. You can read QR codes, Barcodes, and whatever codes, but you cannot read `codabar` codes. This is the only code type that you cannot read on iOS 15.3 or lower.

This is btw a platform restriction. This does not work on any iOS app unless it uses a custom MLKit implementation instead of iOS' default one.

> And must take care not to initialize the CodeReader with this code or else it won't even run? Or is there a workaround?

I don't know what you mean with this.
Just don't pass `codabar` if you don't use it, then you're good

Oliveirax, 2023-10-31:
> don't know what you mean with this.
I mean exactly what I said. If you try to initialize the CodeScanner with "codabar" in the CodeTypes, (in an older version of iOS) the CodeScanner does not even initialize and the camera doesn't read any type of code whatsoever. 

> This is the only code type that you cannot read on iOS 15.3 or lower.
barcoda is just the most common type of barcode in the world, so this is a big issue IMHO. 

Please understand I am not badmouthing your work, it's a fine piece of software, no doubt there!  I just think this should be mentioned in the "known limitations" section. So that this lib doesn't find its way into production code, and some user with a company issued iPhone 6S calls support in panic, because suddenly the app cannot read barcodes and he can't do his job.

zjkuang, 2023-10-31:
> @zjkuang I have an issue with the "useCodeScanner".
> TypeError: 0, _$$_REQUIRE(_dependencyMap[11(...)e-vision-camera").useCodeScanner is not a function (it is
> undefined)
> Do you have any ideas on how to fix it?
> I really appreciate any help you can provide.

I guess you are experiencing the same issue in this post, https://github.com/mrousavy/react-native-vision-camera/issues/2106, and please try the method I suggested there. @ngenerapps

zjkuang, 2023-10-31:
> > > @zjkuang I have an issue with the "useCodeScanner".
> > > TypeError: 0, _$$_REQUIRE(_dependencyMap[11(...)e-vision-camera").useCodeScanner is not a function (it is
> > > undefined)
> > > Do you have any ideas on how to fix it?
> > > I really appreciate any help you can provide.
> > 
> > 
> > I guess you are experiencing the same issue in this post, #2106, and please try the method I suggested there. @ngenerapps
> 
> I got this error
> 
> At line:1 char:24
> 
> * watchman watch-del-all && yarn
>   start --reset-cache
> * ```
>                      ~~
>   ```
> 
> The token '&&' is not a valid statement separator in this version. + CategoryInfo : Parser Error: (:) [], ParentContainsErr orRecordException + FullyQualifiedErrorId : Invali dEndOfLine

Perhaps your environment is different than mine. (I am on Mac with zsh.) Please search online for how to clean caches for React Native Android in your environment.

quocbaooo, 2023-11-01:
I create frame using frame processors to scanner QR code but not working someone help me pls

zjkuang, 2023-11-01:
> I create frame using frame processors to scanner QR code but not working someone help me pls

Frame Processor does not work with scanner. I removed it. Please check my code snippet in the content of this ticket.

trilam1409, 2023-12-28:
> Same here, works fine on iOS, but it does not recognize on Android. plz help me codeTypes: ['qr', 'ean-13'], "react-native-vision-camera": "^3.6.4", <Camera style={{ width: "100%", height: "100%" , backgroundColor:w}} device={device}
> 
> ```
>           isActive={true}
>           {...props} codeScanner={codeScanner}
>         >
> ```

I face the same issue. 

Library version: 3.6.17. 

It works well on iPhone but it doesn't work on Redmi 12 Pro.

quocbaooo, 2023-12-28:
> I face the same issue.
>
> Library version: 3.6.17.
>
> It works well on iPhone but it doesn't work on Redmi 12 Pro.

on Redmi 12 pro black screen ?

trilam1409, 2023-12-28:
> on Redmi 12 pro black screen ?

Yes, it was a black screen. 

There is a tip like this comment to make the camera work. But it's unstable.

#2308

quocbaooo, 2023-12-28:
> Yes, it was a black screen.
>
> There is a tip like this comment to make the camera work. But it's unstable.
>
> #2308

The camera is still visible but it's almost dark, right?

trilam1409, 2023-12-28:
> The camera is still visible but it's almost dark, right?

No, it shows the review screen of the camera. But the QR code scan doesn't wok.

quocbaooo, 2023-12-28:
> No, it shows the review screen of the camera. But the QR code scan doesn't wok.

Can I see your code?

trilam1409, 2023-12-28:
@quocbaooo 

`<Camera
        onInitialized={() => setIsActive(true)}
        style={StyleSheet.absoluteFill}
        device={device}
        isActive={isActive}
        codeScanner={codeScanner}
        onError={(error) => {
          console.log('error', error)
        }}
      />`

trilam1409, 2023-12-28:
> Same here, works fine on iOS, but it does not recognize on Android. plz help me codeTypes: ['qr', 'ean-13'], "react-native-vision-camera": "^3.6.4", <Camera style={{ width: "100%", height: "100%" , [...]

You should change codeTypes to only "qr". It woks on my side.

`codeTypes: ['qr'],`

quocbaooo, 2023-12-28:
> @quocbaooo
> 
> `<Camera onInitialized={() => setIsActive(true)} style={StyleSheet.absoluteFill} device={device} isActive={isActive} codeScanner={codeScanner} onError={(error) => { console.log('error', error) }} />`

still black screen on redmi

billoneillighttiger, 2023-12-28:
Please downgrade to 3.6.16. Are you using expo or cli?

quocbaooo, 2023-12-28:
> Please downgrade to 3.6.16. Are you using expo or cli?
> […](#)

i using cli and version 3.6.16

billoneillighttiger, 2023-12-28:
Is your iOS build (npm run ios) successful with version 3.6.16 ?

quocbaooo, 2023-12-28:
I only have the problem on Android with Redmi devices, the rest on iOS or other Android devices are still normal

billoneillighttiger, 2023-12-28:
Can you share the code snippet including Camera tag? Because I want to know
if you are using frameprocessor

quocbaooo, 2023-12-28:
[image]

> Can you share the code snippet including Camera tag? Because I want to know if you are using frameprocessor
> […](#)

Do you need anything else?

quocbaooo, 2023-12-28:
> Can you share the code snippet including Camera tag? Because I want to know if you are using frameprocessor
> […](#)

{device != null && (
        <ReanimatedCamera
          style={[StyleSheet.absoluteFill]}
          device={device}
          onError={onError}
          isActive={isActive}
          format={format}
          fps={fps}
          zoom={0}
          codeScanner={codeScanner}
          // onInitialized={() => setIsActive(true)}
        />
      )}

billoneillighttiger, 2023-12-28:
@quocbaooo can you let me know your react-native version?

quocbaooo, 2023-12-29:
> @quocbaooo can you let me know your react-native version?

react native: 0.72.7

trilam1409, 2024-01-05:
You should downgrade the library to v3.6.16.
It's working on my side.

#2340

PR #2082 (2023-10-25): fix: Fix `CodeScanner` not being called

## What

Fixes the CodeScanner not being called anymore since the latest update.

## Changes

## Tested on

## Related issues

- Fixes #2078