margelo/react-native-vision-camera #2079: 🔧 VisionCamera.modulemap not found

Skseo91, 2023-10-25:
### How were you trying to build the app?

I ran an iOS app that was using VisionCamera V3 on Xcode, and there was an Xcode update for iOS 17 version in the morning.
After the update, this error occurs, so you cannot build it normally and an error occurs.

### Full build logs

```tsx
The following build errors occur when using my physical device iPhone XR in release mode on Xcode.

/Users/vitasoftm2/Library/Developer/Xcode/DerivedData/hamanbike-csksayeagjsqrububbfdvaukmsgl/Build/Intermediates.noindex/hamanbike.build/Debug-iphoneos/hamanbike.build/DerivedSources/hamanbike_vers.c module map file '/Users/vitasoftm2/Library/Developer/Xcode/DerivedData/hamanbike-csksayeagjsqrububbfdvaukmsgl/Build/Products/Debug-iphoneos/VisionCamera/VisionCamera.modulemap' not found
```

### Project dependencies

```json
"dependencies": {
    "@dlwo159/react-native-nmap": "^0.0.69",
    "@react-native-async-storage/async-storage": "^1.17.11",
    "@react-navigation/drawer": "^6.6.3",
    "@react-navigation/native": "^6.1.7",
    "@react-navigation/native-stack": "^6.9.10",
    "@react-navigation/stack": "^6.3.14",
    "@rneui/base": "^4.0.0-rc.7",
    "@rneui/themed": "^4.0.0-rc.8",
    "@tosspayments/payment-sdk": "^1.6.5",
    "appcenter": "5.0.0",
    "appcenter-analytics": "5.0.0",
    "appcenter-crashes": "5.0.0",
    "axios": "^1.3.3",
    "convert-string": "^0.1.0",
    "geolib": "^3.3.4",
    "iamport-react-native": "^2.0.5",
    "moment": "^2.29.4",
    "react": "18.1.0",
    "react-native": "0.70.6",
    "react-native-ble-manager": "^10.1.2",
    "react-native-bluetooth-state-manager": "^1.3.5",
    "react-native-code-push": "^7.0.4",
    "react-native-fast-image": "^8.6.3",
    "react-native-flash-message": "^0.4.2",
    "react-native-fused-location": "^1.1.2",
    "react-native-geolocation-service": "^5.3.1",
    "react-native-gesture-handler": "^2.12.1",
    "react-native-modal": "^13.0.1",
    "react-native-pager-view": "^6.2.0",
    "react-native-permissions": "^3.9.0",
    "react-native-progress": "^5.0.0",
    "react-native-reanimated": "^2.17.0",
    "react-native-safe-area-context": "^4.5.0",
    "react-native-screens": "^3.20.0",
    "react-native-send-intent": "^1.3.0",
    "react-native-svg": "^13.10.0",
    "react-native-vector-icons": "^10.0.0",
    "react-native-vision-camera": "^3.0.0",
    "react-native-webview": "^11.26.1"
  },
  "devDependencies": {
    "@babel/core": "^7.12.9",
    "@babel/runtime": "^7.12.5",
    "@react-native-community/eslint-config": "^2.0.0",
    "@tsconfig/react-native": "^2.0.2",
    "@types/jest": "^26.0.23",
    "@types/react": "^18.0.21",
    "@types/react-native": "^0.70.6",
    "@types/react-native-version-check": "^3.4.5",
    "@types/react-test-renderer": "^18.0.0",
    "@typescript-eslint/eslint-plugin": "^5.37.0",
    "@typescript-eslint/parser": "^5.37.0",
    "babel-jest": "^26.6.3",
    "eslint": "^7.32.0",
    "jest": "^26.6.3",
    "metro-react-native-babel-preset": "0.72.3",
    "react-native-version-check": "^3.4.7",
    "react-test-renderer": "18.1.0",
    "typescript": "^4.8.3"
  },
```

### VisionCamera Version

3.3.1

### Target platforms

iOS

### Operating system

MacOS

### Can you build the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

Skseo91, 2023-10-25:
Sorry, it was a mistake about my inexperience in using Xcode. Everything worked fine after rebooting.

Stance0102, 2024-01-08:
Hi ,So, how did you solve it?

JB712, 2024-02-14:
@Stance0102 In my case, the error was to open xcodeproj file instead of xcworkspace
found out thanks to https://stackoverflow.com/a/59696931