margelo/react-native-vision-camera #2080: "app:checkDebugDuplicateClasses" error when building on Android

nicklasbring8, 2023-10-25:
### How were you trying to build the app?

I updated react-native-vision-camera from version 2.15.6 to 3.5.1. However, when I attempt to build the app, I encounter the following error, along with numerous duplicate class errors:

```
* What went wrong:
Execution failed for task ':app:checkDebugDuplicateClasses'.
> A failure occurred while executing com.android.build.gradle.internal.tasks.CheckDuplicatesRunnable
   > Duplicate class com.facebook.debug.debugoverlay.model.DebugOverlayTag found in modules jetified-react-android-0.73.0-rc-debug.3-debug-runtime (com.facebook.react:react-android:0.73.0-rc.3) and jetified-react-native-0.70-debug.5-debug-runtime (com.facebook.react:react-native:0.70.5)
     Duplicate class com.facebook.debug.holder.NoopPrinter found in modules jetified-react-android-0.73.0-rc-debug.3-debug-runtime (com.facebook.react:react-android:0.73.0-rc.3) and jetified-react-native-0.70-debug.5-debug-runtime (com.facebook.react:react-native:0.70.5)
     Duplicate class com.facebook.debug.holder.Printer found in modules jetified-react-android-0.73.0-rc-debug.3-debug-runtime (com.facebook.react:react-android:0.73.0-rc.3) and jetified-react-native-0.70-debug.5-debug-runtime (com.facebook.react:react-native:0.70.5)
     Duplicate class com.facebook.debug.holder.PrinterHolder found in modules jetified-react-android-0.73.0-rc-debug.3-debug-runtime (com.facebook.react:react-android:0.73.0-rc.3) and jetified-react-native-0.70-debug.5-debug-runtime (com.facebook.react:react-native:0.70.5)
     Duplicate class com.facebook.debug.tags.ReactDebugOverlayTags found in modules jetified-react-android-0.73.0-rc-debug.3-debug-runtime (com.facebook.react:react-android:0.73.0-rc.3) and jetified-react-native-0.70-debug.5-debug-runtime (com.facebook.react:react-native:0.70.5)
     Duplicate class com.facebook.fbreact.specs.NativeAccessibilityInfoSpec found in modules jetified-react-android-0.73.0-rc-debug.3-debug-runtime (com.facebook.react:react-android:0.73.0-rc.3) and jetified-react-native-0.70-debug.5-debug-runtime (com.facebook.react:react-native:0.70.5)
     Duplicate class com.facebook.fbreact.specs.NativeAccessibilityManagerSpec found in modules jetified-react-android-0.73.0-rc-debug.3-debug-runtime (com.facebook.react:react-android:0.73.0-rc.3) and jetified-react-native-0.70-debug.5-debug-runtime (com.facebook.react:react-native:0.70.5)
     Duplicate class com.facebook.fbreact.specs.NativeActionSheetManagerSpec found in modules jetified-react-android-0.73.0-rc-debug.3-debug-runtime (com.facebook.react:react-android:0.73.0-rc.3) and jetified-react-native-0.70-debug.5-debug-runtime (com.facebook.react:react-native:0.70.5)
```

I'm currently using React Native version 0.70.5, and I'm unsure where the 0.73.0-rc3 comes from, except that it appeared after I installed version 3 of the react-native-vision-camera module.

I recall encountering issues with my previous version of React Native when version 0.71.0 was uploaded to Maven Central. To resolve the problem, I had to update my React Native version to one of their patch versions because my app started pulling the latest version of React Native from Maven Central. This information may be relevant to the current issue.

### Full build logs

```tsx
npx react-native run-android
```

### Project dependencies

```json
"dependencies": {
    "@react-navigation/bottom-tabs": "^6.4.0",
    "@react-navigation/native": "^6.0.13",
    "@react-navigation/native-stack": "^6.9.0",
    "@react-navigation/stack": "^6.3.2",
    "@reduxjs/toolkit": "^1.8.5",
    "@software/react-native-nabto": "^0.3.6",
    "crypto-js": "^4.1.1",
    "i18next": "^22.0.6",
    "react": "18.1.0",
    "react-i18next": "^12.0.0",
    "react-native": "0.70.5",
    "react-native-bouncy-checkbox": "^3.0.6",
    "react-native-dropdown-picker": "^5.4.3",
    "react-native-fs": "^2.20.0",
    "react-native-gesture-handler": "^2.7.1",
    "react-native-html-to-pdf": "^0.12.0",
    "react-native-localize": "^2.2.4",
    "react-native-mmkv": "^2.5.1",
    "react-native-reanimated": "^2.14.2",
    "react-native-safe-area-context": "^4.3.3",
    "react-native-screens": "^3.17.0",
    "react-native-share": "^7.9.0",
    "react-native-svg": "^13.2.0",
    "react-native-svg-transformer": "^1.0.0",
    "react-native-vision-camera": "^2.16.5",
    "react-redux": "^8.0.4"
  },
```

### VisionCamera Version

3.5.1

### Target platforms

Android

### Operating system

Windows

### Can you build the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2023-10-25:
Hey!

I think VisionCamera V3 requires react-native 0.71. I'm not 100% sure but I think it does.

The error you sent also kinda looks like a cache issue. Maybe try upgrading to RN 0.71 first, then clear cache and then install VisionCamera V3 :)

closed 2023-10-25 by mrousavy

nicklasbring8, 2023-10-25:
Hey,

I had the same thought. The only thing is that upgrading from 0.70.5 is not just a small step, but I'll definitely give it a try. :)
I've attempted clearing the cache with no success.

nikoloz110, 2023-11-09:
@nicklasbring8 
hey I have exactly the same issue, (though my rn version is 0.69),  have you had any luck solving it?

nicklasbring8, 2023-11-09:
I apologize for the delay in providing an update.

I successfully updated my React Native version to 0.71.14, which resolved my issue. The React Native CLI upgrader was unable to automatically update my React Native version, so I had to manually update all my files using the helper tool. While it did take some time, it was a worthwhile effort.

@nikoloz110, if you are currently using version 0.69, can you build without the react-native-vision-camera module? I don't recall 0.69 having the necessary patches to address the major build issue on Android introduced around a year ago when they released 0.71. I also encountered significant difficulties building on Android back then, which prompted me to update to the 0.70.5 patch.

rodrigodiasf1984, 2024-03-08:
Same issue, react-native version=  "react-native": "0.70.6" how to solve it ?

rarenatoe, 2024-03-08:
Same issue, shameless bump