margelo/react-native-vision-camera #2085: 🐛 iOS `audio={true}` does not enable audio >v3.3.1

MassimoDeLuca, 2023-10-25:
### What's happening?

When setting `audio={true}` on the camera, the microphone is not utilized, nor is there an audio stream in the output video.

Only occurs on iOS and works fine on Android.

Tested the following:
❌ v3.5.1
❌ v3.4.1
✅ v3.3.1
✅ v3.2.0

### Reproduceable Code

```tsx
Reproducible on latest example https://github.com/mrousavy/react-native-vision-camera/tree/main/package/example

<Camera
  device={activeCamera}
  ref={(ref) => setCamera(ref)}
  style={cameraStyle}
  isActive={isFocused && !appBackgrounded}
  audio={true}
  video={true}
/>
```
```

### Relevant log output

```shell
No logs or errors output
```

### Camera Device

```json
{
    "formats": [
    ],
    "hardwareLevel": "full",
    "hasFlash": true,
    "hasTorch": true,
    "id": "com.apple.avfoundation.avcapturedevice.built-in_video:0",
    "isMultiCam": false,
    "maxZoom": 121.875,
    "minZoom": 1,
    "name": "Back Camera",
    "neutralZoom": 1,
    "physicalDevices": [
      "wide-angle-camera"
    ],
    "position": "back",
    "sensorOrientation": "landscape-right",
    "supportsFocus": true,
    "supportsLowLightBoost": false,
    "supportsRawCapture": false
  },
  {
    "formats": [
    ],
    "hardwareLevel": "full",
    "hasFlash": true,
    "hasTorch": false,
    "id": "com.apple.avfoundation.avcapturedevice.built-in_video:1",
    "isMultiCam": false,
    "maxZoom": 128.875,
    "minZoom": 1,
    "name": "Front Camera",
    "neutralZoom": 1,
    "physicalDevices": [
      "wide-angle-camera"
    ],
    "position": "front",
    "sensorOrientation": "landscape-right",
    "supportsFocus": false,
    "supportsLowLightBoost": false,
    "supportsRawCapture": false
  },
```

### Device

Phone 13 (iOS 13)

### VisionCamera Version

>=3.4.0

### Can you reproduce this issue in the VisionCamera Example app?

Yes, I can reproduce the same issue in the Example app here

### Additional information

- [X] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

james-cohen, 2023-10-25:
Also having the same issue, no problems on Android. Stopped working with no changes apart from package update, no issues previously

spenweb, 2023-10-30:
I was able to confirm downgrading to v3.3.1 in the VisionCamera Example app brought back audio stream in video recording on physical iPhone running iOS 17.

I am also not able to get audio stream on video recordings using the latest (v3.6.4) inside my app on iPhone running iOS 17.

mrousavy, 2023-10-31:
Honestly, this is starting to get annoying. Why is everybody leaving out logs? This is literally what logs are for, and I even have an exact and precise explanation on how you can get the logs so you can put them here.

Without logs, I cannot help you.

yspreen, 2023-10-31:
workaround: set audio to false, use effect to set it to true after 1000ms

yspreen, 2023-10-31:
I also set it to false and back to true after 100ms every time a file has finished recording.

spenweb, 2023-10-31:
Here is Xcode log output from my iPhone 12 mini on iOS 17 from the latest from the main branch (pulled within the last hour) with the `frameProcessor` prop commented out:
```
Successfully registered Frame Processor Plugin "example_plugin"!
Successfully registered Frame Processor Plugin "example_kotlin_swift_plugin"!
Running application VisionCameraExample ({
    initialProps =     {
    };
    rootTag = 1;
})
Thread Performance Checker: Thread running at User-interactive quality-of-service class waiting on a lower QoS thread running at Default quality-of-service class. Investigate ways to avoid priority inversions
PID: 4862, TID: 628064
Backtrace
=================================================================
3   VisionCameraExample                 0x0000000100f4ebd4 -[SRRunLoopThread runLoop] + 44
4   VisionCameraExample                 0x0000000100f50c4c +[NSRunLoop(SRWebSocket) SR_networkRunLoop] + 56
5   VisionCameraExample                 0x0000000100f4d5ec -[SRProxyConnect _openConnection] + 72
6   VisionCameraExample                 0x0000000100f4caa4 -[SRProxyConnect _configureProxy] + 916
7   VisionCameraExample                 0x0000000100f4c298 -[SRProxyConnect openNetworkStreamWithCompletion:] + 92
8   VisionCameraExample                 0x0000000100f52008 -[SRWebSocket open] + 624
9   VisionCameraExample                 0x0000000100c62dc4 -[RCTReconnectingWebSocket start] + 148
10  VisionCameraExample                 0x0000000100c5348c -[RCTPackagerConnection init] + 416
11  VisionCameraExample                 0x0000000100c532cc __49+[RCTPackagerConnection sharedPackagerConnection]_block_invoke + 36
12  libdispatch.dylib                   0x00000001039c2b34 _dispatch_client_callout + 20
13  libdispatch.dylib                   0x00000001039c4690 _dispatch_once_callout + 140
14  VisionCameraExample                 0x0000000100c53280 +[RCTPackagerConnection sharedPackagerConnection] + 88
15  VisionCameraExample                 0x0000000100c85644 -[RCTDevSettings initialize] + 164
16  VisionCameraExample                 0x0000000100b8b834 -[RCTModuleData _initializeModule] + 96
17  VisionCameraExample                 0x0000000100b8b1e4 -[RCTModuleData setUpInstanceAndBridge:] + 2168
18  VisionCameraExample                 0x0000000100b8cc9c -[RCTModuleData instance] + 1168
19  VisionCameraExample                 0x0000000100bb8508 -[RCTCxxBridge moduleForName:lazilyLoadIfNecessary:] + 712
20  VisionCameraExample                 0x0000000100b959b0 -[RCTModuleRegistry moduleForName:lazilyLoadIfNecessary:] + 140
21  VisionCameraExample                 0x0000000100b95918 -[RCTModuleRegistry moduleForName:] + 48
22  VisionCameraExample                 0x0000000100c965d0 -[RCTPerfMonitor devMenuItem] + 92
23  VisionCameraExample                 0x0000000100c9649c -[RCTPerfMonitor initialize] + 88
24  VisionCameraExample                 0x0000000100b8b834 -[RCTModuleData _initializeModule] + 96
25  VisionCameraExample                 0x0000000100b8b1e4 -[RCTModuleData setUpInstanceAndBridge:] + 2168
26  VisionCameraExample                 0x0000000100b8ced4 __25-[RCTModuleData instance]_block_invoke + 44
27  VisionCameraExample                 0x0000000100ba6020 RCTUnsafeExecuteOnMainQueueSync + 52
28  VisionCameraExample                 0x0000000100b8cb3c -[RCTModuleData instance] + 816
29  VisionCameraExample                 0x0000000100bbcd58 __49-[RCTCxxBridge _prepareModulesWithDispatchGroup:]_block_invoke + 156
30  libdispatch.dylib                   0x00000001039c0f50 _dispatch_call_block_and_release + 32
31  libdispatch.dylib                   0x00000001039c2b34 _dispatch_client_callout + 20
32  libdispatch.dylib                   0x00000001039d3158 _dispatch_main_queue_drain + 1448
33  libdispatch.dylib                   0x00000001039d2ba0 _dispatch_main_queue_callback_4CF + 44
34  CoreFoundation                      0x00000001aba4c9bc CEB74A0C-5EA4-3F5A-9DE8-73F0DAD2F4CB + 227772
35  CoreFoundation                      0x00000001aba496c8 CEB74A0C-5EA4-3F5A-9DE8-73F0DAD2F4CB + 214728
36  CoreFoundation                      0x00000001aba48e18 CFRunLoopRunSpecific + 608
37  GraphicsServices                    0x00000001ee50d5ec GSEventRunModal + 164
38  UIKitCore                           0x00000001ade572fc EC1206D7-B4C9-3257-BB63-08E26A802416 + 2290428
39  UIKitCore                           0x00000001ade56938 UIApplicationMain + 340
40  VisionCameraExample                 0x00000001008cc874 main + 100
41  dyld                                0x00000001ce233d44 0D856E72-9A63-3B5B-879F-8D026115A4F0 + 23876
Running "VisionCameraExample" with {"rootTag":1,"initialProps":{}}
nw_socket_handle_socket_event [C5:1] Socket SO_ERROR 61
nw_connection_get_connected_socket_block_invoke [C5] Client called nw_connection_get_connected_socket on unconnected nw_connection
TCP Conn 0x28099d400 Failed : error 0:61 [61]
nw_connection_copy_connected_local_endpoint_block_invoke [C5] Client called nw_connection_copy_connected_local_endpoint on unconnected nw_connection
nw_connection_copy_connected_remote_endpoint_block_invoke [C5] Client called nw_connection_copy_connected_remote_endpoint on unconnected nw_connection
Re-rendering Navigator. Camera: undefined | Microphone: undefined
Loading react-native-worklets-core...
Worklets loaded successfully
VisionCameraProxy: Creating Worklet Context...
VisionCameraProxy: Worklet Context Created!
Re-rendering Navigator. Camera: granted | Microphone: undefined
[... cut ...]
Video loaded. Size: 2160x3840 (landscape, 1.5183333158493042 seconds)
VisionCamera.withDeviceLock(_:): Beginning CaptureDevice configuration...
VisionCamera.withDeviceLock(_:): Committed CaptureDevice configuration!
media has loaded.
VisionCamera.didSetProps(_:): Updating 1 props: [isActive]
VisionCamera.configure(_:): Updating Session Configuration...
VisionCamera.withDeviceLock(_:): Beginning CaptureDevice configuration...
VisionCamera.withDeviceLock(_:): Committed CaptureDevice configuration!
```

anirudhsama, 2023-11-02:
I can also confirm that audio isn't working on 3.6.4.

anitrix, 2023-11-04:
Same issue here, audio is not being captured or utilized when recording a video. v3.5.1

mrousavy, 2023-11-07:
Thanks for the logs @spenweb.

So in the VisionCamera Example app here it does not record any audio at all, correct?

spenweb, 2023-11-07:
@mrousavy, that's correct. No audio in the preview (inside the VisionCamera Example app) and no audio when saved to the native gallery app ("Photos" on iOS).

yspreen, 2023-11-08:
can we reopen this now? it's not resolved just bc logs were missing for a while...

we can duplicate it but valuable discussion would be missing

mrousavy, 2023-11-09:
Okay thanks @spenweb, weird.

mrousavy, 2023-11-10:
fixed ;)

closed 2023-11-10 by mrousavy

LunatiqueCoder, 2024-10-23:
> workaround: set audio to false, use effect to set it to true after 1000ms

WOW! Thanks so much for the workaround! The issue is still here on **v4.5.2**. We looked so much for a fix, and this is it!