margelo/react-native-vision-camera #2086: 🐛 Camera zoom state becomes unsynced when using enableZoomGesture

stevengoldberg, 2023-10-25:
### What's happening?

To reproduce:

1. Pass `enableZoomGesture` to the camera
2. Perform a pinch gesture to zoom in or out
3. Set Camera `isActive` to false (e.g. by taking a photo and displaying a preview)
4. Set Camera `isActive` to true (e.g. by discarding the photo preview)
5. When the Camera re-renders, perform a pinch gesture again

**Expected**: The Camera will begin zooming from the zoom level displayed in the video preview
**Actual**: The Camera snaps back to the zoom level from step 3

Video demo: https://imgur.com/a/3XpEhLt

Relatedly, I think: when zooming out to 0.5x using the triple camera, the result of `Camera.takePhoto` will always be at 1x zoom, despite the video preview being correct.

Video demo: https://imgur.com/a/4CjuYnd

### Reproduceable Code

```tsx
const handleTakePhoto = useCallback(() => {
    setIsShowingPreview(true)
    camera.current
    .takePhoto({
       qualityPrioritization: 'quality',
       enableAutoStabilization: true,
     })
     .then((photo) => {
        setPreviewImageUri(photo.path)
      })    
    }, [])

// ...

<Camera
   style={{ flex: 1, width: windowWidth }}
   ref={cameraRef}
   device={device}
   enableHighQualityPhotos
   isActive={isActive}
   photo
   enableZoomGesture
   orientation="portrait"
   format={format}
   zoom={device.neutralZoom}
/>
```

### Relevant log output

```shell
N/A
```

### Camera Device

```json
{
  "supportsLowLightBoost": false,
  "id": "com.apple.avfoundation.avcapturedevice.built-in_video:7",
  "maxZoom": 189,
  "isMultiCam": true,
  "sensorOrientation": "landscape-right",
  "hasTorch": true,
  "hardwareLevel": "full",
  "supportsFocus": true,
  "supportsRawCapture": false,
  "neutralZoom": 2,
  "physicalDevices": [
    "ultra-wide-angle-camera",
    "wide-angle-camera",
    "telephoto-camera"
  ],
  "position": "back",
  "name": "Back Triple Camera",
  "hasFlash": true,
  "minZoom": 1
}
```

### Device

iPhone 14 Pro

### VisionCamera Version

3.6.3

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [X] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2023-10-31:
Interesting, is this also happening on Android?

stevengoldberg, 2023-10-31:
I'm afraid I don't have an Android device to test with.

mrousavy, 2024-01-15:
Hey - I think I fixed this in the latest version of VisionCamera. :)

closed 2024-01-15 by mrousavy