margelo/react-native-vision-camera #2091: 🐛 Crash when pixelFormat="yuv"

R4tch3t, 2023-10-26:
### What's happening?

Why crash when set prop pixelFormat="yuv" if remove these prop work fine but is needed for frameProcessor for process image in ML Kit 

### Reproduceable Code

```tsx
{device != null && isActive && (
        <PinchGestureHandler onGestureEvent={onPinchGesture} enabled={isActive}>
          <Reanimated.View style={StyleSheet.absoluteFill}>
            <TapGestureHandler onEnded={onDoubleTap} numberOfTaps={2}>
              <ReanimatedCamera
                ref={camera}
                style={StyleSheet.absoluteFill}
                device={device}
                format={format}
                fps={fps}
                hdr={enableHdr}
                lowLightBoost={device.supportsLowLightBoost && enableNightMode}
                isActive={isActive}
                onInitialized={onInitialized}
                onError={onError}
                enableZoomGesture={false}
                animatedProps={cameraAnimatedProps}
                enableFpsGraph={true}                                
                photo={true}
                video={true}
                audio={hasMicrophonePermission}
                frameProcessor={frameProcessor}
                //pixelFormat={"rgb"}
                pixelFormat={'yuv'}                
              />
            </TapGestureHandler>
          </Reanimated.View>
        </PinchGestureHandler>
      )}
```

### Relevant log output

```shell
{ [session/camera-has-been-disconnected: [session/camera-has-been-disconnected] The given Camera device (id: 0) has been disconnected! Error: UNKNOWN_CAMERA_DEVICE_ERROR]
                                                                                                      name: 'session/camera-has-been-disconnected',
                                                                                                      _code: 'session/camera-has-been-disconnected',
                                                                                                      _message: '[session/camera-has-been-disconnected] The given Camera device (id: 0) has been disconnected! Error: UNKNOWN_CAMERA_DEVICE_ERROR',
                                                                                                      _cause: undefined }
```

### Camera Device

```json
{}
```

### Device

Samsung Galaxy S21

### VisionCamera Version

3.6.4

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [ ] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

seanpcoyle, 2023-10-26:
I'm also running into this on a Galaxy A14.

R4tch3t, 2023-10-26:
> I'm also running into this on a Galaxy A14.

mmmm, that works in some phone devices but in others crash....

R4tch3t, 2023-10-26:
Why pixelFormat="native" is data Inaccessible in frame processor ??? or it's possible??, if yes, how can I access to that data??

R4tch3t, 2023-10-26:
@mrousavy Please!
[image]
[image]
 You tell me if fix that if no I downgrade to V2 because I need a HOTfix urgent

zzz08900, 2023-10-27:
Well, I guess nobody here is responsible for your app crashing like ***. 

Stop whining and do something to fix YOUR app.

R4tch3t, 2023-10-27:
> Well, I guess nobody here is responsible for your app crashing like ***.
> 
> Stop whining and do something to fix YOUR app.
> 
> Edit: I do remember mrousavy offer paid services of fixing stuff like this, I think you can contact him if you really need such services.

It only fails due to the update of this library, I had v3rc2 with worklets in its infancy and it did not have the errors that I have now,
1.- YOU!? Have you done something to fix it!??, because that's what I do at least so that the OPEN source library listens well brother OPEN code!!!
2.- If I can pay 100, 1000 $ but it is version 3.6.4, that is what I will end up doing for an open source library...

R4tch3t, 2023-10-27:
@zzz08900  
You dare to ask stupid questions like Unable to call 'setState' within 'onBarcodeScanned' and comment here without contributing anything but hate...

rodgomesc, 2023-10-28:
@R4tch3t  can you ping me on discord? please join discord community if you are not already in it, and ping me.  i also need to solve that problem, but i don't have a samsung device right now, maybe we can jump in a pair programming session and figure that out

rodgomesc, 2023-10-28:
> pixelFormat="native"

unfortunately `native` format can't be read by a frame processor since the layout is private :(

rodgomesc, 2023-10-30:
@R4tch3t fyi https://github.com/mrousavy/react-native-vision-camera/pull/2108

can you run the example app from this PR with `pixelFormat` set to `yuv`?

mrousavy, 2023-10-31:
@R4tch3t 

> You tell me if fix that if no I downgrade to V2 because I need a HOTfix urgent....

If you need something "urgent" then **pay me to fix it** instead of posting spanish screenshots without context.

Also, you checked that you browsed the github repo for duplicates, but there is literally another issue with almost exactly the same title, lol. 

Closing as duplicate of https://github.com/mrousavy/react-native-vision-camera/issues/1947.

mrousavy, 2023-10-31:
Thanks @rodgomesc & @zzz08900 for helping out here ❤️

PR #2108 (2023-11-09): fix crash on samsung devices if using frameProcessor with YUV

## What

 There's a know bug that causes a crash on random samsung devices using `frameProcessor` with `yuv` format
 

## Tested on
-  Galaxy S20 FE, Android 13
-  Galaxy S21 FE, Android 13
-  Galaxy M31, Android 12

## How to Test
- the example app with `pixelformat=yuv` should crash in the `mrousavy:main` branch and work as expected in `rodgomesc:fix/samsung-crash-yuv`

## Related issues
Fixes #2091
closed 2023-11-09 by mrousavy