margelo/react-native-vision-camera #2095: The camera orientation UI rotates from landscape to portrait after upgrade to iOS17  🐛

martinmicunda, 2023-10-27:
### What's happening?

Since I upgraded to iOS17 when I rotate camera to landscape it rotates back to portrait mode but it should behave similer to iOS camera (I don't have this issue with iOS16). I still want to take landscape photo but it should not rotate the camera only provide camera output in landscape.

https://github.com/mrousavy/react-native-vision-camera/assets/1643606/660bacfd-0631-47bf-a6ba-f06543d49538

### Reproduceable Code

```tsx
interface Props {
  ref: React.RefObject<Camera>;
  device: CameraDevice;
  isActive: boolean;
  orientation: CameraOrientationType;
}

const PhotoCamera = forwardRef<Camera, Props>(
  ({ device, isActive, orientation }, ref) => {
    const styles = useStyle(getStyle);

    return (
      <Camera
        ref={ref}
        device={device}
        isActive={isActive}
        photo={true}
        preset='photo'
        enableZoomGesture={true}
        style={styles.camera}
        enableHighQualityPhotos={true}
        orientation={orientation}
        zoom={device.neutralZoom}
      />
    );
  },
);

export default PhotoCamera;
```

### Relevant log output

```shell
VisionCamera.didSetProps(_:): Updating 14 prop(s)...
VisionCamera.configureCaptureSession(): Configuring Session...
VisionCamera.configureCaptureSession(): Initializing Camera with device com.apple.avfoundation.avcapturedevice.built-in_video:6...
VisionCamera.configureCaptureSession(): Adding Video input...
VisionCamera.configureCaptureSession(): Adding Photo output...
VisionCamera.invokeOnInitialized(): Camera initialized!
VisionCamera.configureCaptureSession(): Session successfully configured!
VisionCamera.configureFormat(): Configuring Format...
VisionCamera.configureDevice(): Configuring Device...
VisionCamera.configureDevice(): Device successfully configured!
VisionCamera.didSetProps(_:): Starting Session...
nw_socket_handle_socket_event [C142.1.1:1] Socket SO_ERROR 54
nw_protocol_socket_reset_linger [C142.1.1:1] setsockopt SO_LINGER failed 22
VisionCamera.didSetProps(_:): Started Session!
VisionCamera.didSetProps(_:): Updating 1 prop(s)...
<<<< FigCaptureSessionRemote >>>> Fig assert: "err == 0 " at bail (FigCaptureSessionRemote.m:800) - (err=-12784)
```

### Camera Device

```json
{
  "supportsFocus": true,
  "supportsRawCapture": false,
  "neutralZoom": 2,
  "maxZoom": 189,
  "supportsLowLightBoost": false,
  "id": "com.apple.avfoundation.avcapturedevice.built-in_video:6",
  "minZoom": 1,
  "name": "Back Dual Wide Camera",
  "supportsParallelVideoProcessing": true,
  "hasFlash": true,
  "hasTorch": true,
  "position": "back",
  "devices": [
    "ultra-wide-angle-camera",
    "wide-angle-camera"
  ],
  "isMultiCam": true,
  "supportsDepthCapture": false
}
```

### Device

iPhone 15 Pro (iOS 17.0.3)

### VisionCamera Version

2.16.5

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

nn1900, 2023-10-29:
I have the same issue the camera orientation is updated automatically when I rotate the device from portrait to landscape. This is wired, I don't want the orientation to change automatically.

nn1900, 2023-10-29:
The change was introduced in commit https://github.com/mrousavy/react-native-vision-camera/commit/f35644df3357d36309c1827ef71aa6e383070b6c. The output orientation will be set to user specified orientation value if possible, otherwise, it will be set to the input orientation. Previously, the input orientation is pinned to "portrait". However is was changed to be the **current device orientation**. So, if the user does not manually specify the orientation and device rotated (orientation changed), the output orientation will be **reconfigured** to be current device orientation. I don't think this is the expected behavior. It should behaves the way the native camera app does instead: never reconfigure the output orientation.

nn1900, 2023-10-29:
We finally set the orientation to the value that matches current screen orientation so that the output orientation will be reconfigured when screen rotates on iPad (on iPhone we locked the screen orientation in portrait mode). This way, unless the screen orientation changes (on iPad, with screen orientation not locked on the device), the camera output orientation will not be reconfigured. 

```typescript
// a hook that returns and monitors the current screen orientation
const screenOrientation = useScreenOrientation();

// convert the screen orientation to camera device orientation
const cameraOrientation =  screenOrientationToDeviceOrientation(screenOrientation);

return (
   <Camera orientation={cameraOrientation} ... />
);

```

mrousavy, 2023-10-31:
Orientation support is a pinned issue. Sponsor/Donate so I can implement this: https://github.com/mrousavy/react-native-vision-camera/issues/1891

closed 2023-10-31 by mrousavy