margelo/react-native-vision-camera #2099: I need help to configure the camera❓

douglasndm, 2023-10-28:
### Question

I'm having an issue that I'm not sure if it is a bug or if I am using the lib wrongly. When I try to take the picture the device flash works, and the photo is taken but it looks like to photo is taken after the flash goes off. The photo is always dark and I am receiving many reports from multiple devices with the same issue.

I am using react native 0.72.5 without new arch.

This is the camera component
`
<Camera
  ref={camera}
  style={StyleSheet.absoluteFill}
  device={device}
  isActive={isFocused && !captured}
  orientation="portrait"
  enableZoomGesture={false}
  zoom={device?.neutralZoom || 1}
  lowLightBoost={device?.supportsLowLightBoost}
  photo
/>
`

and the take picture method

`
const photo = await camera.current.takePhoto({
	flash: enableFlash,
	qualityPrioritization: 'speed',
});
`

### What I tried

I tried to use "qualityPrioritization: 'speed'," and it didn't work. I also tried to use torch, torch worked as it expected but the result was the same.

### VisionCamera Version

3.6.4

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

douglasndm, 2023-10-28:
These are examples of how the picture are after the capture.

https://github.com/mrousavy/react-native-vision-camera/assets/7841509/b88b22b0-31
[image]
1a-4bf3-818a-ebb6ad1b5dd6

[image]

mrousavy, 2023-10-31:
Hi, yes Flash requires quite a complex precapture sequence, so I created a sponsor pool / fund raiser so I can dedicate time to implement that. Sponsor/Donate here if you want this happening: https://github.com/mrousavy/react-native-vision-camera/issues/1890

closed 2023-10-31 by mrousavy