margelo/react-native-vision-camera #2100: 🔧  Missing namespace information required by Gradle 8

adolfogc, 2023-10-29:
### How were you trying to build the app?

Tried to build my app using `pnpm run android`

### Full build logs

```tsx
> Task :react-native-vision-camera:processDebugManifest FAILED

Incorrect package="com.mrousavy.camera" found in source AndroidManifest.xml: [redacted]/node_modules/react-native-vision-camera/android/src/main/AndroidManifest.xml.
Setting the namespace via the package attribute in the source AndroidManifest.xml is no longer supported.
Recommendation: remove package="com.mrousavy.camera" from the source AndroidManifest.xml: [redacted]/node_modules/react-native-vision-camera/android/src/main/AndroidManifest.xml.

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':react-native-vision-camera:processDebugManifest'.
> A failure occurred while executing com.android.build.gradle.tasks.ProcessLibraryManifest$ProcessLibWorkAction
   > Incorrect package="com.mrousavy.camera" found in source AndroidManifest.xml: [redacted]/node_modules/react-native-vision-camera/android/src/main/AndroidManifest.xml.
     Setting the namespace via the package attribute in the source AndroidManifest.xml is no longer supported.
     Recommendation: remove package="com.mrousavy.camera" from the source AndroidManifest.xml: [redacted]/node_modules/react-native-vision-camera/android/src/main/AndroidManifest.xml.
```

### Project dependencies

```json
"dependencies": {
    "@aws-amplify/auth": "^5.6.5",
    "@aws-amplify/core": "^5.8.5",
    "@aws-amplify/pushnotification": "^5.0.45",
    "@aws-amplify/rtn-push-notification": "^1.1.7",
    "@react-native-async-storage/async-storage": "^1.19.3",
    "@react-native-community/netinfo": "^9.4.1",
    "@react-native-masked-view/masked-view": "^0.3.0",
    "@react-navigation/native": "^6.1.9",
    "@react-navigation/stack": "^6.3.20",
    "@rescript/core": "^0.5.0",
    "@rescript/react": "^0.11.0",
    "@sentry/react-native": "^5.11.1",
    "@urql/rescript": "^4.0.0",
    "amazon-cognito-identity-js": "^6.3.6",
    "graphql": "^16.8.1",
    "jsc-android": "^250231.0.0",
    "react": "18.2.0",
    "react-intl": "^6.5.1",
    "react-native": "0.72.6",
    "react-native-gesture-handler": "^2.13.4",
    "react-native-get-random-values": "^1.9.0",
    "react-native-gradle-plugin": "^0.71.19",
    "react-native-localize": "^3.0.2",
    "react-native-reanimated": "^3.5.4",
    "react-native-safe-area-context": "^4.7.4",
    "react-native-screens": "^3.25.0",
    "react-native-url-polyfill": "^2.0.0",
    "react-native-vision-camera": "^3.6.4",
    "rescript-react-native": "^0.72.0",
    "rescript-react-navigation": "^6.0.3",
    "rn-slide-button": "^1.0.3",
    "urql": "^4.0.5",
    "zustand": "^4.4.4"
  },
```

### VisionCamera Version

3.6.4

### Target platforms

Android

### Operating system

MacOS

### Can you build the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

zzz08900, 2023-10-29:
I could be wrong on this but, isn't RN going to be using Gradle 8 starting 0.73?
I thought 0.72 does not have that namespace bullshit messing around, am I wrong?

adolfogc, 2023-10-29:
Hi @zzz08900, I'm using 0.72.6

adolfogc, 2023-10-29:
Currently, a namespace is set (incorrectly) in `build.gradle`:

https://github.com/mrousavy/react-native-vision-camera/blob/f4617a3918783ff188c8b63f0694b1ad94ce3fae/package/android/build.gradle#L76

The `.example` needs to be removed.

zzz08900, 2023-10-29:
No idea why I didn't run into this while testing vision camera v3 with RN 0.72 but yeah that does look like a bug.

mrousavy, 2023-10-31:
Whoops my bad. Do we also need it in `AndroidManifest.xml`?

mrousavy, 2023-10-31:
Related PR: https://github.com/mrousavy/react-native-vision-camera/pull/2104

zzz08900, 2023-10-31:
> Whoops my bad. Do we also need it in `AndroidManifest.xml`?

I think we also need that in `AndroidManifest.xml` if namespace is used and both need to match.
It appears @adolfogc also took care of that.

PR #2104 (2023-10-31): fix: set correct namespace in build.gradle

## What

This PR fixes a build error in Android by setting the correct namespace in the `build.gradle` file.

## Changes

* Set correct namespace in `build.gradle`.

## Tested on

macOS

## Related issues

* Fixes #2100
closed 2023-10-31 by mrousavy