margelo/react-native-vision-camera #2106: ❓ UseCodeScanner

VampAK1864, 2023-10-30:
### Question

I have an issue with the "useCodeScanner".
TypeError: 0, _$$_REQUIRE(_dependencyMap[11(...)e-vision-camera").useCodeScanner is not a function (it is
undefined)
Do you have any ideas on how to fix it?
I really appreciate any help you can provide.

### What I tried

import {Camera, useCameraDevices, useCodeScanner, } from 'react-native-vision-camera';
import {useScanBarcodes, BarcodeFormat} from `'vision-camera-code-scanner';

const devices = useCameraDevices();
  const device = devices.back;

  useEffect(() => {
    checkPermission();
  }, []);

  const checkPermission = async () => {
    try {
      const newCameraPermission = await Camera.requestCameraPermission();
      console.log('Camera Permission:', newCameraPermission);
    } catch (error) {
      console.error('Camera Permission Error:', error);
    }
  };

  const [qrCode, setQRCode] = useState('');

  const onCodeScanned = event => {
    console.log('Scanned QR Code:', event.nativeEvent.codeStringValue);
    setQRCode(event.nativeEvent.codeStringValue);
  };

  if (device == null) {
    return (
      <View style={{flex: 1, justifyContent: 'center', alignItems: 'center'}}>
        <ActivityIndicator />
      </View>
    );
  }
  return (
    <SafeAreaView
      style={{
        flex: 1,
        justifyContent: 'center',
        alignItems: 'center',
        backgroundColor: '#000',
      }}>
      <Camera
        style={StyleSheet.absoluteFill}
        device={device}
        isActive={true}
        onCodeScanned={onCodeScanned}
      />
      <View
        style={{
          position: 'absolute',
          bottom: 0,
          padding: 16,
          backgroundColor: 'rgba(0,0,0,0.5)',
        }}>
        <Text style={{color: 'white'}}>
          {qrCode ? `Scanned QR Code: ${qrCode}` : 'Scanning QR code...'}
        </Text>
      </View>
    </SafeAreaView>
  );``

### VisionCamera Version

2.16.5

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

khuepm, 2023-10-30:
have you try new version?

VampAK1864, 2023-10-30:
Yes, I have tried the 3.6.4 and I am still facing the same issue

Antonis Kyriacou

mrousavy, 2023-10-31:
In 3.6.4. `useCodeScanner` is exported in index. It's not possible that this error is being thrown.

closed 2023-10-31 by mrousavy

zjkuang, 2023-10-31:
@VampAK1864  I guess you need to clean your metro cache. e.g. I would do a
```
watchman watch-del-all && yarn start --reset-cache
```

ngenerapps, 2023-10-31:
> @VampAK1864 I guess you need to clean your metro cache. e.g. I would do a
> 
> ```
> watchman watch-del-all && yarn start --reset-cache
> ```
I got this error

At line:1 char:24
+ watchman watch-del-all && yarn      
start --reset-cache
+                        ~~
The token '&&' is not a valid         
statement separator in this version.  
    + CategoryInfo          : Parser  
   Error: (:) [], ParentContainsErr   
  orRecordException
    + FullyQualifiedErrorId : Invali  
   dEndOfLine