margelo/react-native-vision-camera #2109: 🐛 videoCodec setting isn't used on iOS

yspreen, 2023-10-30:
### What's happening?

Setting videoCodec to 264 does not affect the codec. it's always 265

### Reproduceable Code

```tsx
camera.current?.startRecording({
      videoCodec: "h264",
      onRecordingError: console.error,
      onRecordingFinished: handleRecording,
})
```

### Relevant log output

```shell
–
```

### Camera Device

```json
ERROR  TypeError: Cannot convert undefined value to object

This error is located at:
    in observerComponent (created by observerComponent)
    in RCTView (created by View)
    in View (created by observerComponent)
    in observerComponent (created by CameraScreen)
    in RCTView (created by View)
    in View (created by ScreenWithoutScrolling)
    in RCTView (created by View)
    in View (created by ScreenWithoutScrolling)
    in ScreenWithoutScrolling (created by Screen)
    in RCTView (created by View)
    in View (created by KeyboardAvoidingView)
```

### Device

iPhone 15 Pro

### VisionCamera Version

3.6.3

### Can you reproduce this issue in the VisionCamera Example app?

Yes, I can reproduce the same issue in the Example app here

### Additional information

- [X] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

yspreen, 2023-10-30:
I think the issue template needs updating. the code for producing a camera device printout failed

mrousavy, 2023-10-31:
Yup again, as I just commented on your other comment - I can investigate this issue for $50: https://github.com/sponsors/mrousavy

yspreen, 2023-10-31:
yes I have seen that comment

yspreen, 2023-10-31:
Fixed in PR. My Venmo is open for donations of $50 or less @yspreen

yspreen, 2023-10-31:
on a different note, is it just me or am I losing audio recording every time I record more than one video per app session? if I restart the app it's fine but the second video onwards has no audio

yspreen, 2023-10-31:
seems to be https://github.com/mrousavy/react-native-vision-camera/issues/2085. 
I added a comment there

mrousavy, 2023-11-01:
Funny about the venmo.

Anyways, I already fixed this issue **for free** in a previous commit to main but didn't release it to npm yet, so you don't even need to donate me $50 and can take profit from this all for yourself 🙏

closed 2023-11-01 by mrousavy

yspreen, 2023-11-01:
I'm a little confused which commit you're referring to. I could reproduce this bug on main before making these changed.

mrousavy, 2023-11-01:
That's because your main branch is outdated, this is the fix I did in my free time: https://github.com/mrousavy/react-native-vision-camera/commit/cc65ec26a9ba68a9501fe14fc6f1b722368932e1