margelo/react-native-vision-camera #2111: 🐛 Black camera view after application idle

alexey-kuznetsov, 2023-10-31:
### What's happening?

I open my application, do something with camera, everything works good. Then I leave screen with camera and then exit application (just by going to my phone homescreen). If I lock screen for some short period, for example 30 seconds, I can unlock screen, open application and it will work again. But if I leave phone locked for example for 15-20 minutes and then try to open application, I see black sreen instead of camera view. If I do some actions, that cause unmounting of camera component, it will work after it will mount again.

Note: I check app state and page state to set camera inactive when it's page is not focused.

### Reproduceable Code

```tsx
function MyComponent() {
  const [hasPermission, setPermission] = useState(false);
  const cameraRef = useRef(null);
  const device = useCameraDevice('back');  
  const isFocused = useIsFocused();
  const appState = useAppState();

  useEffect(() => {
    const checkCamera = async () => {
      const permission = await Camera.getCameraPermissionStatus();
      if (permission === 'granted') {
        setPermission(true);
      } else {
        const reqPermission = await Camera.requestCameraPermission();
        if (reqPermission === 'granted') {
          setPermission(true);
        }
      }
    };
    checkCamera();
  }, []);

  return (
    <Camera
      ref={cameraRef}
      device={device}
      photo
      isActive={hasPermission && isFocused && appState === 'active' && !textModeActivated}
      style={styles.camera}
      enableZoomGesture
   />
  );
```

### Relevant log output

```shell
-
```

### Camera Device

```json
-
```

### Device

Xiaomi Mi 11 Lite

### VisionCamera Version

3.6.4

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2023-10-31:
Why. Do. You. Not. Include. The. Log. Output. ????

closed 2023-10-31 by mrousavy

joratapok, 2023-11-08:
I faced with same behaviour on android on several phones. If I in camera screen and switch to another app and wait 1-2 min camera session will stop and not recover when I get back.

2023-11-08 17:56:57.595   944-7133  CameraService           cameraserver                         E  block for PID 29189
2023-11-08 17:56:57.596   944-7133  CameraService           cameraserver                         E  block: call notifyError to the client. (pid 0)
2023-11-08 17:56:57.596   944-7133  CameraService           cameraserver                         E  notifyError: errorCode=6
2023-11-08 17:56:57.597 29189-3808  Camera                  com.agrohim                          E  Error 3
2023-11-08 17:56:57.598   669-1566  QCamera                 ven...e.camera.provider@3.0-service  E  <HAL><ERROR> stop_preview: 515: [KPI Perf] stop_preview: X, aleady stop
2023-11-08 17:56:57.598 29189-3808  RequestQueue            com.agrohim                          E  cancel failed: no repeating request exists.
2023-11-08 17:56:57.601 29189-30032 CameraManager           com.agrohim                          E  Camera 0: Error! 3
2023-11-08 17:56:57.604 29189-30032 CameraSession           com.agrohim                          E  Camera Device android.hardware.camera2.impl.CameraDeviceImpl@fa9f1ec has been disconnected!
                                                                                                    com.mrousavy.camera.core.CameraDisconnectedError: [session/camera-has-been-disconnected] The given Camera device (id: 0) has been disconnected! Error: CAMERA_IS_DISABLED_BY_ANDROID
                                                                                                    	at com.mrousavy.camera.extensions.CameraManager_openCameraKt$openCamera$2$callback$1.onError(CameraManager+openCamera.kt:49)
                                                                                                    	at android.hardware.camera2.impl.CameraDeviceImpl$CameraDeviceCallbacks.notifyError(CameraDeviceImpl.java:2042)
                                                                                                    	at android.hardware.camera2.impl.CameraDeviceImpl$CameraDeviceCallbacks.lambda$Sm85frAzwGZVMAK-NE_gwckYXVQ(Unknown Source:0)
                                                                                                    	at android.hardware.camera2.impl.-$$Lambda$CameraDeviceImpl$CameraDeviceCallbacks$Sm85frAzwGZVMAK-NE_gwckYXVQ.accept(Unknown Source:8)
                                                                                                    	at com.android.internal.util.function.pooled.PooledLambdaImpl.doInvoke(PooledLambdaImpl.java:271)
                                                                                                    	at com.android.internal.util.function.pooled.PooledLambdaImpl.invoke(PooledLambdaImpl.java:195)
                                                                                                    	at com.android.internal.util.function.pooled.OmniFunction.run(OmniFunction.java:86)
                                                                                                    	at android.os.Handler.handleCallback(Handler.java:883)
                                                                                                    	at android.os.Handler.dispatchMessage(Handler.java:100)
                                                                                                    	at android.os.Looper.loop(Looper.java:237)
                                                                                                    	at android.os.HandlerThread.run(HandlerThread.java:67)
2023-11-08 17:56:57.604 29189-30032 CameraView              com.agrohim                          E  invokeOnError(...):
2023-11-08 17:56:57.606   669-1566  QCamera                 ven...e.camera.provider@3.0-service  E  <HAL><ERROR> lock_acq: 363: failed to acquire lock
2023-11-08 17:56:57.622   931-3836  mm-camera               mm-qcamera-daemon                    E  <MCT   ><ERROR> 1331: mct_pipeline_decide_hw_wakeup: Couldn't find meta stream
2023-11-08 17:56:57.641   931-4306  mm-camera               mm-qcamera-daemon                    E  <PPROC ><ERROR> 2867: pproc_module_stop_session: pproc_module_stop_session:2867, E
2023-11-08 17:56:57.678   931-4306  QCameraQuadraCfa        mm-qcamera-daemon                    E  ~QCameraQuadraCfa:348 closing libremosaic_wrapper.so
2023-11-08 17:56:57.678   931-4306  mm-camera               mm-qcamera-daemon                    E  <PPROC ><ERROR> 2897: pproc_module_stop_session: pproc_module_stop_session:2897, X
2023-11-08 17:56:57.925   669-1566  QCamera                 ven...e.camera.provider@3.0-service  E  <HAL><ERROR> closeCamera: 3191: closeCamera:  HACK UniPlugin direct UnLoad ++
2023-11-08 17:56:57.927   669-1566  QCamera                 ven...e.camera.provider@3.0-service  E  <HAL><ERROR> closeCamera: 3195: closeCamera:  HACK UniPlugin direct UnLoad --
2023-11-08 17:56:57.957   944-944   CameraClient            cameraserver                         E  attempt to use a camera after disconnect() (pid 29189)
2023-11-08 17:56:59.570  1006-1175  Netd                    netd                                 E  getNetworkForDns: getNetId from enterpriseCtrl is netid 0
[... cut ...]
                                                                                                    	at android.os.Handler.dispatchMessage(Handler.java:100)
                                                                                                    	at android.os.Looper.loop(Looper.java:237)
                                                                                                    	at android.os.HandlerThread.run(HandlerThread.java:67)
2023-11-08 17:57:13.501 29189-30032 CameraView              com.agrohim                          E  invokeOnError(...):

mrousavy, 2023-11-09:
Format your logs. It's really bare minimum effort if I help you fix an issue in your app.

remllov91, 2025-01-17:
Anybody has a solution ? I have the same issue