margelo/react-native-vision-camera #2116: 🐛 [Android] [CodeScanner] Java_com_google_android_libraries_barhopper_BarhopperV3_recognizeBufferNative

iliapnmrv, 2023-10-31:
### What's happening?

I'm using react-native-vision-camera `v3.6.4`. On QR code scan I get google mlkit error `Java_com_google_android_libraries_barhopper_BarhopperV3_recognizeBufferNative`. I've found known issues in [mlkit repository](https://github.com/googlesamples/mlkit/issues?q=Java_com_google_android_libraries_barhopper_BarhopperV3_recognizeBufferNative). Do you have any ideas?

Did not have any issues on `v3.5.1`

Sentry error [here](https://sentry.vibeapp.ru/share/issue/c8d13f431bc7434fa9cf666342ab7b63/) may be useful 

### Reproduceable Code

```tsx
const codeScanner: CodeScanner = {
    codeTypes: ['qr'],
    onCodeScanned: codes => {},
  };

 <Camera
    device={device}
    isActive={isCameraActive}
    codeScanner={codeScanner}
    style={styles.cameraStyle}
/>
```

### Relevant log output

```shell
10-31 18:40:45.183  3395  7552 F libc    : Fatal signal 11 (SIGSEGV), code 1 (SEGV_MAPERR), fault addr 0x749afc3f53 in tid 7552 (pool-9-thread-9), pid 3395 (tscloud.scanner)
10-31 18:40:45.425  9017  9017 F DEBUG   : *** *** *** *** *** *** *** *** *** *** *** *** *** *** *** ***
10-31 18:40:45.425  9017  9017 F DEBUG   : Build fingerprint: 'samsung/a32ser/a32:11/RP1A.200720.012/A325FXXU2AVC1:user/release-keys'
10-31 18:40:45.425  9017  9017 F DEBUG   : Revision: '4'
10-31 18:40:45.425  9017  9017 F DEBUG   : ABI: 'arm64'
10-31 18:40:45.426  9017  9017 F DEBUG   : Timestamp: 2023-10-31 18:40:45+0300
10-31 18:40:45.426  9017  9017 F DEBUG   : pid: 3395, tid: 7552, name: pool-9-thread-9  >>> com.ticketscloud.scanner <<<
10-31 18:40:45.426  9017  9017 F DEBUG   : uid: 10356
10-31 18:40:45.426  9017  9017 F DEBUG   : signal 11 (SIGSEGV), code 1 (SEGV_MAPERR), fault addr 0x749afc3f53
10-31 18:40:45.426  9017  9017 F DEBUG   :     x0  00000074bdd7c878  x1  00000000000000e2  x2  0000007609005d38  x3  0000000000000014
10-31 18:40:45.426  9017  9017 F DEBUG   :     x4  0000000000000014  x5  0000000000000000  x6  0000000000001dff  x7  00000074bdd7d470
10-31 18:40:45.426  9017  9017 F DEBUG   :     x8  0000000000000000  x9  0000000000000071  x10 000000000000006d  x11 b4000074af2c8000
10-31 18:40:45.426  9017  9017 F DEBUG   :     x12 0000000000000070  x13 0000000000057f53  x14 000000749afc3f53  x15 000000749afc3a53
10-31 18:40:45.426  9017  9017 F DEBUG   :     x16 000000749af6c000  x17 000000769ae6e080  x18 00000000000e05fc  x19 00000074bdd7c998
10-31 18:40:45.426  9017  9017 F DEBUG   :     x20 00000074bdd7c938  x21 00000074bdd7cad0  x22 0000000000000071  x23 000000000000006d
10-31 18:40:45.426  9017  9017 F DEBUG   :     x24 b4000074a7372278  x25 00000000000002d0  x26 00000000133a7e60  x27 00000074bdd7d188
10-31 18:40:45.426  9017  9017 F DEBUG   :     x28 00000074bdd7d1c8  x29 00000000000006ef
10-31 18:40:45.426  9017  9017 F DEBUG   :     lr  000000748c46dff4  sp  00000074bdd7c870  pc  000000748c46e0bc  pst 0000000080000000
10-31 18:40:45.809  9017  9017 F DEBUG   : backtrace:
10-31 18:40:45.810  9017  9017 F DEBUG   :     NOTE: Function names and BuildId information is missing for some frames due
10-31 18:40:45.810  9017  9017 F DEBUG   :     NOTE: to unreadable libraries. For unwinds of apps, only shared libraries
10-31 18:40:45.810  9017  9017 F DEBUG   :     NOTE: found under the lib/ directory are readable.
10-31 18:40:45.810  9017  9017 F DEBUG   :       #00 pc 00000000001290bc  /data/user_de/0/com.google.android.gms/app_chimera/m/0000021d/dl-VisionBarcode.optional_234212100400.apk
10-31 18:40:45.810  9017  9017 F DEBUG   :       #01 pc 00000000000816a8  /data/user_de/0/com.google.android.gms/app_chimera/m/0000021d/dl-VisionBarcode.optional_234212100400.apk
10-31 18:40:45.810  9017  9017 F DEBUG   :       #02 pc 000000000007b294  /data/user_de/0/com.google.android.gms/app_chimera/m/0000021d/dl-VisionBarcode.optional_234212100400.apk
10-31 18:40:45.810  9017  9017 F DEBUG   :       #03 pc 000000000007afe8  /data/user_de/0/com.google.android.gms/app_chimera/m/0000021d/dl-VisionBarcode.optional_234212100400.apk
10-31 18:40:45.810  9017  9017 F DEBUG   :       #04 pc 000000000007b544  /data/user_de/0/com.google.android.gms/app_chimera/m/0000021d/dl-VisionBarcode.optional_234212100400.apk
10-31 18:40:45.810  9017  9017 F DEBUG   :       #05 pc 0000000000079924  /data/user_de/0/com.google.android.gms/app_chimera/m/0000021d/dl-VisionBarcode.optional_234212100400.apk
10-31 18:40:45.810  9017  9017 F DEBUG   :       #06 pc 00000000000799c4  /data/user_de/0/com.google.android.gms/app_chimera/m/0000021d/dl-VisionBarcode.optional_234212100400.apk
10-31 18:40:45.811  9017  9017 F DEBUG   :       #07 pc 000000000013eed4  /apex/com.android.art/lib64/libart.so (art_quick_generic_jni_trampoline+148) (BuildId: bd4bbcfee9112dc043c9dc3ab6bd8478)
10-31 18:40:45.811  9017  9017 F DEBUG   :       #08 pc 00000000021d19b8  /memfd:jit-cache (deleted) (offset 0x2000000) (com.google.android.libraries.barhopper.BarhopperV3.recognize+104)
10-31 18:40:45.811  9017  9017 F DEBUG   :       #09 pc 0000000002011204  /memfd:jit-cache (deleted) (offset 0x2000000) (m.pp.g+452)
10-31 18:40:45.811  9017  9017 F DEBUG   :       #10 pc 00000000020fbd20  /memfd:jit-cache (deleted) (offset 0x2000000) (m.pp.b+5376)
10-31 18:40:45.811  9017  9017 F DEBUG   :       #11 pc 00000000021c580c  /memfd:jit-cache (deleted) (offset 0x2000000) (m.bgj.a+396)
10-31 18:40:45.811  9017  9017 F DEBUG   :       #12 pc 00000000021bff08  /memfd:jit-cache (deleted) (offset 0x2000000) (m.bq.onTransact+200)
10-31 18:40:45.811  9017  9017 F DEBUG   :       #13 pc 00000000021113fc  /memfd:jit-cache (deleted) (offset 0x2000000) (android.os.Binder.transact+140)
10-31 18:40:45.811  9017  9017 F DEBUG   :       #14 pc 00000000021caa6c  /memfd:jit-cache (deleted) (offset 0x2000000) (com.google.android.gms.internal.mlkit_vision_barcode.zza.zzb+124)
10-31 18:40:45.811  9017  9017 F DEBUG   :       #15 pc 00000000021cabe0  /memfd:jit-cache (deleted) (offset 0x2000000) (com.google.android.gms.internal.mlkit_vision_barcode.zzvt.zzd+128)
10-31 18:40:45.811  9017  9017 F DEBUG   :       #16 pc 0000000002112d18  /memfd:jit-cache (deleted) (offset 0x2000000) (com.google.mlkit.vision.barcode.internal.zzn.zza+776)
10-31 18:40:45.811  9017  9017 F DEBUG   :       #17 pc 00000000020cde84  /memfd:jit-cache (deleted) (offset 0x2000000) (com.google.mlkit.vision.barcode.internal.zzk.zze+164)
10-31 18:40:45.812  9017  9017 F DEBUG   :       #18 pc 00000000021da77c  /memfd:jit-cache (deleted) (offset 0x2000000) (com.google.mlkit.vision.barcode.internal.zzk.run+92)
10-31 18:40:45.812  9017  9017 F DEBUG   :       #19 pc 00000000021e4cd8  /memfd:jit-cache (deleted) (offset 0x2000000) (com.google.mlkit.vision.common.internal.MobileVisionBase.zza+136)
10-31 18:40:45.812  9017  9017 F DEBUG   :       #20 pc 00000000021e3aac  /memfd:jit-cache (deleted) (offset 0x2000000) (com.google.mlkit.vision.common.internal.zza.call+76)
[... 52 lines cut ...]
  "hasTorch": true,
  "position": "back",
  "id": "0"
}
```

### Device

Android Samsung, Honor, Huawei, Redmi

### VisionCamera Version

3.6.4

### Can you reproduce this issue in the VisionCamera Example app?

Yes, I can reproduce the same issue in the Example app here

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

zzz08900, 2023-11-01:
You can using try another version of MLKit by editing the `build.gradle` of vision-camera.

I have no idea which (or if any) other version works so, good hunting.

rodgomesc, 2023-11-21:
can you guys please test if this change solves the issue for you?

remove `node_modules` modify vision-camera in `package.json` to be like below and see what happens
```json
"dependencies": {
  "react-native-vision-camera": "github:rodgomesc/react-native-vision-camera#cad3699"
}
```

WitchMad, 2023-11-21:
Appears to be working! I will run it on others devices to confirm.

rodgomesc, 2023-11-21:
> Appears to be working! I will run it on others devices to confirm.

looking forward to know if it will work then i'll send a pr

OtayNacef, 2023-11-21:
> can you guys please test if this change solves the issue for you?
> 
> remove `node_modules` modify vision-camera in `package.json` to be like below and see what happens

Thank you for the fix! I've tested it on multiple devices, and it works perfectly. go ahead and create a pull request.

iliapnmrv, 2023-11-22:
does anyone get an npm error once i set `"github:rodgomesc/react-native-vision-camera#cad3699"`?

```
npm ERR! code ENOENT
npm ERR! syscall open
npm ERR! path /Users/user/.npm/_cacache/tmp/git-cloneOHBQrz/package.json                                                  
npm ERR! errno -2
npm ERR! enoent ENOENT: no such file or directory, open '/Users/user/.npm/_cacache/tmp/git-cloneOHBQrz/package.json'      npm ERR! enoent This is related to npm not being able to find a file.
npm ERR! enoent 
npm ERR! A complete log of this run can be found in: /Users/user/.npm/_logs/debug-0.log          
```

netdown, 2023-11-23:
@iliapnmrv Use gitpkg 
`"react-native-vision-camera": "https://gitpkg.now.sh/rodgomesc/react-native-vision-camera/package?cad3699"`

iliapnmrv, 2023-11-23:
> @iliapnmrv Use gitpkg `"react-native-vision-camera": "https://gitpkg.now.sh/rodgomesc/react-native-vision-camera/package?cad3699"`

it works, thanks a lot❤️

SamiChab, 2023-12-06:
Same error happened to me on `v3.6.9`. Going back to `v3.5.1` seems to solve this issue.

mrousavy, 2024-01-08:
Heyo! Does https://github.com/mrousavy/react-native-vision-camera/pull/2355 maybe fix your issue?

4yki, 2024-01-08:
still seems to be an issue in 3.7.0 .. Tried both VisionCamera_enableCodeScanner=true and VisionCamera_enableCodeScanner=false 
Samsung device (SDK 33)

augustjoergensen, 2024-01-11:
I have the same problem, happens to around 4% of my users - so pretty frequent. 3.7.0 did not solve this issue.

marcshilling, 2024-01-11:
This is a strange one. I'm seeing some reports of this crash coming from my prod app (vision-camera v3.6.17).

Last night I was able to reproduce it 100% of the time in my app (crashing right after a barcode scan). This morning: same app, same steps to reproduce and I can't seem to make it happen.

marcshilling, 2024-01-11:
@rodgomesc did you ever make a PR for this fix?

parlet, 2024-01-16:
@mrousavy, Yeah, I remember doing a cursory run through the code diffs before, and I didn't see anything that would attract my attention either, although of course I don't have an in-depth knowledge of the code base.

I hope I'll find some time to debug this issue and dig deeper into what within this library might help, although ultimately, the problem lies in the `libbarhopper_v3.so` and that's where we'd ideally find the solution. Something tells me this might be a race condition.

Anyway, thank you very much for your attention to this issue. =)
Just let me know If there is anything more I could assist with.

marcshilling, 2024-01-16:
I might have found a way to avoid or minimize this. I'm curious if others are doing what I was and unmounting the camera component/navigating away right after they get a scan. I can't seem to reproduce the crash anymore after:
1. Setting `isActive={false}` on the camera after I get a scan
2. Inserting a brief setTimeout delay of 500ms before calling my Navigation.dismiss

Give this a shot in your apps.

marcshilling, 2024-01-22:
I'm still getting reports of it in prod even on 3.5.1 for what it's worth, although definitely seems less frequent

parlet, 2024-01-23:
By the way, for tracking purposes, this bug is logged in the Google's public issue tracker as [#294890843](https://issuetracker.google.com/issues/294890843) and [#314930525](https://issuetracker.google.com/issues/314930525).
We may have better luck if we tackle this issue at its source.

marcshilling, 2024-01-27:
I can also seem to reproduce this every once in a while by simply backgrounding the app while the camera view is active, then resuming the app. On the occasional resume, this crash will occur right away.

marcshilling, 2024-01-30:
Also this guy: https://github.com/googlesamples/mlkit/issues/714

mgcrea, 2024-02-05:
I'm hitting this as well (tried both `3.8.2` and `3.9.0-beta.0`) when using [my barcode scanner plugin](https://github.com/mgcrea/vision-camera-barcode-scanner) that leverages a frame processor and that relies as well on Google MLKit. Can be reproduced pretty quickly with a hard mount/unmount interval:

```tsx
  useEffect(() => {
    const interval = setInterval(() => {
      setIsMounted(prevIsMounted => {
        console.log({isMounted: prevIsMounted});
        return !prevIsMounted;
      });
    }, 3000);
    return () => {
      clearInterval(interval);
    };
  }, []);
```

gaurav00729, 2024-02-08:
Java_com_google_android_libraries_barhopper_BarhopperV3_recognizeBufferNative.
Getting the same issue in production, It occurs randomly while scanning and navigating between screens.

mgcrea, 2024-03-25:
Might be worth trying the latest `4.0.0-beta.10` as it seems to have fixed this race condition issues when unmounting the camera, see https://github.com/mrousavy/react-native-vision-camera/issues/2341#issuecomment-2018139327, worth a shot.

mrousavy, 2024-04-22:
Hey - I think this issue has been fixed in [VisionCamera 4.0.0](https://github.com/mrousavy/react-native-vision-camera/releases/tag/v4.0.0). 🥳 

Please try V4 and let me know if you still experience this issue;
- if not, please consider [💖 **sponsoring me on GitHub** 💖](https://github.com/sponsors/mrousavy/) to support the development of VisionCamera and thank me for my time spent on fixing bugs and building new features.
- if you still see this issue, please comment and we can re-open this. But please update your native logs with the native (Xcode/Android Studio) logs from running VisionCamera V4 so I can investigate this.

closed 2024-04-22 by mrousavy

oscar-b, 2024-04-22:
I haven't been able to crash v4 yet, this issue seems fixed! 🎉

pke, 2024-04-26:
@oscar-b did you upgrade your ReactNative version to use 4.0?

oscar-b, 2024-04-26:
> @oscar-b did you upgrade your ReactNative version to use 4.0?

RNVC v4.0.1 yes.

pke, 2024-04-26:
No I mean the React Native version to 0.72.3?

oscar-b, 2024-04-26:
> No I mean the React Native version to 0.72.3?

No Expo v50, using v0.73.6

spsaucier, 2024-06-05:
We are still seeing this in Vision 4.0.3.

antoinerousseau, 2024-08-14:
A new version 17.3 of mlkit got out a week ago, maybe it's better with it? can't find the changelog
https://mvnrepository.com/artifact/com.google.mlkit/barcode-scanning/17.3.0

mrousavy, 2024-08-14:
you can test https://github.com/mrousavy/react-native-vision-camera/pull/3135

SliverYuki, 2025-01-02:
> I might have found a way to avoid or minimize this. I'm curious if others are doing what I was and unmounting the camera component/navigating away right after they get a scan. I can't seem to reproduce the crash anymore after:我可能已经找到了避免或最小化这种情况的方法。我很好奇其他人是否正在做我所做的事情并卸载相机组件/在扫描后立即导航离开。之后我似乎无法再重现崩溃：
> 
> 1. Setting `isActive={false}` on the camera after I get a scan扫描后在相机上设置`isActive={false}`
> 2. Inserting a brief setTimeout delay of 500ms before calling my Navigation.dismiss在调用我的 Navigation.dismiss 之前插入短暂的 500 毫秒的 setTimeout 延迟
> 
> Give this a shot in your apps. I'm on `v3.8.0` now.在您的应用程序中尝试一下。我现在使用的是`v3.8.0` 。

I tried your method and the crashes became less frequent but still happen occasionally